python-microscopy/python-microscopy #725: [BUG] we don't propagate metadata to density mappings (e.g. Sample.SlideRef)

barentine, 2020-12-18:
**Describe the bug**
I try to access 'Sample.SlideRef' metadata of a density mapping (histogram) and KeyError because we don't copy entries and overwrite voxelsize, we just give it a new one.

**To Reproduce**
Steps to reproduce the behavior:
1. do a density mapping
2. hope to find metadata from the localizations
3. not find it

David-Baddeley, 2020-12-20:
We do record the `SourceImageID`, which should in theory be enough to find the the metadata about the localisations, if in a somewhat roundabout way. 

This probably should fall into a more general "What should be propagated discussion" - it probably does make sense to propagate all the `'Sample.*'` keys, although I I'm not sure about any of the other metadata from the localisation file.

David-Baddeley, 2020-12-20:
I should note that the copy all entries and over-write where needed strategy that is adopted by a lot of modules is a quick and dirty solution, not the way things **should** work. Modules should only copy things which are objectively the same, but not anything that isn't  (potentially preserving the input metadata under the `'Source'` key).

David-Baddeley, 2020-12-20:
Actually I correct myself, for rendered images it should all there under the `'Source.*'` keys - i.e. you'd need to use `Source.Sample.XXX` instead of `Sample.XXX`

David-Baddeley, 2020-12-20:
Changed the title - this isn't a bug, as the metadata is available. There could be an argument, however, for also propagating sample related metadata up to the top level of the metadata hierarchy.

barentine, 2020-12-23:
I would argue we are still looking at the same sample, and therefore it would be nice not to need to know a-priori whether you need to dig or not. are you OK with me explicitly passing through sample related keys then?

barentine, 2021-01-20:
should be set

closed 2021-01-20 by barentine