python-microscopy/python-microscopy #727: [BUG] PYMEAcquire hangs on close

zacsimile, 2020-12-19:
**Describe the bug**
Open PYMEAcquire with default init script (or any init script). If I try to close PYMEAcquire, it hangs and won't close until I sent a Ctrl+C call.

<details>
<summary>Traceback</summary>
<pre>
DEBUG:matplotlib:CACHEDIR=C:\Users\scope_baby\.matplotlib
DEBUG:matplotlib.font_manager:Using fontManager instance from C:\Users\scope_baby\.matplotlib\fontlist-v330.json
DEBUG:matplotlib.pyplot:Loaded backend WXAgg version unknown.
INFO:numexpr.utils:Note: NumExpr detected 12 cores but "NUMEXPR_MAX_THREADS" not set, so enforcing safe limit of 8.
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
C:\Users\scope_baby\miniconda3\envs\pyme\lib\_collections_abc.py:702: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  return len(self._mapping)
C:\Users\scope_baby\miniconda3\envs\pyme\lib\_collections_abc.py:720: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  yield from self._mapping
c:\users\scope_baby\code\python-microscopy\PYME\misc\extraCMaps.py:73: MatplotlibDeprecationWarning:
The revcmap function was deprecated in Matplotlib 3.2 and will be removed two minor releases later. Use Colormap.reversed() instead.
  cmapdat_r = cm.revcmap(ndat[cmapname])
attempting to load load logging config from c:\users\scope_baby\code\python-microscopy\PYME\Acquire\logging.json
INFO:PYME.Acquire.PYMEAcquire:using initialization script c:\users\scope_baby\code\python-microscopy\PYME\Acquire\Scripts\init.py

DEBUG:PYME.Acquire.acquiremainframe:Init run, waiting on background threads

c:\users\scope_baby\code\python-microscopy\PYME\Acquire\ui\splashScreen.py:124: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\scope_baby\code\python-microscopy\PYME\Acquire\ui\splashScreen.py:124: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
[0, 50]
0.1
DEBUG:PYME.Acquire.acquiremainframe:Backround initialization done

DEBUG:PYME.Acquire.acquiremainframe:Starting post-init

DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Simulation UI

[0, 0]
0.1
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Camera controls

c:\users\scope_baby\code\python-microscopy\PYME\ui\manualFoldPanel.py:37: wxPyDeprecationWarning: Using deprecated class. Use Colour instead.
  col = wx.NamedColour(col)
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Sample database

DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Fake DMD

ERROR:PYME.Acquire.acquiremainframe:Error whilst initializing Fake DMD GUI
Traceback (most recent call last):
  File "c:\users\scope_baby\code\python-microscopy\PYME\Acquire\acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "c:\users\scope_baby\code\python-microscopy\PYME\Acquire\ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "c:\users\scope_baby\code\python-microscopy\PYME\Acquire\Scripts\init.py", line 120, in fake_dmd
    from PYMEnf.Hardware import FakeDMD, DMDGui
ModuleNotFoundError: No module named 'PYMEnf'
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Laser controls

DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Focus Keys

DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Action manager

0
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Tiling

DEBUG:PYME.Acquire.acquiremainframe:Run all custom GUI init tasks

c:\users\scope_baby\code\python-microscopy\PYME\IO\DataSources\ArrayDataSource.py:45: FutureWarning: Using a non-tuple sequence for multidimensional indexing is deprecated; use `arr[tuple(seq)]` instead of `arr[seq]`. In the future this will be interpreted as an array index, `arr[np.array(seq)]`, which will result either in an error or a different result.
  r = self.data.__getitem__(keys)
0
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC458B8> Display True True
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC458B8> Integration Time True True
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC458B8> Current Slide True True
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC458B8> Laser Control True True
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC458B8> Positioning True True
seqd: update display
<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC45C18> Z-Stack False True
seqd: update display
File
DEBUG:PYME.Acquire.SpoolController:Updating series counter

DEBUG:PYME.Acquire.SpoolController:Looking for C:\Users\scope_baby\PYMEData\scope_baby\2020_12_19\19_12_series_A.h5 on local machine

WARNING:PYME.Acquire.ui.HDFSpoolFrame:Compression settings invalid, disabling quantization

<PYME.ui.manualFoldPanel.foldPanel object at 0x0000028FFFC45C18> Tiling True True
DEBUG:PYME.Acquire.acquiremainframe:Finished post-init

c:\users\scope_baby\code\python-microscopy\PYME\ui\histLimits.py:289: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\scope_baby\code\python-microscopy\PYME\DSView\scrolledImagePanel.py:52: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\scope_baby\code\python-microscopy\PYME\DSView\arrayViewPanel.py:1154: wxPyDeprecationWarning: Call to deprecated item ImageFromData. Use :class:`Image` instead.
  img = wx.ImageFromData(ima.shape[1], ima.shape[0], ima.ravel())
c:\users\scope_baby\code\python-microscopy\PYME\DSView\arrayViewPanel.py:513: wxPyDeprecationWarning: Call to deprecated item BitmapFromImage. Use :class:`wx.Bitmap` instead
  im2 = wx.BitmapFromImage(im)
Warning: not keeping up with camera, giving up with 7 frames still in buffer
INFO:root:All cleanup functions called

INFO:root:Remaining Threads:
MainThread, None
Thread-1, <function worker at 0x0000028FFBDBBD38>
Thread-2, <function worker at 0x0000028FFBDBBD38>
Thread-3, <function Pool._handle_workers at 0x0000028FFBDB49D8>
Thread-4, <function Pool._handle_tasks at 0x0000028FFBDB4A68>
Thread-5, <function Pool._handle_results at 0x0000028FFBDB4AF8>
Thread-6, <function get_ns at 0x0000028FFF7A1E58>
zeroconf-Engine, None
zeroconf-Reaper, None
Thread-7, <bound method ZCListener._poll_services_open of <PYME.misc.pyme_zeroconf.ZCListener object at 0x0000028FFF894F48>>
zeroconf-ServiceBrowser__pyme-http._tcp.local., None
Thread-8, <bound method ActionManager._monitor_defunct of <PYME.Acquire.ActionManager.ActionManager object at 0x0000028FFF8A0F48>>
</pre>
</details>

If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.7.9

David-Baddeley, 2020-12-20:
This is a semi-common problem due to the number of threads which run, and is the reason for all the thread debugging stuff which gets printed at the end. I would expect the simulator to exit cleanly though ...

zacsimile, 2020-12-20:
Me, too. And the simulator used to, but it didn't with the latest version of PYME when I checked yesterday. Ran it a couple of times to be sure.

David-Baddeley, 2020-12-20:
There might be a regression with the new `_monitor_defunct` action manager stuff. I should change the thread logging code to either show if the threads are deamon threads, or just filter out all the deamon threads.

David-Baddeley, 2020-12-20:
Confirmed, it's `_monitor_defunct` - all the other threads are deamon threads.

David-Baddeley, 2020-12-20:
With line 669 of acquiremainframe.py modified as

```python
msg += '%s, %s, daemon=%s\n' % (t.name, t._target, t.daemon)
```

we get

```
INFO:root:Remaining Threads:
MainThread, None, daemon=False
Thread-1, <function worker at 0x11f41fe60>, daemon=True
Thread-2, <function worker at 0x11f41fe60>, daemon=True
Thread-3, <function Pool._handle_workers at 0x11f422b00>, daemon=True
Thread-4, <function Pool._handle_tasks at 0x11f422b90>, daemon=True
Thread-5, <function Pool._handle_results at 0x11f422c20>, daemon=True
zeroconf-Engine-123145345605632, None, daemon=True
zeroconf-Reaper_123145350860800, None, daemon=True
Thread-9, <bound method ZCListener._poll_services_open of <PYME.misc.pyme_zeroconf.ZCListener object at 0x1c21d4c7d0>>, daemon=True
zeroconf-ServiceBrowser__pyme-http._tcp.local._123145361371136, None, daemon=True
Thread-11, <bound method ActionManager._monitor_defunct of <PYME.Acquire.ActionManager.ActionManager object at 0x1c21d4ccd0>>, daemon=False
```

David-Baddeley, 2020-12-20:
And making that a deamon thread has fixed it. PR to follow.

barentine, 2020-12-20:
Sorry team, this one's on me

zacsimile, 2020-12-21:
Confirmed #728 fixes this on mac and win10.

closed 2020-12-21 by zacsimile