python-microscopy/python-microscopy #732: Closing the main window after performing a channel split function, closes all windows in dh5view.[BUG]

mdlessard75, 2020-12-21:
**Describe the bug**
Closing the main window after performing a channel split function, closes all windows in dh5view.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a multi-channel image.
2. Click on Process --> Channel Split
3. Close the window that contains the multi-channel image (not the resulting seperate channel images)

**Expected behavior**
I would expect only the window that contained the multi-channel image to close, not all the windows to close.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- OS: Win10 64
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2020-12-21:
This is a quirk that had been on the TODO list for a while as a clean solution is a little trickier than it might seem, and the status quo has some advantages (it gives a simple way of closing dh5view and all open files). The fact that the first window you open has a special 'master' window status, and that all subsequently opened windows are children of that is not super intuitive, however. There are a few possible solutions:

- Make the parent of newly opened windows None. This is what we used to have, and means that you have to manually close every window in order to close the program. This gets really tiresome if you've done a lot of different processing steps, which is the reason we switched to the current behaviour. This is especially true when the viewer is launched from PYMEVis
- Do the above, but have a `File-->Quit` menu item as well as a `File-->Close`, which allows you to close everything. The logic here would be a bit messy (we somehow have to record all the windows so we can iterate over and close them), but should be doable.
- Have a separate "master" window / toolbar (as is done for ImageJ, GIMP), of which all viewers are children. This has some advantages, and might make sense when in the context of a common entry point, but is aesthetically pretty gross.
- Change to a tabbed interface where new images open in their own tabs.
- Keep the status-quo, but warn when closing the "master" window. 

The splitting stuff might have a couple of additional complications, as the split image is not a copy/new image, but rather a view of the original image (and as a result needs to keep the original image around, although it probably doesn't need to keep the GUI for the original image around)

David-Baddeley, 2020-12-21:
Yet another option: 

- have a 'hidden' master window (as we do for PYMEClusterOfOne) which gives you a task-bar icon and possibly a tray menu

PR #734 (2020-12-23): Issue731
Addresses issue #731. Partial fix for #732

- fixes `File-->Close` (#731) in dh5view
- adds re-usable `File-->Close` and `File-->Quit` functions to `AUIFrame` (so they work across all PYME programs) (Partial #732 fix)
- To fully fix #732, we would additionally need to change a bunch of `View3D()` and `ViewIm3D()` calls to use `parent=None`
- This is potentially slightly risky for PYMEAcquire - if a user chooses `File-->Quit` in an image displayed in PYMEAcquire, it will quit everything. TODO - do we need to modify this behaviour? Intercept `EVT_CLOSE` in PYMEAcquire and warn?
closed 2020-12-23 by David-Baddeley