python-microscopy/python-microscopy #736: [BUG] if we fail to push acquisition events, or rule pusher fails to find them, we push forever

barentine, 2020-12-23:
**Describe the bug**
```
DEBUG:PYME.cluster.rules:DataSource.__class__: <class 'PYME.IO.DataSources.ClusterPZFDataSource.DataSource'>

PatchedServiceInfo(type='_pyme-taskdist._tcp.local.', name='PYMERuleServer:greenback - PID:18119._pyme-taskdist._tcp.local.', addresses=[b'\n\x96\x07\x84'], port=15346, weight=0, priority=0, server='PYMERuleServer:greenback - PID:18119._pyme-taskdist._tcp.local.', properties={}) b'\n\x96\x07\x84'
DEBUG:PYME.cluster.rules:local_queues: []

INFO:PYME.cluster.rules:no local rule server, choosing one at random

DEBUG:root:resultsURI: http://10.150.7.4:15348/__aggregate_h5r/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis/23_12_series_B.h5r

DEBUG:root:Successfully released tasks (17901:20000)

DEBUG:root:fileFormattedResults - URI: http://10.150.7.4:15348/__aggregate_h5r/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis/23_12_series_B.h5r/MetaData

DEBUG:PYME.IO.clusterIO:Request for b'http://10.150.7.4:15348/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/23_12_series_B.pcs/' failed with error: 404

DEBUG:PYME.IO.clusterIO:Could not find Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/23_12_series_B.pcs/events.json, retrying ...

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:PYME.IO.clusterIO:Could not find Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/23_12_series_B.pcs/events.json, retrying ...

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:PYME.IO.clusterIO:could not find file Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/23_12_series_B.pcs/events.json on cluster: cluster nodes = ['PYMEDataServer [villains]:greene - PID:6367', 'PYMEDataServer [villains]:drno - PID:6704', 'PYMEDataServer [villains]:goldfinger - PID:7078', 'PYMEDataServer [villains]:lechiffre - PID:11956', 'PYMEDataServer [villains]:silva - PID:8146', 'PYMEDataServer [villains]:greenback - PID:18461', 'PYMEDataServer [villains]:scaramanga - PID:12708', 'PYMEDataServer [villains]:elektra - PID:8111', 'PYMEDataServer [villains]:stromberg - PID:6722', 'PYMEDataServer [villains]:blofeld - PID:7696']

WARNING:root:No handler for data type found, using pickle

DEBUG:root:fileFormattedResults - URI: http://10.150.7.4:15348/__aggregate_h5r/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis/23_12_series_B.h5r/Events

'http://10.150.7.4:15348/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis/23_12_series_B.h5r.json'
DEBUG:PYME.cluster.rules:Successfully created rule

DEBUG:root:task pusher poll loop started

Warning: not keeping up with camera, giving up with 2 frames still in buffer
DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 600, _next_release_start: 0

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (0:600)

n live: 0
DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 4700, _next_release_start: 600

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (600:4700)

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 4700, _next_release_start: 4700

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 4700, _next_release_start: 4700

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 4700, _next_release_start: 4700

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 4700, _next_release_start: 4700

[... 2814 lines cut ...]
DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

DEBUG:root:numTotalFrames: 20000, _next_release_start: 20000

INFO:root:Shutting down piezo

```

</details>

barentine, 2020-12-23:
note in this case the events file not being there the first time was probably my bad because I was trying to pop HTTPSpooler poll threads off the megalist of them and forgot to start with the largest index so we don't index error. That said, I hit similar issues yesterday without my hacking around

David-Baddeley, 2021-01-10:
Timeouts for series completion are a rather messy as they introduce some very application-specific logic deep into the core (or need to be passed as a parameter for every series, which also makes things gross). Ultimately we want to be able to support both your fast acquisition and, e.g. a 12hr PAINT experiment, or even a multi-day light-sheet time-lapse recording. Timeouts are also the reason we got here in the first place (a write timeout on the events file), and have a habit of causing different problems further down the track.

Before adding additional timeouts, I think there are a couple of avenues we can pursue:
- making the directory listing fast enough that polling on a couple of series which don't complete is not an issue (once we are consistently adding files to the local listing cache when we write them, we can potentially - at least on the acquisition side - answer the `.exists()` question without any cluster calls as we know it would have been written from our process (and should therefore be in our cache if it exists).
- have the pusher have an alternative completion callback which can get called directly by the spooler
- tear down everything associated with a series on error.