python-microscopy/python-microscopy #737: [BUG] PYMECluster handling of file exists when pushing series/rules from windows  botches path joining

barentine, 2020-12-23:
**Describe the bug**
```
DEBUG:root:we have unpublished frames - push them

DEBUG:PYME.IO.clusterIO:could not find file Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/23_12_series_B.pcs/events.json on cluster: cluster nodes = ['PYMEDataServer [villains]:greene - PID:6367', 'PYMEDataServer [villains]:drno - PID:6704', 'PYMEDataServer [villains]:goldfinger - PID:7078', 'PYMEDataServer [villains]:lechiffre - PID:11956', 'PYMEDataServer [villains]:silva - PID:8146', 'PYMEDataServer [villains]:greenback - PID:18461', 'PYMEDataServer [villains]:scaramanga - PID:12708', 'PYMEDataServer [villains]:elektra - PID:8111', 'PYMEDataServer [villains]:stromberg - PID:6722', 'PYMEDataServer [villains]:blofeld - PID:7696']

WARNING:root:No handler for data type found, using pickle

DEBUG:root:fileFormattedResults - URI: http://10.150.7.8:15348/__aggregate_h5r/Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis\23_12_series_B_1.h5r/Events

Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\unifiedIO.py", line 49, in assert_name_ok
    assert check_name(name) == True
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\frameWrangler.py", line 235, in onExpReady
    self.onFrame.send(sender=self, frameData=d)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in send
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in <listcomp>
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\HTTPSpooler.py", line 292, in OnFrame
    sp.Spooler.OnFrame(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Spooler.py", line 222, in OnFrame
    self.StopSpool()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\HTTPSpooler.py", line 240, in StopSpool
    sp.Spooler.StopSpool(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Spooler.py", line 170, in StopSpool
    self.onSpoolStop.send(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in send
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in <listcomp>
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 500, in SpoolStopped
    self.onSpoolStop.send(self)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in send
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in <listcomp>
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\rules.py", line 379, in post_rules
    rule_factory_chain[0].get_rule(context=context).push()
  File "c:\users\bergamot\code\python-microscopy\PYME\cluster\rules.py", line 239, in push
    post_args = self.prepare()
  File "c:\users\bergamot\code\python-microscopy\PYME\cluster\rules.py", line 546, in prepare
    clusterIO.put_file(self.resultsMDFilename, self.mdh.to_JSON().encode(), serverfilter=self.serverfilter)
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\clusterIO.py", line 961, in put_file
    unifiedIO.assert_name_ok(filename)
  File "c:\users\bergamot\code\python-microscopy\PYME\IO\unifiedIO.py", line 51, in assert_name_ok
    raise AssertionError('Name "%s" is invalid. Names must only include alphanumeric characters and underscore' % name)
AssertionError: Name "Bergamot/2020_12_22/EMC0606_secondaryabcomparison/next_day/B2/000/analysis\23_12_series_B_1.h5r.json" is invalid. Names must only include alphanumeric characters and underscore
DEBUG:root:Successfully released tasks (17300:20000)

```

can lead to getting analysis series not where you expect them too
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. use PYMEAcquire / PYMECluster
2. bog things down/ do something weird so you hit a fileexists error

**Expected behavior**
if we're going to add a _1 that's Ok but would be nice to put it in the right spot / not throw an error in PYMEAcquire about it

David-Baddeley, 2021-01-10:
Will also hit this when using PYMECluster for re-analysis (which is the main use case for subscripting the analysis files).

David-Baddeley, 2021-11-11:
Have a suspicion this has been fixed, need to verify ...

closed 2026-07-02 by David-Baddeley