python-microscopy/python-microscopy #738: [BUG] lack of tolerance on camera integration time means spec'ing it in setstate will restart camera unnecessarily

barentine, 2020-12-23:
**Describe the bug**
I image at 800 Hz, but really it's at 795.53 Hz, and my 'actual' integration time is something like `0.0012570225563909773`
None of that is 'bug' but I do think it's ~buggy to restart the camera/framewrangler on each series if I have the scope state spec'd in a protocol to include the integration time.

What makes sense here as a fix? Do we put the shim as a state handler (target, actual, like stage positions which currently have a comment saying 'boo' next to them?) Or do we stuff it in the camera, e.g. Set/Get return the same, but there's also a GetActual?

barentine, 2020-12-23:
one potential for within the camera class
```
    def GetIntegTime(self):
        return np.round(self.get_actual_integration_time(), 5)
    
    def get_actual_integration_time(self):
        return self.getCamPropValue('EXPOSURE TIME')
```

csoeller, 2020-12-23:
This is also related to subtle issues with camera map naming. I have, for example, noticed that with the Zyla a chosen 100 ms frame time can result in maps named 99 ms or 100 ms depending which preamp mode is active (since that leads to subtle changes in actual integ time).

Not sure which level of rounding should be adopted for the map purpose and perhaps also agreement on how close a map timing needs to be to actual timing to be acceptable (presumably a fractional difference being sensible?).

Maybe conventions could be adopted for both our cases while we are at it?

David-Baddeley, 2020-12-23:
I really don't think this is a bug - and think the status quo (if you set the state you are guaranteed that the state will be set) is far more robust than a solution where, e.g. you either remember the last state you requested (which might end up out of sync with the hardware) or rely on some hard-coded tolerance which might not be appropriate for all use cases. Our current approach is fairly simple and transparent and I think that is desirable. The one area where I could see an argument for some tolerancing is when stage setting positions (but I think that at least some of the stages already have this built in). I'd want a compelling argument for why the current behaviour is a problem for the camera properties.

If the real issue is that restarting the camera takes too long, there are things we could look at here (e.g. re-using buffers if the ROI etc ... is the same to avoid memory allocation overhead), but I'd want some good evidence that this is a problem which really effects realistic imaging scenarios (i.e. is e.g. 100ms down-time once-per series an issue that really effects realistic biological imaging workflows). 

@csoeller - I'm pretty sure camera map selection is already toleranced, although this is something that could definitely use revisiting. If I recall correctly, we use a hard-coded 1ms tolerance, and the interval boundaries fall on integral values of ms. Given the behaviour you describe above, making the tolerance intervals [x-0.5, x+ 0.5) rather than [x, x+1) might be sensible.

David-Baddeley, 2020-12-23:
That said, I would entertain the possibility of adding something like:

`StateHandler.ensure_close(self, state={}, tolerance={})`

I definitely don't think that tolerancing should be built into the camera class, as it's too hard to predict how things could be used and it adds complexity.
I do not think that this should be the default behaviour, however.