python-microscopy/python-microscopy #741: [BUG] clusterIO struggling at high bandwidth at the moment

barentine, 2020-12-23:
**Describe the bug**
A clear and concise description of what the bug is.
performance is high and we're keeping up, but dropping quite some frames on io errors
[image]

```
===========================================================================
Error in rule: 000083-95aa046a0c394811a3df841956cc6c53 running task: 3500 on elektra:7858

taskDescr:
----------
{'id': '000083-95aa046a0c394811a3df841956cc6c53~3500', 'type': 'localization', 'taskdef': {'frameIndex': '3500', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2020_12_23/EMC0607_try2/B1/000/analysis/23_12_series_L.h5r.json'}, 'inputs': {'frames': 'PYME-CLUSTER://villains/Bergamot/2020_12_23/EMC0607_try2/B1/000/23_12_series_L.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.10:15348/__aggregate_h5r/Bergamot/2020_12_23/EMC0607_try2/B1/000/analysis/23_12_series_L.h5r/FitResults', 'driftResults': 'http://10.150.7.10:15348/__aggregate_h5r/Bergamot/2020_12_23/EMC0607_try2/B1/000/analysis/23_12_series_L.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 308, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 420, in __call__
    bufferManager.updateBuffers(md, self.dataSourceModule, self.bufferLen)
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 94, in updateBuffers
    self.dBuffer = buffers.dataBuffer(DataSource(md.dataSourceID, md.taskQueue), bufferLen)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 58, in __init__
    self.mdh.update(json.loads(clusterIO.get_file(mdfn, self.clusterfilter)))
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 806, in get_file
    raise IOError("Specified file could not be found: %s" % filename)
OSError: Specified file could not be found: Bergamot/2020_12_23/EMC0607_try2/B1/000/23_12_series_L.pcs/metadata.json

```

**additional context**
posting localization rules using the new refactor on spool-stop.  Imaging 256 x 1024 at 800 Hz

David-Baddeley, 2021-01-10:
Would be interesting to know what was behind that IO error. The code that throws that also generates a debug log output listing which cluster nodes could be found which might be interesting.

```python
    if (len(locs) == 0):
        # we did not find the file
        logger.debug('could not find file %s on cluster: cluster nodes = %s' % (filename, get_ns().list()))
        raise IOError("Specified file could not be found: %s" % filename)
```

There is a chance that the tidy-up/delisting code is kicking in and you are dropping data servers for some reason. We should add some debug logging to the delisting code so we can see if that is the case (the delisting code has a pretty generous timeout of 10s - if a server doesn't respond within that time it's not usefully there and there are other issues we need to address, but it would be good to know if this is happening).

barentine, 2021-01-17:
Worth noting things currently seem to run fine up until a number of localization rules have already been completed, then eventually I start getting these socket timesouts and get_file IOErrors

```
===========================================================================
Error in rule: 000020-a9dad3bb66cb43e49395727b88854f5c running task: 14398 on stromberg:26450

taskDescr:
----------
{'id': '000020-a9dad3bb66cb43e49395727b88854f5c~14398', 'type': 'localization', 'taskdef': {'frameIndex': '14398', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2021_1_15/EMC0610S1/A1/002/analysis/15_1_series_IY.h5r.json'}, 'inputs': {'frames': 'pyme-cluster://villains/Bergamot/2021_1_15/EMC0610S1/A1/002/15_1_series_IY.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.5:15348/__aggregate_h5r/Bergamot/2021_1_15/EMC0610S1/A1/002/analysis/15_1_series_IY.h5r/FitResults', 'driftResults': 'http://10.150.7.5:15348/__aggregate_h5r/Bergamot/2021_1_15/EMC0610S1/A1/002/analysis/15_1_series_IY.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connection.py", line 160, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 392, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 1254, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 1300, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 1249, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 1036, in _send_output
    self.send(msg)
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 974, in send
    self.connect()
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connection.py", line 187, in connect
    conn = self._new_conn()
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connection.py", line 167, in _new_conn
    % (self.host, self.timeout),
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.HTTPConnection object at 0x7f0f3b24ef28>, 'Connection to 10.150.7.132 timed out. (connect timeout=0.5)')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/util/retry.py", line 446, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='10.150.7.132', port=15348): Max retries exceeded with url: /Bergamot/2021_1_15/EMC0610S1/A1/002/15_1_series_IY.pcs/frame14366.pzf (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f0f3b24ef28>, 'Connection to 10.150.7.132 timed out. (connect timeout=0.5)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 309, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 437, in __call__
    bufferManager.bBuffer.calc_background(self.bgindices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 256, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 219, in update_buffer
    frame_data = np.ascontiguousarray(self.data_buffer.getSlice(frame), dtype=np.float32)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/buffers.py", line 28, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 71, in getSlice
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
[... 138 lines cut ...]
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 309, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 437, in __call__
    bufferManager.bBuffer.calc_background(self.bgindices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 256, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 219, in update_buffer
[... cut ...]
    r = adapter.send(request, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/adapters.py", line 504, in send
    raise ConnectTimeout(e, request=request)
requests.exceptions.ConnectTimeout: HTTPConnectionPool(host='10.150.7.132', port=15348): Max retries exceeded with url: /Bergamot/2021_1_15/EMC0610S1/A1/002/15_1_series_IY.pcs/frame13410.pzf (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f2f19af30f0>, 'Connection to 10.150.7.132 timed out. (connect timeout=0.5)'))

```

barentine, 2021-01-17:
That things work fine at the start makes me wonder whether any of our caches loose some efficiency as they increase size (the ordered part of an ordered dict?). Short of a larger clusterIO rewrite like we're gearing up for in #742, @David-Baddeley do you mind if I increase the default get_file timeout? The directory listing timeout is 5 s

barentine, 2021-01-17:
Might be worth increasing the directory listing timeout as well though. Does 10 s seem reasonable?
```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/util/retry.py", line 368, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/packages/six.py", line 686, in reraise
    raise value
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 306, in _raise_timeout
    raise ReadTimeoutError(self, url, "Read timed out. (read timeout=%s)" % timeout_value)
urllib3.exceptions.ReadTimeoutError: HTTPConnectionPool(host='10.150.7.132', port=15348): Read timed out. (read timeout=5)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 309, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 437, in __call__
    bufferManager.bBuffer.calc_background(self.bgindices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 256, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 219, in update_buffer
    frame_data = np.ascontiguousarray(self.data_buffer.getSlice(frame), dtype=np.float32)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/buffers.py", line 28, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 71, in getSlice
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 792, in get_file
    locs = locate_file(filename, serverfilter, return_first_hit=True)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 365, in locate_file
    dirList, dt = _listSingleDir(dirurl)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 248, in _listSingleDir
    r = s.get(url, timeout=timeout)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 546, in get
    return self.request('GET', url, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='10.150.7.132', port=15348): Read timed out. (read timeout=5)

```

barentine, 2021-01-20:
as @David-Baddeley suggested in meeting yesterday: running with less worker processes allows things to run fine - the hand-ins were really hammering the dataservers, and as David mentioned and proposes a fix in #795 it looks like locking is a big sink of dataserver performance
[clusterIO.pdf](https://github.com/python-microscopy/python-microscopy/files/5843679/clusterIO.pdf)
[clusterListing.pdf](https://github.com/python-microscopy/python-microscopy/files/5843680/clusterListing.pdf)
[HTTPDataServer.pdf](https://github.com/python-microscopy/python-microscopy/files/5843681/HTTPDataServer.pdf)

barentine, 2021-01-30:
Updated, running with #795 and 2 workers per node in a 10 node cluster
[clusterListing.pdf](https://github.com/python-microscopy/python-microscopy/files/5898695/clusterListing.pdf)
[HTTPDataServer.pdf](https://github.com/python-microscopy/python-microscopy/files/5898696/HTTPDataServer.pdf)
[remFitBuf.pdf](https://github.com/python-microscopy/python-microscopy/files/5898697/remFitBuf.pdf)

David-Baddeley, 2021-01-31:
Interesting numbers here:

588,366 calls to HTTPDataServer `list_directory()`
201,572 calls to `cl.dir_cache.list_directory()`
16,712 calls to `os.listdir()`

This tells us that the `dir_cache` is doing it's job (mostly) with a 10:1 hit to miss ratio. The json cache in the dataserver is not as convincingly  - with a ~1:1 hit to miss ratio. This is probably not that important, however as the jsoning that the cache is trying to save is pretty fast.

Locking seems to be good enough (lock wait time is < time spent in lock).

Worry we might be running into the limits of profiler overhead once we get down into the _file_info stuff.

barentine, 2021-01-31:
Just noticed the following on one of the other dataservers while restarting them
[image]

barentine, 2021-02-11:
All set here, worth noting that with full N-cpu workers per node I can a) pull substantially more than the 50 tasks per worker want in the rulenodeserver (100 still is maybe less than half full load on the GPUs, I'm currently running with 400). With these 120 workers wanting 400 tasks, localizing series of 40k frames, I'm hitting an occasional frame or two fail on a socket error, which should be handled well by frame retries

```
===========================================================================
Error in rule: 000772-f36ca85653e14c4a8ab377ee432f5aef running task: 1732 on lahontan:9784

taskDescr:
----------
{'id': '000772-f36ca85653e14c4a8ab377ee432f5aef~1732', 'type': 'localization', 'taskdef': {'frameIndex': '1732', 'metadata': 'PYME-CLUSTER://villains/Bergamot/2021_2_8/EMC0104S4_2/E1/analysis/8_2_series_BUI.h5r.json'}, 'inputs': {'frames': 'pyme-cluster:///Bergamot/2021_2_8/EMC0104S4_2/E1/8_2_series_BUI.pcs'}, 'outputs': {'fitResults': 'http://10.150.7.4:15348/__aggregate_h5r/Bergamot/2021_2_8/EMC0104S4_2/E1/analysis/8_2_series_BUI.h5r/FitResults', 'driftResults': 'http://10.150.7.4:15348/__aggregate_h5r/Bergamot/2021_2_8/EMC0104S4_2/E1/analysis/8_2_series_BUI.h5r/DriftResults'}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 384, in _make_request
    six.raise_from(e, None)
  File "<string>", line 2, in raise_from
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 380, in _make_request
    httplib_response = conn.getresponse()
  File "/home/ubuntu/miniconda3/lib/python3.7/http/client.py", line 1336, in getresponse
    response.begin()
  File "/home/ubuntu/miniconda3/lib/python3.7/http/client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "/home/ubuntu/miniconda3/lib/python3.7/http/client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/home/ubuntu/miniconda3/lib/python3.7/socket.py", line 589, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/util/retry.py", line 368, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/packages/six.py", line 686, in reraise
    raise value
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/urllib3/connectionpool.py", line 306, in _raise_timeout
    raise ReadTimeoutError(self, url, "Read timed out. (read timeout=%s)" % timeout_value)
urllib3.exceptions.ReadTimeoutError: HTTPConnectionPool(host='10.150.7.133', port=15348): Read timed out. (read timeout=10)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 309, in computeLoop
    res = task()
  File "/home/ubuntu/code/python-microscopy/PYME/localization/remFitBuf.py", line 437, in __call__
    bufferManager.bBuffer.calc_background(self.bgindices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 256, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 219, in update_buffer
    frame_data = np.ascontiguousarray(self.data_buffer.getSlice(frame), dtype=np.float32)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/buffers.py", line 28, in getSlice
    sl = self.dataSource.getSlice(ind)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/DataSources/ClusterPZFDataSource.py", line 71, in getSlice
    sl = PZFFormat.loads(clusterIO.get_file(frameName, self.clusterfilter))[0]
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 796, in get_file
    locs = locate_file(filename, serverfilter, return_first_hit=True)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 349, in locate_file
    dirList, dt = _listSingleDir(dirurl)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/clusterIO.py", line 248, in _listSingleDir
    r = s.get(url, timeout=timeout)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 546, in get
    return self.request('GET', url, **kwargs)
[... 59 lines cut ...]
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 256, in calc_background
    self.update_buffer(bg_indices)
  File "/home/ubuntu/code/pyme-warp-drive/warpdrive/buffers.py", line 219, in update_buffer
    frame_data = np.ascontiguousarray(self.data_buffer.getSlice(frame), dtype=np.float32)
  File "/home/ubuntu/code/python-microscopy/PYME/IO/buffers.py", line 28, in getSlice
    sl = self.dataSource.getSlice(ind)
[... cut ...]
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.7/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='10.150.7.133', port=15348): Read timed out. (read timeout=10)

```

closed 2021-02-11 by barentine