python-microscopy/python-microscopy #746: [BUG] clusterUI downloading pcs series as h5 fails

barentine, 2021-01-04:
**Describe the bug**
```
Traceback:
File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)
File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)
File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  187.                 response = self.process_exception_by_middleware(e, request)
File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  185.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)
File "/home/smeagol/code/python-microscopy/PYME/cluster/clusterUI/clusterbrowser/views.py" in file
  35.             img.Save(outf.name)
File "/home/smeagol/code/python-microscopy/PYME/IO/image.py" in Save
  1138.                 openImages.pop(ofn)
File "/home/smeagol/miniconda3/lib/python3.6/weakref.py" in pop
  271.                 raise KeyError(key)
Exception Type: KeyError at /files/download/Bergamot/2020_12_9/GB_STORM_Slide2/try2/000/A1.pcs/
Exception Value: 'pyme-cluster://villains/Bergamot/2020_12_9/GB_STORM_Slide2/try2/000/A1.pcs'
```

**To Reproduce**
Steps to reproduce the behavior:
1. try to download a pcs series (pzf spooled) from the cluster via clusterUI
4. see error

**Expected behavior**
download the pcs as an h5

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10 computer talking to ubuntu 14.04/18.04 cluster
 - Python version: 3.6
 - numpy version: 1.16 / 1.19
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
filing so I remember to look into it tomorrow

David-Baddeley, 2021-01-12:
This is odd ... looks to be the renaming logic on saving files that is getting into problems (not really relevant for the cluster case). In the worst case we could bypass this, but still confusing as to why it's causing problems.

adrianhollow, 2021-03-15:
Tried to reproduce this with `ClusterOfOne`, but instead hit #956 . With the change proposed in #957 , #956 does not occur and the download works just fine.
Maybe the bug is located in a gap between real clusters as in #918 and `ClusterOfOne`, but just wanted to give that note here.

David-Baddeley, 2021-04-18:
Suspect we should be able to close this with #943

closed 2021-04-18 by David-Baddeley