python-microscopy/python-microscopy #750: [BUG] TooManyFieldsSent: possible to queue too many recipe tasks at once through clusterUI

barentine, 2021-01-06:
**Describe the bug**
using the templated recipe view through clusterui_tornado, I tried to queue a recipe for 4757 files at once
```
Environment:

Request Method: POST
Request URL: http://10.150.7.130:8889/recipes/run_template/

Django Version: 1.11.10
Python Version: 3.6.9
Installed Applications:
['django.contrib.admin',
 'django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.messages',
 'django.contrib.staticfiles',
 'clusterbrowser',
 'clusterstatus',
 'localization',
 'recipes',
 'accounts']
Installed Middleware:
['django.middleware.security.SecurityMiddleware',
 'django.contrib.sessions.middleware.SessionMiddleware',
 'django.middleware.common.CommonMiddleware',
 'django.middleware.csrf.CsrfViewMiddleware',
 'django.contrib.auth.middleware.AuthenticationMiddleware',
 'django.contrib.auth.middleware.SessionAuthenticationMiddleware',
 'django.contrib.messages.middleware.MessageMiddleware',
 'django.middleware.clickjacking.XFrameOptionsMiddleware']

Traceback:

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/exception.py" in inner
  41.             response = get_response(request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _legacy_get_response
  249.             response = self._get_response(request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  178.             response = middleware_method(request, callback, callback_args, callback_kwargs)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/middleware/csrf.py" in process_view
  300.                     request_csrf_token = request.POST.get('csrfmiddlewaretoken', '')

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/wsgi.py" in _get_post
  126.             self._load_post_and_files()

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/http/request.py" in _load_post_and_files
  311.             self._post, self._files = QueryDict(self.body, encoding=self._encoding), MultiValueDict()

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/http/request.py" in __init__
  394.             for key, value in limited_parse_qsl(query_string, **parse_qsl_kwargs):

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/utils/http.py" in limited_parse_qsl
  442.                 'The number of GET/POST parameters exceeded '

Exception Type: TooManyFieldsSent at /recipes/run_template/
Exception Value: The number of GET/POST parameters exceeded settings.DATA_UPLOAD_MAX_NUMBER_FIELDS.

```

barentine, 2021-01-07:
Note: 588 is alright, but 1009 is too many

barentine, 2021-01-07:
Looks like we can configure it to either a larger value or disable it completely if we want https://stackoverflow.com/questions/47585583/the-number-of-get-post-parameters-exceeded-settings-data-upload-max-number-field

David-Baddeley, 2021-01-10:
Suggest we set it to `None` to disable. Could potentially also consider changing things so that file names are passed as json in the body, but would require a bit of js wrangling.

closed 2021-01-11 by barentine