python-microscopy/python-microscopy #759: [BUG] pyme-cluster:// scheme for output modules fails if the directory doesn't exist on the local node

barentine, 2021-01-07:
**Describe the bug**
A clear and concise description of what the bug is.
[image]
[image]
[image]

for the following recipe
```
- localisations.AddPipelineDerivedVars:
    inputEvents: ''
    inputFitResults: FitResults
    outputLocalizations: input
- tablefilters.FilterTable:
    filters:
      LLH:
      - -250.0
      - 0.0
      error_x:
      - 0.0
      - 20.0
      error_y:
      - 0.0
      - 20.0
    inputName: input
    outputName: filtered
- multiview.Fold:
    input_name: filtered
    output_name: folded
- multiview.CalibrateShifts:
    input_name: folded
    output_name: shiftmap
    search_radius_nm: 500.0
- output.HDFOutput:
    filePattern: '{output_dir}/{file_stub}.sf'
    inputVariables:
      shiftmap: shift_map
    scheme: pyme-cluster://

```

barentine, 2021-01-07:
@David-Baddeley wouldn't be tough to check clusterIO dir exists for pyme-cluster:// scheme, if we wanted to preserve the default crash on trying to write a table if the directory tree doesn't already exist. Otherwise we could just make it. Any preference? The first seems more safe

David-Baddeley, 2021-01-10:
I think we should just create it if it doesn't exist - that's the default cluster logic.

closed 2021-01-11 by barentine