python-microscopy/python-microscopy #761: [ENHANCEMENT] Maintain layer display settings when loading a new dataset

lukasfue, 2021-01-08:
Visgui used to keep the layer settings set for one dataset even after loading a new dataset. I think this was considered a bug and was changed in a recent update so that the layers resets when the new dataset is loaded. However, I thought this was nice as I could compare different datasets under the same visualization setting without having to go through it all each time. If this isn't desirable for others, maybe there could be an option to keep the settings or not when loading new datasets.

David-Baddeley, 2021-01-10:
As a default it gets super-messy if the data is not substantially the same as the previous data and somewhat messy even if it is for multi-colour data. 

I think we have support for doing what you want in the backend, though - just hasn't made it into the GUI yet. It's probably best that we keep the full pipeline as well as the display settings, however, as it's quite possible that layers will depend on stuff which has been generated in the pipeline.

Proposed solution would be to add, e.g., an `Open with current pipeline` menu item.

lukasfue, 2021-01-11:
I think that sounds like a good solution. It would give users the option at least and avoids the potential for serious bugs that could arise by forcing it to be that way every time.

zacsimile, 2021-03-16:
@David-Baddeley what is the best backend for this? Do we want to try to store "most recent settings" in the user `config.yaml` and pull from those?