python-microscopy/python-microscopy #762: [ENHANCEMENT] Ability to set layer display defaults in visgui

lukasfue, 2021-01-08:
When I visualize data in visgui, I pretty much always use the same LUTs, point sizes, alphas , pointsprites, etc. It would be really nice to be able to choose my own default visualization settings (instead of having it always be points, gist_rainbow, 30 nm points,  colored in time, etc). Additionally, it would be even better to have the option to do this for multiple layers. So for example, when I first load up my data, it will display a layer colored in time with pointsprites, 10 nm points, 0.1 alpha, and Jet LUT and another layer colored in Z with pointsprites, 10 nm points, 0.1 alpha, and gist_rainbow LUT. Then I can easily toggle on/off which layer I want to see without having to spend time setting each of them up.

David-Baddeley, 2021-01-10:
Can't recall the precise issue (or whether this was an offline discussion), but something similar is already on the TODO list. Somewhat alluded to in #700. Basic idea is to open with most recently used settings, starting with colour maps and expanding to other parameters once we have some idea of what gets used.

lukasfue, 2021-01-11:
The ideas in #700 sound nice as well, but seem different to me than what I'm thinking here. In this case, I'm proposing that one could save how they want their data to be displayed initially. So each time one loads a different dataset it is displayed that way from the initial load without having to change all the settings. Its more of a quality of life enhancement than anything else I suppose.

closed 2021-12-14 by David-Baddeley