python-microscopy/python-microscopy #765: [BUG] PYME.LMVis.gl_render3D_shaders.LMGLShaderCanvas.getIm() returns bad snapshot for meshes

zacsimile, 2021-01-09:
**Screenshot on my mac**

[image]

** getIm() result of the same view **

[image]

David-Baddeley, 2021-01-11:
I suspect that depth checking is not working as we don't seem to attach a depth-buffer for the offscreen rendering. By the look of things, we also don't do anti-aliasing (which will make the images more jagged). Will see if it's easy to fix ... 

See also:
https://stackoverflow.com/questions/5751081/how-to-use-depth-testing-when-rendering-to-an-offscreen-buffer-then-onto-texture

David-Baddeley, 2021-01-11:
Immediate issue fixed in #771

[image]

Images are still a little more pixilated due to lack of multi-sampling / antialiasing.

David-Baddeley, 2021-01-11:
[image]

now with antialising :)

zacsimile, 2021-01-11:
Very cool! Thank you, @David-Baddeley!

David-Baddeley, 2021-01-11:
Transparency also works (I was a rather worried that it wouldn't due to the two-pass rendering approach).

closed 2021-01-12 by David-Baddeley