python-microscopy/python-microscopy #766: [BUG] Focus mapping in PYMEVis confused by piezo offset update event

barentine, 2021-01-09:
**Describe the bug**
I haven't chased down exactly what's going on in the mapping, but I've messed it up somewhere. Might need to add an event to log the focus at the start of the series (or rather, when the focus lock is disabled)
[image]
Zoom in on PYMEVis events at the start of the series
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. run automated imaging using the offset piezo REST client/server and reflection based focus lock client/server

**Expected behavior**
don't get the wrong focus for the first z-dwell frames

barentine, 2021-01-09:
events:
```
image.events[np.argsort(image.events[:]['Time'])]
array([(b'ProtocolFocus', 1.60998114e+09, b'0, 49.677'),
       (b'PiezoOnTarget', 1.60998114e+09, b'27.051'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.671'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.596'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.559'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.557'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.568'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.576'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.595'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.593'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.601'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.635'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.652'),
       (b'StartAq', 1.60998114e+09, b'0'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.670'),
       (b'PiezoOnTarget', 1.60998114e+09, b'26.637'),
       (b'ProtocolTask', 1.60998114e+09, b"-1, update, {'Lasers.MPB560.On': False, 'Lasers.MPB560.Power': 600.0, 'Lasers.MPB642.On': True, 'Lasers.MPB642.Power': 600.0, 'Multiview.ActiveViews': [0, 1, 2, 3], 'Multiview.ROISize': [256, 256], 'Camera.IntegrationTime': 0.00126}"),
       (b'ProtocolTask', 1.60998114e+09, b'-1, DisableLockAfterAcquiringIfEnabled, '),
       (b'PiezoOffsetUpdate', 1.60998114e+09, b'-23.0440'),
       (b'ProtocolFocus', 1.60998114e+09, b'1001, 50.677'),
       (b'PiezoOnTarget', 1.60998114e+09, b'27.607'),
       (b'ProtocolFocus', 1.60998114e+09, b'2001, 51.677'),
       (b'PiezoOnTarget', 1.60998114e+09, b'28.618'),
       (b'ProtocolFocus', 1.60998115e+09, b'3001, 52.677'),
       (b'PiezoOnTarget', 1.60998115e+09, b'29.600'),
       (b'ProtocolFocus', 1.60998115e+09, b'4001, 53.177'),
       (b'PiezoOnTarget', 1.60998115e+09, b'30.120'),
       (b'ProtocolFocus', 1.60998115e+09, b'5001, 52.177'),
       (b'PiezoOnTarget', 1.60998115e+09, b'29.120'),
       (b'ProtocolFocus', 1.60998115e+09, b'6001, 51.177'),
       (b'PiezoOnTarget', 1.60998115e+09, b'28.098'),
       (b'ProtocolFocus', 1.60998115e+09, b'7001, 50.177'),
       (b'PiezoOnTarget', 1.60998115e+09, b'27.099'),
       (b'ProtocolTask', 1.60998115e+09, b'8000, TurnOn, '),
       (b'ProtocolFocus', 1.60998115e+09, b'8001, 49.177'),
       (b'PiezoOnTarget', 1.60998115e+09, b'26.119'),
       (b'ProtocolFocus', 1.60998115e+09, b'9001, 49.677'),
       (b'PiezoOnTarget', 1.60998115e+09, b'26.598'),
       (b'ProtocolFocus', 1.60998115e+09, b'10001, 50.677'),
       (b'PiezoOnTarget', 1.60998115e+09, b'27.600'),
       (b'ProtocolFocus', 1.60998116e+09, b'11001, 51.677'),
       (b'PiezoOnTarget', 1.60998116e+09, b'28.620'),
       (b'ProtocolFocus', 1.60998116e+09, b'12001, 52.677'),
       (b'PiezoOnTarget', 1.60998116e+09, b'29.605'),
       (b'ProtocolFocus', 1.60998116e+09, b'13001, 53.177'),
       (b'PiezoOnTarget', 1.60998116e+09, b'30.100'),
       (b'ProtocolFocus', 1.60998116e+09, b'14001, 52.177'),
       (b'PiezoOnTarget', 1.60998116e+09, b'29.120'),
       (b'ProtocolFocus', 1.60998116e+09, b'15001, 51.177'),
       (b'PiezoOnTarget', 1.60998116e+09, b'28.102'),
       (b'ProtocolFocus', 1.60998116e+09, b'16001, 50.177'),
       (b'PiezoOnTarget', 1.60998116e+09, b'27.108'),
       (b'ProtocolFocus', 1.60998116e+09, b'17001, 49.177'),
       (b'PiezoOnTarget', 1.60998116e+09, b'26.120'),
       (b'ProtocolFocus', 1.60998116e+09, b'18001, 49.677'),
       (b'PiezoOnTarget', 1.60998116e+09, b'26.619'),
       (b'ProtocolFocus', 1.60998117e+09, b'19001, 50.677'),
       (b'PiezoOnTarget', 1.60998117e+09, b'27.620'),
       (b'PiezoOnTarget', 1.60998117e+09, b'26.113'),
       (b'ProtocolTask', 1.60998117e+09, b'EnableLock, ')],
      dtype=[('EventName', 'S32'), ('Time', '<f8'), ('EventDescr', 'S256')])
```

observations:
- setting the initial focus position happens before the focus lock is shut off. If the focus lock position is not the same as this zero position... it will be (the offset will just be adjusted until it is).
- as things stand, it would be OK to account for the updated offset and use the frame 0 ProtocolFocus event to map z for every frame after the PiezoOffsetUpdate event, and mark everything between that frame 0 and the PiezoOffsetUpdate as piezo moving.
- if we want to allow the focus lock position to be different than the 0th z position in a stack we need to change something. One potential would be to move the 0 frame ProtocolFocus to after the -1 frame events. Currently this happens in the `Init` method of the `ZStackTaskListProtocol`, which gets called before the `TaskListProtocol.Init` where the frame -1 tasks are actually done. 
- it does strike me as weird that we do a frame 0 task before all the -1 frame tasks. Any thoughts on how to resolve this @David-Baddeley? I think it would make more sense to append the frame 0 focus position to the task list for frame zero and let it get carried out by OnFrame rather than explicitly in `ZStackTaskListProtocol.Init`. This would be a change over the current behavior, but one that would make for a more consistent framework. The current behavior could also be achieved by creating a flag which lets people push the 0 frame ProtocolFocus as the first frame -1 task rather than just a frame zero task as I am proposing

David-Baddeley, 2021-01-10:
The initial z move being given a `frameNum` of 0 rather than -1 (it definitely has all the features of a -1 task) is a little confusing, but shouldn't be of any practical consequence for how things are calculated as this only relies on timestamps. `StartAq` events are the only ones for which the value of `frameNum` has any practical consequence. We should probably change it to -1 for consistency though.

Ultimately the logic error that gives rise to your observed problem is that `offset_map` (line 132 in piezo_movement_correction.py) is initialised with a bogus value (0) rather than the value of the offset at the start of the acquisition. In general, however, having un-offsetted  positions in the events is not at all desirable and prone to generating errors. The analysis code should not need to know the details of focus offsetting and I'd be in favour of deprecating this completely. 

A short-term, backwards compatible, workaround would be to have `PiezoOffsetUpdate` events also spoof `ProtocolFocus` events.

barentine, 2021-01-10:
Hi @David-Baddeley - I think the frame 0 rather than -1 is more than a confusing vs not issue. If I have a -1 event to turn off my focus lock, it currently gets executed _after_ the first MoveTo, which for an offset piezo w/ focus lock will change the offset rather than the position (since it's still locked). I think the solution which allows the most flexibility / practical use of protocols would be to actually first the first MoveTo on frame 0, and if someone isn't using PiezoOnTarget calls they should at least move data starts at to X frames or so.

David-Baddeley, 2021-01-10:
The reasons I don't think that changing where we do the z focus is a good idea / would fix things are as follows:

- it's reasonable to expect that we are at the starting position on the first frame
- you currently turn the focus lock off on frame -1, but a protocol could equally do it on, e.g. frame 50, or 100, so there is no way of choosing a safe time to do it other than before the sequence starts.
- `MoveTo` with focus lock on *should* change the **target** position of the lock (will implicitly also change offset, but that's fine). IE z-stepping should also work with focus lock on. This is how things work with the other focus locks and is important for slower imaging, but might require modification to the reflection focus lock.  

Something I'd consider as a work-around to get your case working fast is to make it a `frameNum=-1` task, but have it execute after all the other -1 priority tasks (i.e. after `TaskListProtocol.OnFrame(self, frameNum)` on line 234 of protocol.py).

David-Baddeley, 2021-01-10:
Will try and put this in a PR (as I have stuff open already)

barentine, 2021-01-10:
Fair point about being able to choose an arbitrary frame to shut off the focus lock. I'm not opposed to making the reflection focus lock and offset piezo less agnostic of each other, but appreciate the quick workaround in #767 (and think it's not super unreasonable long term either)

David-Baddeley, 2021-01-28:
**Current status:** event order still off due to round-trip latency in reflection focus lock implementation.

barentine, 2021-02-16:
event order addressed in #839

closed 2021-02-16 by barentine