python-microscopy/python-microscopy #775: [BUG] events handling issue with multiview extract channel?

barentine, 2021-01-12:
[image]
**Describe the bug**
Not immediately sure how this happens since the crop datasource should pass through getEvents

```
- rcnn_segment.RCNNDetect:
    device: /gpu:0
    input_name: supertile
    intensity_factor: 0.1
    model_dir: /home/ubuntu/keras_models
    output_mask: masks
    output_rois: detected
- supertile.TilePhysicalCoords:
    input_measurements: detected
    input_supertile: supertile
    measurement_units: um
    output_name: physical_coords
- output.HDFOutput:
    filePattern: '{output_dir}/{file_stub}_detections.hdf'
    inputVariables:
      filtered: filtered
      overlap_identified: og_detected
      subset: roi_locations
    scheme: pyme-cluster://
- measurement.IdentifyOverlappingROIs:
    input: physical_coords
    output: overlap_identified
- tablefilters.FilterTable:
    filters:
      rejected:
      - -0.5
      - 0.5
      x_width:
      - 0.0
      - 30.0
      y_width:
      - 0.0
      - 30.0
    inputName: overlap_identified
    outputName: filtered
- acquisition.QueueAcquisitions:
    action_server_url: http://10.150.7.131:9393
    input_positions: subset
    max_duration: 200.0
    spool_settings:
      maxFrames: 30000
      protocol: htsms-flow.py
      stack: true
      subdirectory: '{file_stub}'
- supertile.Supertile:
    input_name: extracted
    level: 2
    output_name: supertile
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_detections.h5'
    inputName: masks
    scheme: pyme-cluster://
- tablefilters.RandomSubset:
    input: filtered
    num_to_select: 30
    output: subset
- multiview.ExtractMultiviewChannel:
    input_name: input
    output_name: extracted
    view_number: 1

```

**To Reproduce**
Steps to reproduce the behavior:
1. run the above recipe on a tile pcs

**Expected behavior**
same recipe sans the extract multiview channel module, on a single view tile pcs, works fine

David-Baddeley, 2021-01-12:
Not much, but error message implies that `events` is a list, rather than a `recarray`. This could be due to loading from json.

David-Baddeley, 2021-01-12:
Doesn't explain why the same recipe works without the multiview module though.

barentine, 2021-01-12:
huh, kind of wild but I'm running a multiwell experiment at the moment and it's failed on _some_ wells
```
===========================================================================
Error in rule: 000216-ff32d4d6d8964422821aac7b458e3cee running task: 3 on drno:13243
taskDescr:
----------
{'id': '000216-ff32d4d6d8964422821aac7b458e3cee~3', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2021_1_12/Hess1/try2/002/B4.pcs'}, 'output_dir': 'pyme-cluster://villains/Bergamot/2021_1_12/Hess1/try2/002/', 'taskdef': {'recipe': "- rcnn_segment.RCNNDetect:\r\n    device: /gpu:0\r\n    input_name: supertile\r\n    intensity_factor: 0.1\r\n    model_dir: /home/ubuntu/keras_models\r\n    output_mask: masks\r\n    output_rois: detected\r\n- supertile.TilePhysicalCoords:\r\n    input_measurements: detected\r\n    input_supertile: supertile\r\n    measurement_units: um\r\n    output_name: physical_coords\r\n- output.HDFOutput:\r\n    filePattern: '{output_dir}/{file_stub}_detections.hdf'\r\n    inputVariables:\r\n      filtered: filtered\r\n      overlap_identified: og_detected\r\n      subset: roi_locations\r\n    scheme: pyme-cluster://\r\n- measurement.IdentifyOverlappingROIs:\r\n    input: physical_coords\r\n    output: overlap_identified\r\n- tablefilters.FilterTable:\r\n    filters:\r\n      rejected:\r\n      - -0.5\r\n      - 0.5\r\n      x_width:\r\n      - 0.0\r\n      - 30.0\r\n      y_width:\r\n      - 0.0\r\n      - 30.0\r\n    inputName: overlap_identified\r\n    outputName: filtered\r\n- acquisition.QueueAcquisitions:\r\n    action_server_url: http://10.150.7.131:9393\r\n    input_positions: subset\r\n    max_duration: 400.0\r\n    spool_settings:\r\n      maxFrames: 45000\r\n      protocol: htsms-flow.py\r\n      stack: true\r\n      subdirectory: '{file_stub}'\r\n- supertile.Supertile:\r\n    input_name: input\r\n    level: 2\r\n    output_name: supertile\r\n- output.ImageOutput:\r\n    filePattern: '{output_dir}/{file_stub}_detections.h5'\r\n    inputName: masks\r\n    scheme: pyme-cluster://\r\n- tablefilters.RandomSubset:\r\n    input: filtered\r\n    num_to_select: 20\r\n    output: subset\r\n"}}
Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 340, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 685, in execute
    m.execute(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/supertile.py", line 120, in execute
    x, y = tile_pyramid.get_position_from_events(stack.events, stack.mdh)
  File "/home/ubuntu/code/python-microscopy/PYME/Analysis/tile_pyramid.py", line 559, in get_position_from_events
    xm = piecewiseMapping.GeneratePMFromEventList(events, mdh, mdh['StartTime'], x0, b'ScannerXPos', 0)
  File "/home/ubuntu/code/python-microscopy/PYME/Analysis/piecewiseMapping.py", line 210, in GeneratePMFromEventList
    for e in events[events['EventName'] == eventName]:
TypeError: list indices must be integers or slices, not str
```

barentine, 2021-01-12:
note that this is without the extract channel bit

closed 2021-01-13 by barentine