python-microscopy/python-microscopy #784: [BUG] DH5view won't save raw data to TIFs past frame 13416

lukasfue, 2021-01-14:
**Describe the bug**
I have loaded a couple dcimg files with 40-50k frames into DH5View and attempted to save as a TIF. DH5View does save a TIF file but it only ever saves the first 13416 frames. It does give a traceback.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch DH5View
2. Load data (I've only tested data with more than 40k frames)
3. Save as TIF
4. See saved TIF is only 13416 frames
5. See following Traceback:
`Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\DSView\dsviewer.py", line 323, in OnSave
    self.image.Save()
  File "c:\users\laf62\python-microscopy\PYME\IO\image.py", line 1132, in Save
    self.filename = dataExporter.ExportData(self.data, self.mdh, self.events, filename=filename, progressCallback=progressCallback)
  File "c:\users\laf62\python-microscopy\PYME\IO\dataExporter.py", line 588, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "c:\users\laf62\python-microscopy\PYME\IO\dataExporter.py", line 251, in Export
    tifffile.imsave_f(outFile, dw, description = description)
  File "c:\users\laf62\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 636, in imsave_f
    write(offset_format, pos)
  File "c:\users\laf62\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 620, in write
    fd.write(pack(arg, *args) if args else arg)
  File "c:\users\laf62\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 541, in pack
    return struct.pack(byteorder+fmt, *val)
struct.error: argument out of range`

**Expected behavior**
A saved TIF that includes all frames

David-Baddeley, 2021-01-14:
TIFF has a hard limit at 2GB  (or potentially 4GB, I need to check) file size, which is probably what you are running into here. It would be helpful if we gave a more informative error message, however. To verify whether this is the issue, it would be good to know what your frame size was (pixels). We should also add a check on this to the exporter.

There is a non-standard  BigTIFF extension to the TIFF format, which lets you go past this but that's not well supported and/or portable, and I'd be reluctant to add support unless it was absolutely critical. TIFF is not really a suitable format for storing such large files (the structure means that IO will always be pretty slow if you have a large number of frames). If the goal is to get it into ImageJ, there are ImageJ extensions which let you read, e.g. HDF5 which does have good large file support. 

That said, when we get around to using an updated version of tifffile we might well get BigTIFF support for free (not 100% sure on this).

A workaround might be to break the series up into ~2GB chunks using the crop exporter (a bit tedious, but would be a way of getting stuff saved/loaded).

David-Baddeley, 2021-01-14:
Just double checked the TIFF max size - the answer is 4GB in theory, but with a lot of implementations only supporting 2GB in practice (i.e. if you save a 4GB tiff, you might not be able to open it in all software).

lukasfue, 2021-01-18:
I see, thank you for the info. It would be nice to not have to use TIFFs for what I'm doing but unfortunately it appears to be the only way. Thankfully, I have found a workaround. Thanks again.

David-Baddeley, 2021-01-28:
Closing, as we now inform users about TIFF limitations rather than giving a cryptic error message. Opened a new enhancement/ discussion issue on BigTIFF just to keep it on the radar.

closed 2021-01-28 by David-Baddeley

lukasfue, 2022-02-17:
Hi @David-Baddeley, this has become relevant again for me. Would it be possible to implement the solution of breaking up the TIFF into multiple files? I think that is probably the most straightforward solution, at least from my perspective. 

Also, in regards to your last comment, the error message is still not very clear. It just says "Error whilst running File>&Save As" which seems evident without that error message. This part isn't super important, but I just wanted to point it out.

Here's the error message I get currently.

Error whilst running File>&Save As
==============================================
python-microscopy=22.01.12.post0.dev[git]7f16e87989d8
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\code\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\code\python-microscopy\PYME\DSView\dsviewer.py", line 312, in OnSave
    self.image.Save()
  File "c:\users\laf62\code\python-microscopy\PYME\IO\image.py", line 1335, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, filename=filename, progressCallback=progressCallback)
  File "c:\users\laf62\code\python-microscopy\PYME\IO\dataExporter.py", line 504, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "c:\users\laf62\code\python-microscopy\PYME\IO\dataExporter.py", line 316, in Export
    tifffile.imsave_f(outFile, data, description = description)
  File "c:\users\laf62\code\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 636, in imsave_f
    write(offset_format, pos)
  File "c:\users\laf62\code\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 620, in write
    fd.write(pack(arg, *args) if args else arg)
  File "c:\users\laf62\code\python-microscopy\PYME\contrib\gohlke\tifffile.py", line 541, in pack
    return struct.pack(byteorder+fmt, *val)
struct.error: argument out of range