python-microscopy/python-microscopy #789: [ENHANCEMENT] allow ridiculous numbers of localization tasks to be queued

barentine, 2021-01-17:
**Is your feature request related to a problem? Please describe.**
Should you manually queue analysis for thousands of series rather than handling them as they come in, you can kill the ruleserver ungracefully when it runs out of RAM for task status arrays

<details>  <summary> Traceback </summary>

```
DEBUG:ruleserver:release_rule_tasks(ruleID = 003269-bb1d660964a94e4ba242ddf291369e50, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003271-c651e1e81a674eb5969361284090d9ee, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003272-6fa95b8bd1e847a68d365a39c1b92c31, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003273-dfbd923b40584294a3cfecca9246c913, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003274-097ca3435fb646e183eec724569e7a9e, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003275-fd9fe4422cb74957813238e2186a97e5, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003276-3800b32f396148d6bc1054d80fc5cd4e, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003277-3910c7cfbe4849c1a2ddf303001a8acf, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003278-ea228e19ae5c4e529f32edcaa671e240, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003279-f89dd659f65f4faf8369ec2b50107919, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003280-d3bd358ec15d465fb1d640a1618dfe1a, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003281-8e71878c986d416194f5ea78b5256d58, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003282-04bd184202944b07acdff74b12f98a6a, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003283-212236aaf60b4ff1a6685b73aea323e7, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003285-9b0838f1283a47ad9c80576a30dd17e2, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003284-d6835bec5ea44dc4b89739e6e615c2bc, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003286-4fe2273f16f54974915690fbfdac7502, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003287-18d9608313d24e4cb8d121cf094087c5, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003288-5bf513a044b84af4b576e0a4cc7aea33, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003289-99460f4f86d346239992223f3ee48beb, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003290-00417b91b9a24fafb420c2f6cafe4f7c, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003291-943d357d8d4f4a059b3830848e9d26fd, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003292-e2d5b9792de84cba8c1fd7043b2856df, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003293-4c75cdd0dac74205bab5911424ca0464, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003295-7498d0352ccf402e928e47b229c24628, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003294-75bc4a8e5d3347fe95b37c7484f7307a, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003296-c6c9549e6dec4ee4bec5cb8e528f0232, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003297-5bd9e3c4db614470906ea75b6dd4400a, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003298-bcf8c0ea6335403e9b9c494344c8bfa5, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003300-c0f5c2056c5d400dba1f6982ec8d78cb, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003299-9927559246614c5e98e4f341356d1a38, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003301-18c08511f743493898c203589d604dd2, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003302-ea3a86c1d0554cc8b15bbf4813132ab8, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003303-af8ca8a64503470884777e687857d9b2, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003304-76b61f448b76488ba322502736c328af, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003305-7ff7b36a5ed341ceaf9aaa482b68932a, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003306-b8d02d48007342f6b117cbdad3d680fe, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003308-7a96cb3b9c0b466894c008f4521ac7b6, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003307-f81563a7db69473498a46a36e0ebd332, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003309-5c491d33fcfb4ad68b9a2b575401d99d, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003310-678ab8fd3e01450290dc4d098dae83ff, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003311-1fe596e966cb4fdb9ebabd94aca608d5, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003312-b2ba9478f7634aa5b919b6a84d07d15c, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003313-a6ea6f4721ca4749a9145dac2fd8e6b4, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003314-7e55915dd1464dca9e47e55638d40e96, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003315-776d988f778e4f95b0bd6641ed0bcfe2, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003316-8c5fd0412c8a4be88a29872bbab85ae1, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003317-cab076eb8cc24b00825b87337a0068b8, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003318-5cef08ae0770428c9364e9fd79d6eaa3, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003319-020c0ccbba334156a9a21940a275b939, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003320-c163c96f947e45c98ab3e8c61994576c, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003321-fb6b3139627b4753adc23272ed0859d7, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003322-b6da592324f048e9bada9d1cbd4306bb, release_start=0, release_end=20000)
DEBUG:ruleserver:release_rule_tasks(ruleID = 003323-d9aa29a7479b446db1ec80d621b0fe2e, release_start=0, release_end=17800)
10.150.7.130 - - [17/Jan/2021 17:51:40] code 500, message Server Error
 Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request
    resp = handler(**kwargs)
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/ruleserver.py", line 706, in add_integer_id_rule
    inputs_by_task=rule_info.get('inputsByTask', None), on_completion=rule_info.get('on_completion', None))
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/ruleserver.py", line 171, in __init__
    self._task_info = np.zeros(max_task_ID, self.TASK_INFO_DTYPE)
MemoryError: Unable to allocate 9.54 MiB for an array with shape (1000000,) and data type [('status', 'u1'), ('nRetries', 'u1'), ('expiry', '<f4'), ('cost', '<f4')]

[... 517 lines cut ...]
  File "/home/ubuntu/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request
    resp = handler(**kwargs)
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/ruleserver.py", line 836, in mark_release_complete
    self._rules[ruleID].mark_release_complete(n_tasks)
KeyError: 'None'

10.150.7.130 - - [17/Jan/2021 17:51:54] code 500, message Server Error
 Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/util/webframework.py", line 285, in _process_request
    resp = handler(**kwargs)
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/ruleserver.py", line 747, in release_rule_tasks
    rule = self._rules[ruleID]
KeyError: 'None'

Killed

```

</details>

**Describe the solution you'd like**
maybe keep a queue of N active rules, and store anything beyond that in the str template format until there is room in the queue to create the array.

**Describe alternatives you've considered**
not launching 10**8 frames 'at once'

barentine, 2021-01-18:
worth noting this was pushing using cluster.rules rather than `HTTPRulePusher`. If a series is already complete when we got to push it we'll want to specify the `max_task_ID`. In this case, pushing 20k frame series, we would have used 20x less RAM on the status arrays

David-Baddeley, 2021-01-18:
Not an entirely trivial change
- storing rules in a compressed format would require putting the rule pushing logic on the ruleserver (not completely out of the question, but also not trivial, and potentially with performance implications). 
- one option might be to make pushing block until memory was available
- another option would be to fail (and return the failure to the pusher) if no memory was available
- 3rd option might be to use disk storage for rules on the ruleserver

barentine, 2021-01-19:
I'm going to close the issue as allocating known length when we have it is good enough for now and I think as you allude to David the alternatives will take some stronger motivation I suspect. Thanks!

closed 2021-01-19 by barentine