python-microscopy/python-microscopy #794: [PROPOSAL] add endpoint to actionserver to allow multiple acquisition events to queue at once

barentine, 2021-01-19:
**Is your feature request related to a problem? Please describe.**
We currently have a race condition, and if you have multiple recipes running which will queue acquisition events, you might get them interweaved, which can result in e.g. spooling a series to the wrong directory, etc. etc.
[image]

**Describe the solution you'd like to implement**
add an endpoint for queueing multiple actions, and a lock to the action server so you can queue actions of the same nice in the desired order

David-Baddeley, 2021-01-19:
I'd been thinking about this problem in a slightly different context, specifically what happens when you are actually using `nice` values to, e.g., insert a high priority task into the queue. Such a task could easily execute between a `state.update` and a `startSpooling`. Adding a new queueing endpoint and lock wouldn't solve this variant of the problem.

I think the right approach is probably going to be to reformulate the tasks so that the combination of state update and action appear as one task ... i.e. something like:

```
with scope.state=...:
    spool_series()
```

barentine, 2021-01-19:
That would be interesting. I've found it useful/necessary to put the starting state in the front of all my protocols. Would this replace that, or would we want both? I could see this state definition combined with your suggestion in #738  replacing preflights.

In any event, I don't think it would be the end of the world to queue multiple tasks at once rather than throttling a for loop so we don't lock the GUI (as I currently do). Thoughts on that in the mean time David?

barentine, 2021-01-19:
I could also see being even more flexible with making the action queue task format be a list of chained actions. The typical one would be to update position followed by an acquisition, but could be nice if you wanted to e.g.
`[state.update({...}), spoolController.StartSpooling(protocol='dic-stack'), spoolController.StartSpooling(protocol='smlm')]`,

closed 2021-02-06 by barentine