python-microscopy/python-microscopy #803: [BUG] django version ratcheting is incomplete

barentine, 2021-01-25:
**Describe the bug**
django 2.1.7:
```
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner
    response = get_response(request)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = self.process_exception_by_middleware(e, request)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py", line 124, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/cluster/clusterUI/localization/views.py", line 104, in settings
    'categorized_fields' : fields_by_category})
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/shortcuts.py", line 36, in render
    content = loader.render_to_string(template_name, context, request, using=using)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/loader.py", line 62, in render_to_string
    return template.render(context, request)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/backends/django.py", line 61, in render
    return self.template.render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 171, in render
    return self._render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 163, in _render
    return self.nodelist.render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 937, in render
    bit = node.render_annotated(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/defaulttags.py", line 209, in render
    nodelist.append(node.render_annotated(context))
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/defaulttags.py", line 209, in render
    nodelist.append(node.render_annotated(context))
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
    return self.render(context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 993, in render
    return render_value_in_context(output, context)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/template/base.py", line 972, in render_value_in_context
    value = str(value)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/utils/html.py", line 397, in <lambda>
    klass.__str__ = lambda self: mark_safe(klass_str(self))
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/forms/boundfield.py", line 33, in __str__
    return self.as_widget()
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/forms/boundfield.py", line 93, in as_widget
    renderer=self.form.renderer,
TypeError: render() got an unexpected keyword argument 'renderer'

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to clusterUI
2. Click on a series
3. click 'Localize Molecules'
4. See error

barentine, 2021-01-25:
Additionally, file uploads are broken
[image]

barentine, 2021-01-25:
I get the 403 on /login too which I didn't expect since we use the `{5 csrf_token %}` inside the post there

barentine, 2021-01-25:
we use crsf_protect decorator on upload_files, but changing that to crsf_exempt is a temporary work around. Looks like `form_upload_file` might just need a `{{ csrf_token }}`

barentine, 2021-02-16:
@David-Baddeley do you care what we do here? OK to CSRF exempt everywhere I'm hitting the verification failure?

barentine, 2021-02-16:
Not sure if this is separate or not, but the glob'ing seems to also be broken?
```
Feb/2021 15:38:07] "GET /status/load/ HTTP/1.1" 200 21617
Not Found: /recipes/find_inputs
[16/Feb/2021 15:38:07] "GET /recipes/find_inputs?glob=Bergamot/2021_2_8/EMC0104S4_2/[A-Z][0-9]*/analysis/*.h5r HTTP/1.1" 404 4569

```

barentine, 2021-02-16:
Note also that we are still droken on django 1.11 (which again, I think is fine as long as we're not on 2.0 or 2.1)
```
Unhandled exception in thread started by <function check_errors.<locals>.wrapper at 0x7fd4a8b490d0>
Traceback (most recent call last):
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/utils/autoreload.py", line 228, in wrapper
    fn(*args, **kwargs)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/management/commands/runserver.py", line 125, in inner_run
    self.check(display_num_errors=True)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/management/base.py", line 359, in check
    include_deployment_checks=include_deployment_checks,
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/management/base.py", line 346, in _run_checks
    return checks.run_checks(**kwargs)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/checks/registry.py", line 81, in run_checks
    new_errors = check(app_configs=app_configs)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/checks/urls.py", line 16, in check_url_config
    return check_resolver(resolver)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/checks/urls.py", line 26, in check_resolver
    return check_method()
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/urls/resolvers.py", line 254, in check
    for pattern in self.url_patterns:
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/urls/resolvers.py", line 405, in url_patterns
    patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/utils/functional.py", line 35, in __get__
    res = instance.__dict__[self.name] = self.func(instance)
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/urls/resolvers.py", line 398, in urlconf_module
    return import_module(self.urlconf_name)
  File "/home/smeagol/miniconda3/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/smeagol/code/python-microscopy/PYME/cluster/clusterUI/clusterUI/urls.py", line 17, in <module>
    from django.urls import path
ImportError: cannot import name 'path'

```

David-Baddeley, 2021-02-17:
re crsf stuff:

- not OK to exempt things where we have asked to protect (it kind of defeats the purpose)
- you shouldn't expect to access the upload URLs directly, but rather only from the POST of the relevant upload form (they need the POST data in any case)
- getting a 403 when you try and access the URL directly is therefore the correct behaviour

there might be something odd going on, however, if it's not working in the correct context

barentine, 2021-02-17:
It's not working in the correct context. correct modulo I've never set up any of the login stuff which I know you added at some point

David-Baddeley, 2021-02-17:
yeah - the auth stuff doesn't do anything useful yet. Super confusing as both forms have the tokens. Which django version?

David-Baddeley, 2021-02-25:
Have an inkling the csrf stuff might be down to binding on localhost vs non-localhost

David-Baddeley, 2021-02-25:
nope, but have a fix.

David-Baddeley, 2021-02-25:
potentially still broken on 1.11, think we've now hit everything for 2.0, 2.1 (fingers crossed)

barentine, 2021-03-02:
All set on 2.1 anyway. Thanks @David-Baddeley . I'm going to close this but reopen a PR suggesting new installs use 2.0 and 2.1 unless either of us feel like testing 1.11 (I for one clearly do not ;) )

closed 2021-03-02 by barentine