python-microscopy/python-microscopy #805: [ENHANCEMENT] apply trait settings in apply_simple

barentine, 2021-01-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I use the GUI to resample a z stack and it doesn't care what spacing I say because we have
`ResampleZStack().apply_simple(input=self.image, z_sampling=float(stuff))` instead of  `ResampleZStack( z_sampling=float(stuff)).apply_simple(input=self.image)`

**Describe the solution you'd like**
either 'fix' apply_simple if we think it's broken, or document that kwargs should only be for inputs/outputs and fix our code everywhere

David-Baddeley, 2021-01-25:
I'd argue that the current behaviour (parameters are in the constructor, input in the `apply(...)` args) is the correct and most intuitive behaviour.  We could probably change the docs (suggestions welcome as I can't see anything which suggests you can pass parameters other than inputs to the `apply()` function). Suggest we also add a check around l185 of `recipes.base` that throws an error if `k` is not in the module inputs so that incorrect usage will generate an error message rather than being silently ignored.

David-Baddeley, 2021-07-12:
Strongly suspect this only hit that particular instance, closing. Can re-open if it proves a recurring problem.

closed 2021-07-12 by David-Baddeley