python-microscopy/python-microscopy #811: [BUG] PYMEImage YZ projection... not working?

barentine, 2021-01-29:
**Describe the bug**
Looks like side view projection is getting the wrong indices... maybe transposing?

**To Reproduce**
Steps to reproduce the behavior:
1. render
[image]

2. crop
[image]

3. swap to a side view
[image]

4. is this a single slice?

**Expected behavior**
project along X within my XY crop

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: osx 10.15
 - Python version: 3.7.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2021-02-02:
As it stands, the xz and yz views do not project (viewer projections are strictly x-y only). This is partly by design to avoid performance issues on large data sets, although the GUI is not clear that this is not being performed. It probably wouldn't hurt to expand the projection logic to xz and yz views, but we'd need to think a bit about how to do so in a high performance way.

David-Baddeley, 2021-04-11:
Mostly fixed with alternative functionality in #972, with removing 'live' projection option in the display settings on the TODO pending a functional colour coded z projection mode.

closed 2021-04-11 by David-Baddeley