python-microscopy/python-microscopy #815: [ENHANCEMENT] ModuleCollection from module list import/register modules

barentine, 2021-02-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I programmatically try and run a recipe in a notebook and forgot to first import PYME.recipes.modules

```
KeyError                                  Traceback (most recent call last)
~/code/python-microscopy/PYME/recipes/base.py in _update_from_module_list(self, l)
    864             try:
--> 865                 mod = all_modules[mn](self)
    866             except KeyError:

KeyError: 'supertile.Supertile'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
<ipython-input-18-24346f5d2b16> in <module>
      1 for well in d.keys():
      2     output_dir = os.path.join(d[well]['directory'], 'recipe_outputs', recipe_stub)
----> 3     recipe = ModuleCollection.fromYAML(recipe_yaml)
      4 
      5     if not os.path.exists(output_dir):

~/code/python-microscopy/PYME/recipes/base.py in fromYAML(cls, data)
    891 
    892         l = yaml.safe_load(data)
--> 893         return cls._from_module_list(l)
    894 
    895     def update_from_yaml(self, data):

~/code/python-microscopy/PYME/recipes/base.py in _from_module_list(cls, l)
    882         yaml or json. Do not call directly"""
    883         c = cls()
--> 884         c._update_from_module_list(l)
    885 
    886         return c

~/code/python-microscopy/PYME/recipes/base.py in _update_from_module_list(self, l)
    867                 # still support loading old recipes which do not use hierarchical names
    868                 # also try and support modules which might have moved
--> 869                 mod = _legacy_modules[mn.split('.')[-1]](self)
    870 
    871             mod.set(**md)

KeyError: 'Supertile'
```

**Describe the solution you'd like**
import PYME.recipes.modules in ModuleCollection init or so?

**Describe alternatives you've considered**
not forgetting

David-Baddeley, 2021-02-01:
If we import `modules` from `base` we run the risk of circular imports. I would also like to have the option of having recipes which didn't import everything (but am not 100% sure that this is needed). Ultimately some of the `recipes.base` stuff is probably due a re-factor (moving `ModuleCollection` into it's own file and potentially renaming to `Recipe` or similar). Probably makes sense to address this then.

David-Baddeley, 2021-11-11:
Refactoring of `ModuleCollection` has now been done for a while which means that we could actually progress this.