python-microscopy/python-microscopy #816: [BUG]Cannot use dSView.module.camera_maps

HelenSun09, 2021-02-01:
**Describe the bug**

In the miniconda3 prompt where I typed "dh5view":
c:\users\helen sun\python-microscopy\PYME\DSView\splashScreen.py:117: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\helen sun\python-microscopy\PYME\DSView\dsviewer.py:144: wxPyDeprecationWarning: Call to deprecated item. Use Append instead.
  tmp_menu.AppendMenu(-1, 'Save &Results', self.save_menu)
WARNING:PYME.DSView.modules:Plugin [arrayView] injects into dsviewer namespace, could result in circular references
WARNING:PYME.DSView.modules:Plugin [shell] injects into dsviewer namespace, could result in circular references
0
WARNING:PYME.DSView.modules:Plugin [metadataView] injects into dsviewer namespace, could result in circular references
ERROR:PYME.recipes.modules:Could not import skimage
Traceback (most recent call last):
  File "c:\users\helen sun\python-microscopy\PYME\recipes\modules.py", line 25, in <module>
    from . import skfilters
  File "c:\users\helen sun\python-microscopy\PYME\recipes\skfilters.py", line 14, in <module>
    import skimage.filters as skf
ModuleNotFoundError: No module named 'skimage'
0
Creating fold panel
c:\users\helen sun\python-microscopy\PYME\DSView\scrolledImagePanel.py:52: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())
c:\users\helen sun\python-microscopy\PYME\DSView\arrayViewPanel.py:1154: wxPyDeprecationWarning: Call to deprecated item ImageFromData. Use :class:`Image` instead.
  img = wx.ImageFromData(ima.shape[1], ima.shape[0], ima.ravel())
c:\users\helen sun\python-microscopy\PYME\DSView\arrayViewPanel.py:513: wxPyDeprecationWarning: Call to deprecated item BitmapFromImage. Use :class:`wx.Bitmap` instead
  im2 = wx.BitmapFromImage(im)
Creating fold panel
Traceback (most recent call last):
  File "c:\users\helen sun\python-microscopy\PYME\DSView\arrayViewPanel.py", line 784, in OnLeftUp
    self.OnSetPosition(event)
  File "c:\users\helen sun\python-microscopy\PYME\DSView\arrayViewPanel.py", line 856, in OnSetPosition
    self.do.xp =int(pos[0]/sc)
  File "c:\users\helen sun\python-microscopy\PYME\DSView\displayOptions.py", line 130, in xp
    raise IndexError('X position out of bounds')
IndexError: X position out of bounds
c:\users\helen sun\python-microscopy\PYME\ui\histLimits.py:289: wxPyDeprecationWarning: Call to deprecated item EmptyBitmap. Use :class:`wx.Bitmap` instead
  MemBitmap = wx.EmptyBitmap(s.GetWidth(), s.GetHeight())

In the miniconda3 prompt where I typed "PYMEClusterofOne":
INFO:root:Serving from directory: C:\Users\Helen Sun\PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 53353 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
dict_values(['http://127.0.0.1:53349/'])
DEBUG:nodeserver:Starting to poll for tasks
nodeserver:INFO:Starting nodeserver on 127.0.0.1:0
INFO:nodeserver:Starting nodeserver on 127.0.0.1:0
127.0.0.1 - - [01/Feb/2021 13:20:50] code 404, message No handler for /favicon.ico
DEBUG:nodeserver:Launching worker processors
usage: taskWorkerHTTP.py [-h] [-p] [--profile-dir PROFILE_DIR]
                         [-s SERVER_PORT]
taskWorkerHTTP.py: error: unrecognized arguments: Sun\PYMEData\LOGS\DESKTOP-DRM5F88\mProf

**To Reproduce**
Steps to reproduce the behavior:
1. Open miniconda3 prompt -> conda activate pyme ->PYMEClusterofOne
2. Open miniconda3 prompt-> conda activate pyme -> dh5view -> load the .dcimg file of dark maps
3.  Click on module -> camera_maps
4. See error

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Windows 10
 PYMEDataServer, running on python 3.6.12 |Anaconda, Inc.| (default, Sep  9 2020, 00:29:25) [MSC v.1916 64 bit (AMD64)]
INFO:root:Serving from directory: C:\Users\Helen Sun\PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 53353 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
dict_values(['http://127.0.0.1:53349/'])

**Additional context**
Add any other context about the problem here.

zacsimile, 2021-02-01:
Do you know the full path it was trying to access when it looked in `Sun\PYMEData\LOGS\DESKTOP-DRM5F88\mProf`?

HelenSun09, 2021-02-01:
The full path: \Users\Helen Sun\PYMEData\LOGS\DESKTOP-DRM5F88\mProf
Dark map DCIMG file: F:\FPLAM Raw Data\20210109

barentine, 2021-02-01:
another space in file path issue, along the lines of #619

barentine, 2021-02-01:
should be fixed in #817. @HelenSun09 thank you for filing the issue - sorry for the trouble. Looks like it didn't have anything to do with camera maps, but please let us know if you run into anything there (or elsewhere!)

David-Baddeley, 2021-02-01:
#817 only modifies archived code which is not used, so is unlikely to fix the issue. 

I think there are two separate issues here:

- a path issue which is preventing one of the workers from coming up (this is unlikely to be fatal, as it will just mean that you have e.g. 3 workers rather than 4 workers on a 4-core machine). This can probably be solved by applying changes analogous to #817 to the current rulenodeserver code.
- a separate issue related to the camera map module. This is a bit trickier, as I don't see anything in the provided traceback output that points to a cause (are we potentially missing an error that gets displayed in the GUI when you call modules->camera_maps?).

HelenSun09, 2021-02-01:
I just tried again. Still gives the same error.

David-Baddeley, 2021-02-01:
dry-running here, I think you are missing a step. `modules->camera_maps` only loads the camera map module. Other than adding a `Processing->Create dark and variance maps` it doesn't do anything. To actually create the maps, you need to select this menu item.

The error you see in the dh5view console about `xp` is harmless.

HelenSun09, 2021-02-01:
@David-Baddeley Thank you! I tried it, you are absolutely right. Thank you very much! I missed that.

closed 2021-02-01 by HelenSun09