python-microscopy/python-microscopy #822: [ENHANCEMENT] retries on failed software trigger frames

barentine, 2021-02-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when the ocassional tile overview locks on
```
DEBUG:PYME.Acquire.HTTPSpooler:Starting spooling: Bergamot/2021_2_2/EMC0104S3_4/D4.pcs
'http://10.150.7.4:15348/Bergamot/2021_2_2/EMC0104S3_4/D4.pcs/metadata.json'
INFO:PYME.Acquire.htsms.rule_ui:protocol name : htsms-tile
DEBUG:PYME.Acquire.Utils.pointScanner:Current positions: {'x': 36000.0, 'x_target': 36000.0, 'y': 27000.0, 'y_target': 27000.0, 'z': 49.9697, 'z_target': 50.0}
Warning: not keeping up with camera, giving up with 0 frames still in buffer
nse
ERROR:PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuORCA:DCAM error for dcamwait_start with return value -2147482879: Lost Frame detected!
```

**Describe the solution you'd like**
when this happens it can be rescued through the shell with
```
scope.frameWrangler.stop()
scope.frameWrangler.Prepare()
scope.frameWrangler.start()
scope.cam.FireSoftwareTrigger()
```
I do not know how necessary the framewrangler restart is, but in the past when we've tried simply pausing and following with another software trigger we haven't had much luck. If I hit this again I'll try just with the trigger first and see what happens.

**Describe alternatives you've considered**
failure

David-Baddeley, 2021-02-04:
a couple of quick thoughts:

- it would probably be good to handle this on the DCAM side
- might be worth replacing the `logger.error(...)` call on line 349 of HamamatsuORCA with `logger.exception()` so we get a traceback and a slightly more informative message.
- potential options for handling on the camera end would be:
    - call `fire_trigger()`again if in triggered mode??
    - return a null frame
    - propagate the exception upwards

for error checking, in the camera driver it might be worth refactoring `DCAMException` as, e.g. `DCAMException(fn, ret_val, message='')`, and keeping `ret_val` as, e.g. an `.error_code` attribute so that the exception handler can check the error code before deciding whether to ignore or propagate the exception.

For this error (which points to buffer corruption) I'd be tempted to propagate the error upwards, and handle at the frame-wrangler level as we are likely to need the camera to be re-started to refresh buffers.

A couple of potential underlying causes:

- stuff bogs down and not reading out the buffer fast enough => camera buffer over-run
- some form of race condition where the camera is re-started and allocates new buffers without releasing old ones (or, after stopAq, releases buffers before they get read out).

David-Baddeley, 2021-02-06:
It might be worth looking at the sequence which generates the error in a bit more detail:

- we print 'nse' which means that the `frameWrangler.bufferOverflowed` flag is set, and also means that we are going to trigger a camera restart.
- the only place frameWrangler.bufferOverflowed` gets set should be on line 274 (at least for the Orca), but this should also print a warning message: 'Warning: Camera buffer overflowing - purging buffer', and then call `StopAq()`, albeit in a different thread.
- we don't see the  'purging buffer' message in the above log (which we should, ideally, see before 'nse')

I think we might be hitting a race condition:

- `Notify()` and `_poll_loop()` are running in separate threads. 
- `_poll_loop()` sets `.bufferOverflowed` while `Notify` is running.
-  `_poll_loop()` does a `wx.CallAfter()` (`_event_loop.call_in_main_thread()`) for `StopAq`
-  because `Notify()` is running in the main thread, this means that `StopAq()` will run after `Notify()` completes
- `Notify()` sees the `bufferOverflowed` flag and restarts the camera, re-allocating buffers etc ... **before** `StopAq()` has been called to stop it.

I'll see if this can be fixed with a little locking ...

David-Baddeley, 2021-02-06:
@barentine - is there a reliable way of triggering this? if I'm correct it's likely to be the result of a (rare) race condition that should be fixed in #827.

barentine, 2021-02-06:
Hi @David-Baddeley - thanks much for looking into this! I haven't tried writing a 'fail' script for it - this one is definitely rare. It seems to happen most commonly at the start of the tiles, but in my mind I worry I'm conflating it with #517.

barentine, 2021-02-08:
Hit a similar error this weekend before pulling #827. Have pulled that in now and will keep you posted
```
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\frameWrangler.py", line 235, in onExpReady
    self.onFrame.send(sender=self, frameData=d)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in send
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\contrib\dispatch\dispatcher.py", line 174, in <listcomp>
    for receiver in self._live_receivers(sender)
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Utils\pointScanner.py", line 220, in tick
    self.scope.cam.FireSoftwareTrigger()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuORCA.py", line 194, in FireSoftwareTrigger
    self.checkStatus(dcam.dcamcap_firetrigger(self.handle, 0), 'dcamcap_firetrigger')
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\Hardware\HamamatsuDCAM\HamamatsuDCAM.py", line 500, in checkStatus
    self.getCamInfo(fn_return))
PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuDCAM.DCAMException: DCAM error for dcamcap_firetrigger with return value -2147483385: Status is not busy!
DEBUG:PYME.Acquire.microscope:restarted framewrangler
```

barentine, 2021-02-11:
The following just occurred (at the head) 30 frames into a tile overview. Notably it did not require a frameWrangler restart manually, just a manual software trigger to kick it back into action
```
0.01
DEBUG:PYME.Acquire.microscope:restarting framewrangler
DEBUG:PYME.Acquire.microscope:restarted framewrangler
DEBUG:htsms-tile:Circular tiler target radius in units of (overlapped) FOVs: 24
DEBUG:PYME.Acquire.Utils.pointScanner:Current positions: {'x': 54000.0, 'x_target': 54000.0, 'y': 9000.0, 'y_target': 9000.0, 'z': 49.9691, 'z_target': 50.0}
DEBUG:PYME.Acquire.HTTPSpooler:Starting spooling: Bergamot/2021_2_8/EMC0104S4_post_testing_2/B2.pcs
'http://10.150.7.1:15348/Bergamot/2021_2_8/EMC0104S4_post_testing_2/B2.pcs/metadata.json'
INFO:PYME.Acquire.htsms.rule_ui:protocol name : htsms-tile
DEBUG:PYME.Acquire.HTTPSpooler:Stopping spooling Bergamot/2021_2_8/EMC0104S4_post_testing_2/A1/10_2_series_S.pcs
'http://10.150.7.5:15348/Bergamot/2021_2_8/EMC0104S4_post_testing_2/A1/10_2_series_S.pcs/final_metadata.json'
'http://10.150.7.3:15348/Bergamot/2021_2_8/EMC0104S4_post_testing_2/A1/10_2_series_S.pcs/events.json'
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
Warning: Camera buffer overflowing - purging buffer
DEBUG:PYME.Acquire.Utils.pointScanner:Current positions: {'x': 54000.0, 'x_target': 54000.0, 'y': 9000.0, 'y_target': 9000.0, 'z': 49.9714, 'z_target': 50.0}
ERROR:PYME.Acquire.Hardware.HamamatsuDCAM.HamamatsuORCA:DCAM error for dcamwait_start with return value -2147482879: Lost Frame detected!
nse

```

David-Baddeley, 2021-02-11:
I think it's a race condition in stop spool, as follows:

- ActionManager polls actions to see if they've finished
- StopSpooling changes the action status to finished
- StopSpooling writes the events file in the main/GUI thread (the same one that is polling the FrameWrangler), locks the GUI, and causes a buffer overflow
- In the meantime, ActionManager is starting the next acquisition, which runs into the pre-existing buffer overflow.