python-microscopy/python-microscopy #826: [ENHANCEMENT] SpoolController.onSpoolStop.connect to pointscanner stop

barentine, 2021-02-05:
**Is your feature request related to a problem? Please describe.**
at least in htsms-tile.py with the circular scanner I connect the scanner `on_stop` signal to the spoolController StopSpool in order to stop spooling at the end of the tile. It is also possible for a user to click `Stop` in the HDFSpoolFrame though, at which point currently the spooling stops but the `pointScanner` keeps going until it finishes. So naturally I'd like to `scope.spoolController.onSpoolStop.connect(self.stop)` in the scanner, but that sounds like it'll either call StopSpooling twice or the scanner's stop twice. Maybe it wouldn't matter. Haven't thought much about it beyond that but am filing this issue so I don't forget to have a look at it some time.

David-Baddeley, 2021-02-06:
The right way of handling this is probably to refactor the scanners so they connect to (or over-ride) the protocol `OnFrame` rather than the `frameWrangler.onFrame` if (and only if) they are being used in a protocol. It's likely to be a bit messy, as they still need to connect to the frame wrangler when not being used in a protocol.

David-Baddeley, 2021-12-05:
Just revisiting this ...  You can already achieve some tidy up with protocols as they currently exist, without the need to hook events.

- clicking `Stop` calls `Spooler.StopSpool()` 
- `Spooler.StopSpool()` calls `Protocol.OnFinish`
- `Protocol.OnFinish()` executes all remaining protocol tasks (there is a seperate argument about whether it should do this)
- A protocol task with an timeslot of `maxint` will get executed after the user clicks on stop. (See, e.g. commented out `ps.stop` tasks in the tiling protocol)

Note the reason these are commented out in current tiling protocols is that you generally want to let the scanner stop the protocol, not the other way around, although it shouldn't hurt to put them back in as a failsafe.

closed 2021-12-05 by David-Baddeley