python-microscopy/python-microscopy #829: [BUG] Spheres display method not working.

lukasfue, 2021-02-08:
**Describe the bug**
Using point display method "spheres" does not work. I do not recieve a traceback for it, the points just disappear when I switch to spheres. Switching to another method, such as pointsprites, works fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Load data
2. Display data as spheres
3. See that they disappear.
4. See error

**Expected behavior**
Points are displayed as spheres.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
python-microscopy=20.10.05
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

David-Baddeley, 2021-02-08:
Probably related to #98

It does work correctly on OSX. Presumably some quirk of the underlying OpenGL implementations.

David-Baddeley, 2021-02-08:
Could possibly be display driver / graphics card dependant

zacsimile, 2021-02-08:
I wonder if there's like a single GL call we use that only works on macs for some reason. There's a couple of platform-specific functions. Could also be different GL versions.

zacsimile, 2021-02-08:
Funny enough my mac has OpenGL 2.1 and my work Windows has OpenGL 1.1 so I'm thinking that might be it. Can @David-Baddeley or @barentine check their Linux OpenGL version with 1glxinfo | grep "OpenGL version"`?

David-Baddeley, 2021-02-11:
going to be optimistic and close this .... would be good if someone could test on linux.

closed 2021-02-11 by David-Baddeley