python-microscopy/python-microscopy #832: [BUG] can click-through recipe configure traits dialogs, opening new ones on accident

barentine, 2021-02-09:
**Describe the bug**
in the recipeGUI, at least in PYMEImage, I'm hitting a funny bug where I can accidentally open a fair number of recipe module configuration dialogs while trying to edit the first one I pulled up

**To Reproduce**
Steps to reproduce the behavior:
1. PYMEImage something
2. Click on 'Recipe'
3. add some stuff
4. click a module to configure it
5. click in that dialog into a field which happens to be over the tile for another recipe module in the gui behind it
6. notice that you get another recipe module dialog open (behind?) the one you were working on in the first place (maybe it's only for the same one you have up? Didn't try others)

**Expected behavior**
one dialog at a time

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] osx 10.15
 - Python version: 3.7.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

zacsimile, 2021-02-10:
Yep, thanks for flagging that @barentine. Same issue. I ran into this in PYMEVis.

David-Baddeley, 2021-02-10:
Hmm, this looks like an error with traitsui and/or matplotlib. Might be hard to work around without significant re-writes to recipe display and/or editor.

Simplest potential fix would be to make the pop-up editors modal, but not sure that will be enough.

csoeller, 2021-02-10:
I can also attest to this behaviour. A bit annoying, can be worked around (move window before doing anything with it), but would be good to fix for sure.

David-Baddeley, 2021-02-10:
A bit more on potential solutions:

- try making the dialogs modal (this might also improve cancel handling - see #838). Potentially combine with a new editor dialog which shows the module help (on TODO list in any case).
- make editor pop up as a panel on the side/bottom of the recipe display rather than in a separate window
- change to using the html based recipe editor (advantageous in that we'd only need to maintain one editor, but potentially problematic on linux and/or windows, also not there yet)

David-Baddeley, 2021-02-18:
should be fixed with #869 (at least until we refactor stuff properly)

closed 2021-02-18 by David-Baddeley