python-microscopy/python-microscopy #834: [BUG] Recipe editor mouse target doesn't understand layers, focus not set

zacsimile, 2021-02-10:
**Describe the bug**
I add a `FilterTable` to my recipe, I double click it, open the filter dialog, right click and hit `Add` (we should have an `Add` button so this is clearer). If there's another tile behind `Add`, the dialog for that tile open as well when I press `Add`, presumably due to a mouse pointer target checking operation that doesn't understand layers. Clearly the focus is still on the recipe editor window. If this wasn't annoying enough, when I close the new dialog it re-reuns the recipe, so I'm left sitting there for a few seconds before I can use the dialog I wanted.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

barentine, 2021-02-10:
sounds like a duplicate of #832

zacsimile, 2021-02-10:
Moved conversation to #832.

closed 2021-02-10 by zacsimile