python-microscopy/python-microscopy #835: [ENHANCEMENT] Eliminate all continuous execution of the pipeline

zacsimile, 2021-02-10:
Force the user to press enter before the pipeline updates. I can't think of single time I've ever wanted the recipe to execute as soon as I type something in. 

Particularly annoying is when I want to go from a single- to a double-digit number in the pipeline. The default will be `1.0` and if I want to go to `50.0` then I have to type `5`, wait, `0`, wait or (because I'm tricky and know how this works now) add a zero `10`, wait, and then replace the `1` with a `5` for `50`, wait. 

I'd like to not game execution time every time I want to update something in the pipeline. If we could just wait until I pressed enter, all of this would be solved.

lukasfue, 2021-02-10:
I fully agree with Zach on this.

barentine, 2021-02-10:
same, I always toggle `pipeline.recipe.execute_on_invalidation` in the shell to get around this

David-Baddeley, 2021-02-10:
This is already fixed on a few of the more computationally expensive
modules (e.g. meshing) - we just need to make that work across all modules.

zacsimile, 2021-02-10:
It’s not fixed on meshing. Density is still continuous execution.

zacsimile, 2021-02-10:
Can we just set `execute_on_invalidation` to `False` across the board, as Andrew suggests?

David-Baddeley, 2021-02-10:
We need to do something so that we don't update on incremental parameter changes - especially for the more computationally expensive modules - but setting `execute_on_invalidation` to `False` across the board is pretty much a non-starter for the following reasons:

- It would require adding an 'execute' button, and an additional manual step - if PYME has one weakness it is the number of manual steps that are required and I'd be loath to add another.
- it would require some clear visual cue-ing that the pipeline was invalid and the user needed to click 'execute' (I can't really think of a way of making this slick so that it was obvious)
- It adds way too much overhead to simple pipeline operations such as filtering
- it will mess up display of anything which depends on a module that got invalidated (as it stands, it would crash, and whilst we might be able to find a way of handling it, it's not going to be clean/pretty - you risk, e.g. loosing colour scaling settings and the like).
- We could no-longer really claim that we are interactive

I was incorrect on where we had fixed it - we fixed the incremental update problem in the layers (for point size and alpha), not any of the recipe modules - see e.g. line 286 of LMVis/layers/pointcloud.py. Applying a similar fix to the recipe modules, so that the traits text editor had `TextEditor(auto_set=False, enter_set=True,...)` across the board would go a long way, and might be enough (this might involve monkey-patching traits so that behaviour was the default). In the short term we could potentially do this with custom views for the most painful modules.

Even better would be if the update was delayed until the entire recipe module lost focus (or you hit enter in one of it's controls), although this would be quite a bit harder to implement robustly (how do you, e.g., handle dropdowns or checkboxes), and having to click out of something before it updates is a bit problematic from a user-experience perspective.

A compromise might be: 
- each control only updates when you either loose focus or hit enter
- you can double click on the caption of a module in the vertical display to pop up a modal editor (as with the full recipe editor) which only updates when you close the dialog (so you can alter multiple parameters at once).

David-Baddeley, 2021-02-10:
Actually the monkey patching might be easier than I'd thought - we should try:

```python
from traits.api import Bool
import traitsui.editors.text_editor
traitsui.editors.text_editor.TextEditor.auto_set = Bool(False)
traitsui.editors.text_editor.TextEditor.enter_set = Bool(True)
```

David-Baddeley, 2021-02-10:
above doesn't work :(

David-Baddeley, 2021-02-11:
this does though ...

```python
#Monkey-patch traits so that editors need enter / focus change to update by default
from traits import trait_types
trait_types.TraitType.auto_set = False
trait_types.TraitType.enter_set = True
```

closed 2021-11-11 by David-Baddeley