python-microscopy/python-microscopy #836: [ENHANCEMENT] Make data pipeline pane resizable, scrollable, or both

zacsimile, 2021-02-10:
Often, the `Data Pipeline` exceeds the size of its pane.

[image]

This forces me to un-pin the layers to show everything.

[image]

Of course, there was plenty of space in the layers pane, if I could've resized. At some point resizing won't do, so making the data pipeline scrollable would be nice. A third option is to make the data pipeline window dockable elsewhere (or just drag it out of the panel and have it float on the side).

David-Baddeley, 2021-02-10:
The pipeline pane automatically resizes to take up all the available space. We might be able to look at doing a bit more dynamic resizing in the layers panel (it currently uses enough space to be able to display the largest of the layer controls - meaning that there can be a bit of extra space when, e.g. using solid colours such that the histogram display is not shown).

The potential downside to messing with the layer panel sizing is that it might make things unpleasant when - e.g. changing from solid colour to colour mapped and back again as a) it would be hard to get all the resizing to happen correctly and b) if resizing was implemented correctly, the controls would jump up and down whenever you changed something which changed the display size.

I've toyed with the idea of moving the layer panel to the RHS of the window, which would be one way of sorting this out, albeit at the cost of less display room.

David-Baddeley, 2021-02-10:
**TLDR** - the pipeline pane is already resizeable and resizes automatically to fill the available space

zacsimile, 2021-02-10:
I guess by resizable I mean I'd like to click-and-drag to resize, which I currently can't do.

David-Baddeley, 2021-02-11:
Re scrolling idea - this is hard to realise when something is not a fixed size, but probably worth looking at in a bit more detail.

David-Baddeley, 2021-02-11:
clicking and dragging wouldn't really help, as it is already at the maximum size it can be (the layer panel potentially reserves too much space when you have all layers collapsed, and we should look at this, but it's going to be tricky).

zacsimile, 2021-02-11:
What about arbitrary resize/arbitrary docking a la Adobe products?

David-Baddeley, 2021-12-06:
Revisiting this after ages, but adobe resizing also has minimum sizes that you can't resize smaller than - we're effectively hitting the minimum size limit already on the layers pane. We have the arbitrary docking with the right hand side panels, but it's not super slick, and adds a lot of window chrome (and associated dead space).  It's also really hard to get the dockable panels to only use the amount of space they require, rather than a fixed proportion of the overall space (this is a wx.aui limitation, but one we have to work with) To get something as slick as the adobe products, we'd need to effectively write our own windowing toolkit from scratch - I don't think it's something you can easily cobble on top of wx (or any other mainstream windowing library). It might be doable at reasonable cost with an HTML5 based ui, so it's something to consider long term.