python-microscopy/python-microscopy #837: [ENHANCEMENT] Don't collapse every un-pinned pane when I open a collapsed pane

zacsimile, 2021-02-10:
Proportionally resize instead. For example,

[image]

And then I open Output Filter

[image]

There was plenty of space to expand output filter while keeping Layers up. Even if part of layers gets hidden, that would be preferable to them disappearing entirely.

David-Baddeley, 2021-02-10:
This is related to how the layer pane is sized (max possible layer size), also probably (in terms of solution) a duplicate of #836

David-Baddeley, 2021-02-11:
Also, partially hidden doesn't work for a number of reasons - not least how do you re-show /access the hidden bits. I think it has to be all or nothing.

zacsimile, 2021-02-11:
I guess I'm advocating for partially hidden + a scroll bar in each pane. As you mention in #836, this might be tricky, though.

zacsimile, 2021-02-11:
Another option would be arbitary docking/resizing of the different panes, a la Adobe products.

zacsimile, 2021-02-11:
I'm going to close this as I agree this entire conversation belongs in #836.

closed 2021-02-11 by zacsimile