python-microscopy/python-microscopy #838: [BUG] Don't run recipes if I hit cancel

zacsimile, 2021-02-10:
If I click on a recipe tile in the `Pipeline Recipe` tab in `PYMEVis` and the dialog pops up and then I hit cancel, the recipe re-runs. This shouldn't happen.

David-Baddeley, 2021-02-10:
Which traits / traitsui version? This smells a little like a traits error.

David-Baddeley, 2021-02-11:
Actually, there is no way of robustly handling a cancel - traits are updated as you edit them. This makes cancel functionally equivalent to OK. Have removed the cancel button as it was confusing.

David-Baddeley, 2021-02-11:
PS - I assume it only runs if you have changed a parameter

closed 2021-02-11 by David-Baddeley

zacsimile, 2021-02-11:
It runs even if I don’t change a parameter. If I just hit cancel, it runs.

David-Baddeley, 2021-02-11:
That's odd, as it should check if new_params==old_params and only run if they are different.

David-Baddeley, 2021-02-11:
see `ModuleBase.edit_no_invalidate()`

zacsimile, 2021-02-11:
You sufficiently convinced me to try again. I even add a print statement "Executing..." to a few of the `execute()` commands in some recipes (e.g. DBSCAN). If I click and then hit cancel, I get the print statement, even if I change no parameters.

David-Baddeley, 2021-02-11:
I've found the bug here which is making things execute when they shouldn't. Fix to follow (probably the *real* fix to this issue - I might have closed prematurely)

zacsimile, 2021-02-11:
Thanks for pushing on this, @David-Baddeley!