python-microscopy/python-microscopy #840: [BUG] actions.FunctionAction comparison

barentine, 2021-02-10:
**Describe the bug**
```
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ActionManager.py", line 184, in Tick
    self.currentTask = self.actionQueue.get_nowait()
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 192, in get_nowait
    return self.get(block=False)
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 174, in get
    item = self._get()
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 230, in _get
    return heappop(self.queue)
TypeError: '<' not supported between instances of 'FunctionAction' and 'FunctionAction'
Traceback (most recent call last):
  File "c:\users\bergamot\code\python-microscopy\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ActionManager.py", line 184, in Tick
    self.currentTask = self.actionQueue.get_nowait()
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 192, in get_nowait
    return self.get(block=False)
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 174, in get
    item = self._get()
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\queue.py", line 230, in _get
    return heappop(self.queue)
TypeError: '<' not supported between instances of 'FunctionAction' and 'FunctionAction'
```

David-Baddeley, 2021-02-10:
This looks like it has to be a `nice` collision - is there a race condition when adding actions from multiple workers? Potentially resolvable with some locking on the endpoint.

barentine, 2021-02-11:
closing, will reopen if we hit this again

barentine, 2021-02-13:
this is still very much an issue

barentine, 2021-02-13:
I thought maybe it was queueing these turnAllLasersOff  with `np.iinfo(int).max` for their nice, but 
```
In [5]: np.iinfo(int).max + (time.time() * 1e-10) == np.iinfo(int).max
Out[5]: False

In [6]: np.iinfo(int).max + (time.time() * 1e-10) > np.iinfo(int).max
Out[6]: True
```

David-Baddeley, 2021-02-14:
Using `nice=maxint` is definitely outside what I'd expected when writing the code (where I'd been expecting that the same conventions as for the command line / system call would be used - i.e. a reasonably small integer).  I'll probably add docs and an assertion on `nice` to force it to, e.g. -20:20.

Using `maxint` will definitely give issues depending on floating point precision. A little bit of testing suggests this is the case for all reasonable delays between subsequent calls.

```python
>>># small delay between calls (how long it takes time.time() to execute)
>>> np.iinfo(int).max + ((time.time())* 1e-10) == np.iinfo(int).max + (time.time() * 1e-10)
True
>>># 1 sec delay between calls
>>> np.iinfo(int).max + ((time.time() +1)* 1e-10) == np.iinfo(int).max + (time.time() * 1e-10)
True

....
>>> # 5e12 s delay between calls
>>>np.iinfo(int).max + ((time.time() +5e12)* 1e-10) == np.iinfo(int).max + (time.time() * 1e-10)
True
>>> # 5e14s delay between calls
>>>np.iinfo(int).max + ((time.time() +5e14)* 1e-10) == np.iinfo(int).max + (time.time() * 1e-10)
False
```

David-Baddeley, 2021-02-14:
Note - with the current -20:20 bounds, we could still run into issues if timing between calls is <50 usec

David-Baddeley, 2021-02-14:
Actually, we won't run into that, as we enforce 1ms minimum increment to the timestamp between calls.

barentine, 2021-02-16:
should be set with #854

closed 2021-02-16 by barentine