python-microscopy/python-microscopy #845: [BUG] Measure2D breaks when using 'output' as target name

csoeller, 2021-02-11:
**Describe the bug**

I believe one could previously select 'output' as the name for the result of the Measure2D recipe module and it would overlay the contours on the image. This seems now broken. Not sure if this is a known bug.

**To Reproduce**
Steps to reproduce the behavior:
1. run yaml recipe
4. See error as below

 
```
Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/recipes.py", line 158, in RunCurrentRecipe
    self.dsviewer.pipeline.OpenFile(ds = cache)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/pipeline.py", line 825, in OpenFile
    self.recipe.execute()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/base.py", line 718, in execute
    m.execute(self.namespace)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/localisations.py", line 169, in execute
    mdh = fitResults.mdh
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/tabular.py", line 195, in __getattr__
    raise AttributeError("'%s' has not attribute '%s'" % (self.__class__, item))
AttributeError: '<class 'PYME.IO.tabular.CachingResultsFilter'>' has not attribute 'mdh'
```

**Expected behavior**
Should just overlay contours.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- python-microscopy=20.12.10
- python=2.7.17, platform=darwin
- numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

**Additional context**

yaml recipe

```yaml
- processing.FlexiThreshold:
    inputName: input
    method: fractional
    outputName: thresh
    parameter: 0.8
- processing.BinaryFillHoles:
    inputName: thresh
    outputName: tfilled
- processing.Label:
    inputName: tfilled
    minRegionPixels: 35
    outputName: labels
- measurement.Measure2D:
    inputIntensity: input
    inputLabels: labels
    measureContour: true
    outputName: measurements
```

David-Baddeley, 2021-02-14:
I think we can separate this into two parts:
 
- the error you get in the traceback 
- getting the contour display working again.

The first one should be reasonably simple to fix. The second, unfortunately, probably needs a bit more work. Displaying the contours was/is not implemented in a particularly robust and/or portable way - particularly as we support more complex recipe workflows, and the current contour code has a lot of ad-hoc, case specific logic in it.

David-Baddeley, 2021-02-14:
On second thoughts, if we fix part a, we might also fix the contours (at least for now - although they will still need revisiting)

The first error looks like metadata is not being propagated through the modules in the recipe. Where does `FlexiThreshold` come from? This is not part of the standard `processing` module, and this might be the problem.

csoeller, 2021-02-15:
Agreed. Let's briefly focus on the first error - missing `mdh`. It occurs even with the following very simple recipe that uses only base modules:

```yaml
- processing.SimpleThreshold:
    inputName: input
    outputName: filtered_image
    threshold: 20000.0
- processing.Label:
    inputName: filtered_image
    minRegionPixels: 50
    outputName: labels
- measurement.Measure2D:
    inputIntensity: input
    inputLabels: labels
    outputName: output
```

## Important observations

1. It only occurs when the special `outputName` `output` is chosen, works fine with any other name.
2. The error is thrown in the `AddPipelineDerivedVars` module which appears to be only added and executed when the `output` name is chosen.

My wild guess at the moment, without any actual drilling down, is that this error is only triggered when a tabular object is using the `output` name in a recipe.
For reference, the error is reproduced below.

```
 Error whilst running Recipes>Run Current Recipe	F5
==============================================
python-microscopy=20.12.10
python=2.7.17, platform=darwin
numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/progress.py", line 107, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/recipes.py", line 158, in RunCurrentRecipe
    self.dsviewer.pipeline.OpenFile(ds = cache)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/pipeline.py", line 825, in OpenFile
    self.recipe.execute()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/base.py", line 718, in execute
    m.execute(self.namespace)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/localisations.py", line 169, in execute
    mdh = fitResults.mdh
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/tabular.py", line 195, in __getattr__
    raise AttributeError("'%s' has not attribute '%s'" % (self.__class__, item))
AttributeError: '<class 'PYME.IO.tabular.CachingResultsFilter'>' has not attribute 'mdh'

```

csoeller, 2021-02-15:
I had a further look. Selecting the special `output` name tag triggers the following code in `RunCurrentRecipe`:

```python
            if isinstance(self.outp, ImageStack):
                _display_output_image(self.outp)
            elif not self.outp is None:
                from PYME.IO import tabular
                import six
                
                cache = tabular.CachingResultsFilter(self.outp)
                self.dsviewer.pipeline.OpenFile(ds = cache)
                self.dsviewer.pipeline.filterKeys = {}
                self.dsviewer.pipeline.Rebuild()
                self.dsviewer.view.filter = self.dsviewer.pipeline
```
Here, our output is not an image stack, so it is assumed to be a an event table, and the code tries to make it into a VisGUI pipeline. In our case, the output is not an image but also is not an events table, but rather a table of measurements. I believe this is where it currently bombs.

Maybe this is where the additional check for our type of data table needs to be inserted and the contour plotting code needs to be invoked?

David-Baddeley, 2021-02-15:
You are correct so far (I should have explained it better in my previous response) but the underlying problem would seem to be that the tabular output doesn't have any metadata. It should have got this from the original image, propagated through each step of the recipe. Somewhere along the way this is falling out. 

Ideally we'd find and fix this. We could also make it so that the subsequent handling didn't depend on having metadata, but on some level this masks the problem.

In terms of special casing the output - as it stands, it uses the pipeline to handle any tabular data, and that is OK - it doesn't strictly need to be localisation events. The reason behind using the pipeline is just to give a consistent interface. The contour code checks for a `'contour'` key in the pipeline output and plots them if it's present.

David-Baddeley, 2021-02-16:
should hopefully be fixed in #856, along with #846 

@csoeller - could you give it a go?

csoeller, 2021-02-16:
Hi @David-Baddeley, I still needed the below patch, but then it worked.

Note, in the past the viewer had a panel to disable overlays which could switch the green center positions off. But not essential. Also not sure if patch below is preferred way to do it but as said makes it work.

[Update] Noticed that selecting `LMDisplay` seemed to bring up `Overlays` panel and can also see positions of clusters in LMDIsplay which seems a bonus.

[image]

```patch
diff --git a/PYME/LMVis/pipeline.py b/PYME/LMVis/pipeline.py
index 2af33327b..3d6d48215 100644
--- a/PYME/LMVis/pipeline.py
+++ b/PYME/LMVis/pipeline.py
@@ -824,7 +824,10 @@ class Pipeline:
         
         self.recipe.execute()
         self.filterKeys = {}
-        self.selectDataSource('filtered_localizations') #NB - this rebuilds the pipeline
+        if 'filtered_localizations' in self.dataSources.keys():
+            self.selectDataSource('filtered_localizations') #NB - this rebuilds the pipeline
+        else:
+            self.selectDataSource(self.dataSources.keys()[0])
 
         # FIXME - we do this already in pipelinify, maybe we can avoid doubling up?
         self.ev_mappings, self.eventCharts = _processEvents(ds, self.events,
```

csoeller, 2021-02-17:
fixed by #866

closed 2021-02-17 by csoeller