python-microscopy/python-microscopy #846: [BUG] writing table from Measure2D generates broken csv and hdf

csoeller, 2021-02-11:
**Describe the bug**
Generated table of measurements with Measure2D module. Saving using the table viewer seems to result in csv and hdf that are unreadable by all readers that I tried.

If these are indeed bugs (and not errors in myusage) then PYME is doing a poor job in preventing and detecting regressions. I know it's hard but this dents confidence. If it is all my fault I'd be happy to learn what the issues area and hang my head in shame.

**To Reproduce**
Steps to reproduce the behavior:
1. Run yaml recipe (see below)
2. Open generated table and save either in cvs or hdf format
4. Despair trying to read generated files

Question: is there now any other way to save these measurements?

**Expected behavior**
Before I could just open these with excel, pandas csv and hdf readers.

**Screenshots**

Table viewer:

[image]

Unreadable csv in Emacs:

[image]

Unreadable hdf in HDFViewer:

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**

   -  python-microscopy=20.12.10
   -  python=2.7.17, platform=darwin
   - numpy=1.16.6, wx=3.0.2.0 osx-cocoa (classic)

**Additional context**

yaml recipe
```yaml
- processing.FlexiThreshold:
    inputName: input
    method: fractional
    outputName: thresh
    parameter: 0.8
- processing.BinaryFillHoles:
    inputName: thresh
    outputName: tfilled
- processing.Label:
    inputName: tfilled
    minRegionPixels: 35
    outputName: labels
- measurement.Measure2D:
    inputIntensity: input
    inputLabels: labels
    measureContour: false
    outputName: measurements
```

David-Baddeley, 2021-02-11:
`Measure2D` has always been a broken in terms of outputs - it generates a bunch of columns with vector or array entries for each object rather than scalars and these do not trivially map to ".csv". HDFs generated from `Measure2D` have never been readable using anything other than `pytables` (the vector/array entries map to a numpy 'Object' dtype, and are stored in a `pytables` specific format in the .hdf). If .csv output has worked at points in the past, it has been a fluke (likely depends on the size of the detected objects and/or the numpy version, as the problem is in how arrays are represented as text). This has changed a little recently to make things more or less functional on py3 (where it previously didn't work at all).

To generate sane output that you stand a good chance of reading elsewhere, you need to select only those columns you are interested in for the output. The `tablefilters.SelectTableColumns` module exists for this purpose. 

Ultimately this should all go away with #524 when that gets done. The vision for #524 is to have a re-factored interface where you can select which measurements you want to perform, with only those which have sane, saveable, output selected by default.  In the meantime we should probably mention the saveability issue and `SelectTableColumns` workaround in the `Measure2D` docs.

csoeller, 2021-02-11:
Thanks for explanations. I will use the column filter. It would be good to see how the contour visualisation can be enabled in some way to see what the settimngs have come up with.

David-Baddeley, 2021-02-11:
I'll probably also change the .csv exporter so that it skips anything in a dtype that is not going to be readable.

csoeller, 2021-02-16:
I checked the effect of #856 by leaving the contours in the filtered measurements and saved that as CSV. This still retains all the newlines in the written contours since the use of `strip` only fixes leading and trailing characters. I did the following experiment using `replace` instead and replacing delimiters and newlines with spaces, just in case. This makes things behave more nicely. For good measure also do this on other strings, just in case we encounter this in already stringified entries, as below. If useful could raise a PR.

```patch
diff --git a/PYME/IO/tabular.py b/PYME/IO/tabular.py
index f0f08cbbb..204ae12da 100644
--- a/PYME/IO/tabular.py
+++ b/PYME/IO/tabular.py
@@ -166,12 +166,12 @@ class TabularBase(object):
         def fmt(d):
             if np.isscalar(d):
                 if isinstance(d, six.string_types):
-                    return str(d)
+                    return str(d).replace(delim,' ').replace('\n',' ')
                 else:
                     return '%e' % d
             else:
                 # try to make sure odd objects don't break file formatting
-                return ('"%s"' % str(d).strip(delim).strip('\n'))
+                return ('"%s"' % str(d).replace(delim,' ').replace('\n',' '))
     
         of = open(outFile, 'w')
     
```

csoeller, 2021-02-17:
fixed by #866

closed 2021-02-17 by csoeller