python-microscopy/python-microscopy #851: [BUG] shadey shaders segfaulting my gaussian render in PYMEVis

barentine, 2021-02-12:
**Describe the bug**

<details> <summary> grotesque amount of stuff, including a traceback </summary> 

```

(base) smeagol@smeagol:~/code/python-microscopy$ PYMEVis '/media/smeagol/d5b5c790-4eba-4212-b990-062bba0b7a3b/data/2019_01_21_atubulinCF568_sec61bAF647/recipe_outputs/mapped/fov8_8kframes_filtered.hdf' 
/home/smeagol/miniconda3/lib/python3.6/_collections_abc.py:702: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  return len(self._mapping)
/home/smeagol/miniconda3/lib/python3.6/_collections_abc.py:720: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  yield from self._mapping
/home/smeagol/code/python-microscopy/PYME/misc/extraCMaps.py:73: MatplotlibDeprecationWarning: 
The revcmap function was deprecated in Matplotlib 3.2 and will be removed two minor releases later. Use Colormap.reversed() instead.
  cmapdat_r = cm.revcmap(ndat[cmapname])
Trying to load 3rd party recipe module nep_fitting.recipe_modules.nep_fits
Loaded 3rd party recipe module nep_fitting.recipe_modules.nep_fits
Trying to load 3rd party recipe module pyme_rcnn.recipe_modules.rcnn_segment
Loaded 3rd party recipe module pyme_rcnn.recipe_modules.rcnn_segment
Trying to load 3rd party recipe module bleaching_tools.python_microscopy.recipe_modules.masked_statistics
ERROR:PYME.recipes.modules:Error loading plugin: bleaching_tools.python_microscopy.recipe_modules.masked_statistics
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'bleaching_tools'
Trying to load 3rd party recipe module pyme_omero.recipe_modules.omero_upload
Loaded 3rd party recipe module pyme_omero.recipe_modules.omero_upload
Trying to load 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
Loaded 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
Trying to load 3rd party recipe module localization_survey.recipe_modules.roi_cropping
Loaded 3rd party recipe module localization_survey.recipe_modules.roi_cropping
Trying to load 3rd party recipe module cb_analysis.python_microscopy.recipe_modules.cb_clustering
Loaded 3rd party recipe module cb_analysis.python_microscopy.recipe_modules.cb_clustering
Trying to load 3rd party recipe module nucleus.python_microscopy.recipe_modules.nucleus_metrics
ERROR:PYME.recipes.modules:Error loading plugin: nucleus.python_microscopy.recipe_modules.nucleus_metrics
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'nucleus'
Trying to load 3rd party recipe module autoroi.recipe_modules.retrieval
Loaded 3rd party recipe module autoroi.recipe_modules.retrieval
Trying to load 3rd party recipe module nucleus.python_microscopy.recipe_modules.distance_to_membrane
ERROR:PYME.recipes.modules:Error loading plugin: nucleus.python_microscopy.recipe_modules.distance_to_membrane
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'nucleus'
Trying to load 3rd party recipe module nucleus.python_microscopy.recipe_modules.shell_fitting
ERROR:PYME.recipes.modules:Error loading plugin: nucleus.python_microscopy.recipe_modules.shell_fitting
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'nucleus'
Trying to load 3rd party recipe module autoroi.recipe_modules.mrcnn_segment
Loaded 3rd party recipe module autoroi.recipe_modules.mrcnn_segment
Trying to load 3rd party recipe module autoroi.recipe_modules.features
Loaded 3rd party recipe module autoroi.recipe_modules.features
Trying to load 3rd party recipe module cell_death_detection.python_microscopy.recipe_modules.cell_death
ERROR:PYME.recipes.modules:Error loading plugin: cell_death_detection.python_microscopy.recipe_modules.cell_death
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'cell_death_detection'
Trying to load 3rd party recipe module nucleus.python_microscopy.recipe_modules.astigmatism_tools
ERROR:PYME.recipes.modules:Error loading plugin: nucleus.python_microscopy.recipe_modules.astigmatism_tools
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/recipes/modules.py", line 41, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named 'nucleus'
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'
quaternion module not found, disabling custom clip plane orientations
[... 332 lines cut ...]
   backtrace from your debugger if you break on the gdk_x_error() function.)
Segmentation fault (core dumped)

```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. PYMEImage /file
2. Generate > Gaussian
3. oh wait, that's 2D, that's not what I wanted
4. Generate > 3D Gaussian
5. set it up
6. run
7. see error

**Expected behavior**
no seg fault

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: ubuntu 16.04
 - Python version: 3.6
 - numpy version: 1.16
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2021-02-14:
It's highly likely that this is a bug in one of PyOpenGL, wxPython, or GTK, which makes it rather hard to track down (and to some extent not our problem). It's rather unlikely that the shader is at fault (note the warning that these errors are reported asynchronously.

First debugging steps would be to see if this error can be reproduced in any other way (is it specific to 3D Gaussians). If (and this seems rather unlikely) the exact same sequence of steps is required to reproduce, then we could look for memory corruption / out of bounds access / reference counting errors in the 3D Gaussian code.

David-Baddeley, 2021-02-14:
Hmm, I get a similar, but different error on 18.0.4

I get a `BadMatch` error from the x server, no segfault

David-Baddeley, 2021-02-14:
A few more details:

- does not require generating a 2D image first
- not specific to Gaussian rendering (also effects 3D histogram, and 3D triangulation)
- does not effect any of the 2D modes.

These lead me to suspect that it's something about the display of 3D image stacks which is to blame (potentially GLComp).

David-Baddeley, 2021-02-14:
Narrowed down a bit further - it's the GL image viewer on 3D data

David-Baddeley, 2021-02-15:
Because the GL image viewer is currently a bit broken for stacks anyway (no way of setting the z-position & just displays the first frame), a temporary workaround would be to make lines 140 and 142 of PYME.DSView.modules.visgui only run for 2D images.

David-Baddeley, 2021-02-25:
workaround in #892, closing, but might need to revisit when we get glCanvas working for 3D data.

closed 2021-02-25 by David-Baddeley