python-microscopy/python-microscopy #855: [ENHANCEMENT] hook laser on/off/power change events even outside of protocol

barentine, 2021-02-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I look back to see laser intensities as I e.g. turned on 405 with a new dye in a new buffer and find there's nothing in the events for it

David-Baddeley, 2021-02-15:
Definitely on the TODO list. Best long term solution is probably hook laser GUI through `scope.state` rather than the lasers directly and log changes to the scope state (probably in parallel to the current event log). This would give a much more generalisable and robust solution than the current events approach where every bit of hardware adds it own custom events.

barentine, 2021-02-16:
That'd be nice