python-microscopy/python-microscopy #859: [discussion] does psfExtraction `Tag Points` do anything?

barentine, 2021-02-16:
**What are you trying to do?**
be lazy and not `Tag` each of my beads
**Where are you stuck?**
nothing happens and i haven't looked at the code yet

David-Baddeley, 2021-02-17:
Yep, although you need to use the `blobFinding` module first to detect the beads.

1) load the `blobFinding` and `psfExtraction` modules
2) use the `Find` button in the `ObjectFinding` module 
    
[image]

3) use `Tag Points` to automatically tag all of the found beads

David-Baddeley, 2021-03-01:
@barentine - can we close this? 

Alternatively, put it on the docs TODO

barentine, 2021-03-01:
Sounds good

barentine, 2021-03-01:
Thanks David

closed 2021-03-01 by barentine