python-microscopy/python-microscopy #861: [DISCUSSION]

csoeller, 2021-02-16:
**Describe the bug**
Hi @David-Baddeley, @barentine, @zacsimile, I am not sure if I have a hosed installation or if something changed in latest git. Opening any `.h5r` I now get:

```
Traceback (most recent call last):

  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_misc.py", line 1367, in Notify

    self.notify()

  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 16869, in Notify

    self.result = self.callable(*self.args, **self.kwargs)

  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/visCore.py", line 754, in OpenFile

    self.pipeline.OpenFile(filename, **args)

  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/pipeline.py", line 825, in OpenFile

    self.recipe.execute()

  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/base.py", line 718, in execute

    m.execute(self.namespace)

  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/recipes/localisations.py", line 291, in execute

    cached_output.mdh = output.mdh

AttributeError: can't set attribute
```

Nothing to look at in the viewer at this stage, obviously.

**To Reproduce**

Open any `h5r` file.

**Expected behavior**
Not like this.

csoeller, 2021-02-16:
Looks we need to add a setter for the `mdh` property to the cached class. With that it works again. Patch below.

```patch
diff --git a/PYME/IO/tabular.py b/PYME/IO/tabular.py
index f0f08cbbb..a78b02176 100644
--- a/PYME/IO/tabular.py
+++ b/PYME/IO/tabular.py
@@ -917,6 +917,10 @@ class CachingResultsFilter(TabularBase):
     def mdh(self):
         return self.resultsSource.mdh
 
+    @mdh.setter
+    def mdh(self, mdh):
+        self.resultsSource.mdh = mdh
+
 @deprecated_name('mappingFilter')
 class MappingFilter(TabularBase):
     _name = "Mapping Filter"
```

barentine, 2021-02-16:
Yikes, and can second the bug. Thanks for flagging it @csoeller 

```
===========================================================================
Error in rule: 000006-086300e0c164486f8c7e96426e9cbbca running task: 1765 on drno:29411

taskDescr:
----------
{'id': '000006-086300e0c164486f8c7e96426e9cbbca~1765', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2021_2_8/EMC0104S4_2/H9_2/analysis/8_2_series_CLB.h5r'}, 'output_dir': 'pyme-cluster://villains/Bergamot/2021_2_8/EMC0104S4_2/', 'taskdef': {'recipe': "- localisations.AddPipelineDerivedVars:\n    inputEvents: ''\n    inputFitResults: FitResults\n    outputLocalizations: Localizations\n- localisations.ProcessColour:\n    input: Localizations\n    output: colour_mapped\n- tablefilters.FilterTable:\n    filters:\n      A:\n      - 5\n      - 20000\n      error_x:\n      - 0\n      - 30\n      error_y:\n      - 0\n      - 30\n    inputName: colour_mapped\n    outputName: filtered_localizations\n- multiview.Fold:\n    input_name: filtered_localizations\n    output_name: folded\n- multiview.ShiftCorrect:\n    input_name: folded\n    output_name: shift_corrected\n    shift_map_path: 'pyme-cluster://villains/CALIBRATION/SHIFTMAPS/2021_02_11_reg.sf'\n- multiview.FindClumps:\n    input_name: shift_corrected\n    output_name: with_clumps\n    probe_aware: true\n    radius_offset: 150.0\n- multiview.MergeClumps:\n    input_name: with_clumps\n    output_name: clumped\n- multiview.MapAstigZ:\n    astigmatism_calibration_location: 'pyme-cluster://villains/CALIBRATION/ASTIGMATISM/2021_2_11_psf_beadconv110nm.am'\n    input_name: clumped\n    output_name: z_mapped\n    rough_knot_spacing: 80.0\n- localisations.ProcessColour:\n    input: z_mapped\n    output: color_mapped\n- localisations.DensityMapping:\n    colours:\n    - chan0\n    - chan1\n    inputLocalizations: color_mapped\n    manualXYBounds:\n    - 0.0\n    - 0.0\n    - 30000.0\n    - 30000.0\n    outputImage: hist\n    pixelSize: 40.0\n    renderingModule: Histogram\n    xyBoundsMode: manual\n- omero_upload.RGBImageUpload:\n    colorblind_friendly: false\n    filePattern: '{file_stub}.png'\n    input_image: hist\n    input_localization_attachments:\n      FitResults: FitResults\n      color_mapped: mapped\n    omero_dataset: '{Sample.Well}'\n    omero_project: '{Sample.SlideRef}'\n"}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/cluster/taskWorkerHTTP.py", line 341, in computeLoop
    recipe.execute()
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/base.py", line 718, in execute
    m.execute(self.namespace)
  File "/home/ubuntu/code/python-microscopy/PYME/recipes/localisations.py", line 291, in execute
    cached_output.mdh = output.mdh
AttributeError: can't set attribute
```

David-Baddeley, 2021-02-16:
Should hopefully already be fixed in #858

csoeller, 2021-02-16:
fixed in #858

closed 2021-02-16 by csoeller