python-microscopy/python-microscopy #867: [PROPOSAL] Distributed pyramid support

David-Baddeley, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
Pyramid creation (and subsequent operations on pyramids) are time consuming, especially for large tiled areas. It should be possible to store and process these in parallel ... There are currently a couple of barriers, the two most critical being: 
 
- pyramid creation requires modifying existing tiles which is not really compatible with the write-once nature of the cluster file system.
- data from the camera does not align 1:1 with tiles (due to both chosen tile sizes, and the need for overlap). This means that a) a single frame will touch multiple tiles and b) tiles will be touched by multiple different frames (due to overlap). This makes some operations intrinsically not thread / process safe (i.e. we need locking and/or some additional smarts).

Once created, and you are just reading tiles, everything should work fine with the current clusterIO.

A few subsidiary issues (easier):

- ensuring data locality for creation of higher levels in tile pyramid
- if division of tile into frames happens on the server, will need to touch data on multiple different nodes.

## pyramid creation I/O [server side]

There are a couple of solutions here:
### Add a custom endpoint to HTTPDataserver
as is done for the aggregation endpoints

**pros:**

- reasonably simple
- precedent in, e.g. __aggregate_h5
- keeps everything in one server
- easy to, e.g., update directory caches

**cons:**

- tiles are quite a lot larger (more overhead) than what is currently handled by aggregate endpoints
- significant processing is required when adding a frame (load impacted tiles, decompress, add new frame data and weights, compress, save). I'm more than a little concerned about putting significant processing in the dataservers for performance reasons.
- thread-safety - data server is multi-threaded (each request handled in a separate thread), tiling (by it's nature) is not really thread safe, as things might need to deal with multiple overlapping tiles. 

### Have a separate server specifically for tile pyramids

**pros:**

- keeps processing out of the data server
- can potentially be single-threaded to avoid concurrency issues

**cons:**

- another process to run and another protocol
- very hard to keep directory caches etc ... up to date
- processing still happens on the same CPU (you avoid the GIL, but still competing for memory and CPU resource)

### Hybrid
After writing the above I'm leaning towards a hybrid solution, outlined as follows:

- custom end point in dataserver
- endpoint drops incoming data onto a `queue.Queue`
- single thread pulls data from queue and does processing
    - could start out with a python implementation
    - completely c / cython -coded version of tile loading, addition, and saving **which releases the GIL** as fallback / end goal if in-process processing creates performance problems

## Pyramid creation [microscope side]
This likely needs a specific handler, not just clusterIO and/or spoolers.

- to ensure data locality when creating higher levels, we probably need to pre-assign data-servers to areas of the pyramid, and always send the corresponding tiles to that server.
- the nature of tiling, however, makes this a bit problematic with data throughput - we can't just divide the pyramid into 10 contiguous chunks and save to each in turn as we'll saturate the connection to that server while we are saving. We probably need smaller contiguous chuncks - e.g. 16, or 64 tiles.
- because of overlap, we need to do the separation of a frame into tiles on the microscope side, so that 2 servers never have to work on the same tile

For the initial levels, the distribution from the microscope dictates who is responsible for which chunk, but once we get to higher levels, the servers have to have some way of knowing which of the higher level chunks they are responsible for. This means we probably need to come up with a deterministic 'chunk distribution algorithm' which says, *given N servers, server x is responsible for chunk i at level j*.

David-Baddeley, 2021-02-17:
server distribution function could be quite simple:

```python
def server_distribution_function(x, y, z=0, chunk_shape=[8,8,1], n_servers=10):
    return (np.floor(x/chunk_shape[0]) + 
            np.floor(y/chunk_shape[1])*n_servers*.3 + 
            np.floor(z/chunk_shape[2]))%n_servers
```

gives
[image]
[image]

Tiles at higher levels above what we know will be present on one machine (i.e. the chunk level) are going to be hard to auto-generate/ update, although pragmatically if we tile up as far as we can with the, e.g. 8x8 tile chunks, we will have broken the back of the work, and might be able to generate the remaining levels in a single process later.

barentine, 2021-02-18:
@adrianhollow

barentine, 2021-02-23:
There are two issues at play I think - whether we construct the pyramid distributed vs. whether we can overlap its acquisition with it's construction remotely (but still potentially in a single process). Obviously the first would be cool, but I wonder if the second would be more bang for the buck (in terms of establishing what we need to more generally run recipes live).
What do you think @David-Baddeley about pushing events as we go (solving some other issues there, potentially to hdf using an aggregate endpoint) then releasing tasks (now per-frame in the original tiling series) for e.g. `recipes.supertile.Supertile`. Processing would still be a single machine, but it'd overlap with imaging time and would then be straightforward to use in a supertile/detect/queue recipe.

As to clusterIO overhead / data locality - I'm not sure how big of an issue that's going to be for tile series. If you're acquiring triggered as we do the bandwidth is much much lower. Most tiles are in the thousands of frames, biggest ones over 10,000, and often smaller xy frame sizes

David-Baddeley, 2021-02-23:
@barentine - I don't think that there is much advantage to separating the distributed and overlapped bits of the acquisition when a solution to to both at once would likely be easier, cleaner, and more scalable. Addressing specific points:

### Pyramid generation has a number of features which make it a poor fit for a recipe module:

- Recipes operate on images (or, in the future, chunks of images) which can fit comfortably in memory and can be read and written atomically.
- Recipe modules (other than output modules) should not have side effects. Output modules should put results atomically.
- Pyramid input and output are too large to fit in comfortably in memory
- Pyramid generation requires incremental tile updates (not atomic writes - at least for any reasonably sized pyramid)
- Because of the interdependencies - both between raw frames and tiles, and between layers of the pyramid), pyramids cannot be trivially chunked in a way which permits pyramid generation in memory. You are always going to need some special case recipe logic.

For these reasons, I don't think a) more generic support for running recipes live will really help pyramids and b) anything we do for pyramids will be that applicable to other recipe tasks. As discussed previously, I view the pyramid (and supertile) recipe modules as temporary hacks which will hopefully be rendered unnecessary once we have more complete tiling support. The are very much a square peg in a round hole.

### Pushing events in real time, rather than at series completion:

This has been tried in the past, and was a major bottleneck in the previous (`launchWorkers`) spooling scheme. The performance issues are inherent in the fact that this requires many small requests to the server, each incurring a little overhead. Although we could probably (eventually) address some of the issues, by e.g. batching the requests, this would be a lot of work and would have a high risk of introducing critical performance regressions along the way.

The more compelling reason for not doing this, however, is that the events are not really usable until they are all there and the series is complete. An analysis scenario which required looking at events before they were all complete would need to constantly poll the events file, holding analysis of frames until it was sure that all events which effected them had arrived. This would be complex, error-prone, and would significantly increase the load on the server holding the events file. For this reason, any analysis which requires events is effectively a non-starter for running before spooling is complete. 

The originally  proposed solution where the tile division is performed acquisition side would obviate the need for any event information in the processing.   
 
### Data locality

Whilst likely not strictly necessary for small, slowly-tiled series, this should be fairly simple and cheap to implement and gives a number of advantages.

- reduces load on dataservers (always good)
- sets things up so that subsequent analysis tasks (e.g. object detection, deconvolution, etc ...) which need to run distributed for performance will run with existing task scoring and allocation code, avoiding the need for special case distribution code.
- scalable,  so that we don't have to re-write again when we do want to tile a very large area or tile faster

adrianhollow, 2021-02-24:
@David-Baddeley I think I understand the principle of your proposed solution, though I don't know enough about the acquisition and cluster workflow to be 100% sure how to implement it. From what I know, pyramid creation is implemented in PYME/Analysis/tile_pyramid right now and dataserver endpoints are configured in PYME/cluster/HTTPDataServer, but I have no idea how to make 'specific handler' on the microscope side. Is there some context you can point me to?

barentine, 2021-02-24:
@adrianhollow , I suspect @David-Baddeley means something along the lines of the `SpoolController`/`HTTPSpooler`. At the moment, the `Tiler` is sort of a local handler, and as David mentioned an HTTPTiler or something like it to handle chunking things up/pushing them to servers intelligently

David-Baddeley, 2021-02-24:
@adrianhollow As things stand, pyramid creation is handled in `PYME.Acquire.Util.tiler.Tiler` (ignore the circular and multiwell tilers for now). This uses `PYME.Analysis.tile_pyramid` under the hood. `PYME.Acquire.Util.tiler.Tiler` is a simple handler for making a pyramid on disk, but it's completely unaware of the cluster. What I had in mind is something along following lines (function names etc not prescriptive, and missing loads and loads of hairy detail).

### Step 1: refactor `tile_pyramid.ImagePyramid.add_base_tile`

As it stands, the function is poorly named (it really should be `update_base_tiles_from_frame`). It probably makes sense to divide into 3 functions, e.g.: 

`update_base_tile()` - which updates a single base tile (basically the contents of the inner `for tile_y in tile_ys:` loop)

`frame_to_tiles()` - which subdivides a frame into individual base tiles

`update_base_tiles_from_frame()` - which is a drop in replacement for the current function and looks more or less like (pseudocode)

```python
def update_base_tiles_from_frame(self, frame, ...):
    tiles = self.frame_to_tiles(frame, ...)
    for tile in tiles:
        self.update_base_tile(tile, ...)
```

### Step 2: make cluster aware

Probably best done in a subclass

- `update_base_tile` moves behind a dataserver REST endpoint.
- `update_base_tiles_from_frame()` starts to look something like the following:

```python
def update_base_tiles_from_frame(self, frame, ...):
    tiles = self.frame_to_tiles(frame, ...)
    for tile in tiles:
        server = choose_server_for_tile(...) #use deterministic `server_distribution_function()` as introduced above
       # note that we are completely bypassing clusterIO here to enable deterministic distribution
        requests.put('http://server/__pyramid_update_base_tile/path/to/pyramid/tile/name', tile)
```

### Step 3: make `ImagePyramid.update_pyramid()` (which generates higher pyramid levels from base tiles) cluster aware

Here's where things get a bit hairy / uncertain. Initial thoughts as follows:

- have nodes build the pyramid during acquisition (much as is done for the current in-process pyramid creation) up to the level for which the chunking ensures data-locality - i.e. the 3rd level for 8x8 chunks.
- this should break the back of the work, allowing higher levels to be constructed quite quickly after all the data is there.
- subsequent levels could be easily done on a single node (at least in the first instance), but distributing the higher level generation as well would have advantages when it came to scheduling subsequent recipe tasks to operate on those tiles.
- this higher level generation would likely require an additional HTTP endpoint, and would require some form of synchronisation (i.e. all nodes would have to complete a level before anyone started on the next.
- Probably sensible to do higher level generation on a single node in the first instance.

### Step4: optimisations

- make tile weight calculation server-side (need to include some hints as to where in the frame the tile came from as parameters to `update_base_tile`) to avoid transmitting extra weight data
- support sending partial tiles and zero-padding on receipt rather than in `frame_to_tiles()`
- persistent sessions, request batching, higher performance socket IO (see, e.g. `PYME.Acquire.HTTPSpooler` and `clusterIO`)

David-Baddeley, 2021-02-24:
PS - also ignore protocol based tiling

adrianhollow, 2021-02-24:
Thank you both, that helps a lot! I'll try to go from here then

adrianhollow, 2021-03-05:
I've implemented the changes to the pyramids in [#929](https://github.com/python-microscopy/python-microscopy/pull/929) and tested via PYMEClusterOfOne and running the `main` of `distributed_pyramid.py`. However, distributed pyramid creation seems two orders of magnitude slower than normal pyramid creation, especially while updating the base tiles. I'm assuming this is because of data transfer via network plus data conversion to json/bytes for the network transfer.

@David-Baddeley , how close is this to what you imagined? I just wanted to know before I start streamlining the process if I'm missing something else.

David-Baddeley, 2021-07-13:
With #929 in we can probably close, and address any remaining niggles as bugs.

closed 2021-07-13 by David-Baddeley