python-microscopy/python-microscopy #876: [BUG] can't turn off PYMEImage crosshairs + IndexError hangs gui til you click back in

barentine, 2021-02-19:
**Describe the bug**
I don't know how to turn off the crosshairs in PYMEImage, which is annoying because I'd like to export a png without them (also without the 2.0 um written under the scalebar).

Additionally, if you click outside of the image, then maybe resize, or click something like changing a LUT, nothing happens until you click back in (other than the following traceback)
```
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 784, in OnLeftUp
    self.OnSetPosition(event)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/arrayViewPanel.py", line 856, in OnSetPosition
    self.do.xp =int(pos[0]/sc)
  File "/Users/Andrew/code/python-microscopy/PYME/DSView/displayOptions.py", line 130, in xp
    raise IndexError('X position out of bounds')
IndexError: X position out of bounds

```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] osx 10.15
 - Python version: 3.7
 - numpy version: 1.19
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2021-02-24:
Workaround: 
```
view.crosshairs=False
view.showScaleBar=False
```

David-Baddeley, 2021-02-24:
Fix to the `IndexError` bug in #887. 

Turning this into an enhancement and renaming for a GUI way of turning off the crosshairs etc ... This has been on the TODO list for a while.

closed 2021-11-24 by David-Baddeley