python-microscopy/python-microscopy #879: [BUG] can't run longer recipes in PYMEVis without bus error or segfault

barentine, 2021-02-20:
**Describe the bug**
If I try to develop a recipe interactively in PYMEVis, everything is grand up until I start running anything that takes a bit more time. If it takes a long time and throws an error in my code, I see the error Traceback and then a segfault. If my code doesn't hit an error but still takes awhile, I see a bus error
<details>

```
(base) Andrews-MacBook-Pro-5:htsmlm-2021 Andrew$ PYMEVis ~/Downloads/mapped.hdf
Trying to load 3rd party recipe module pyme_omero.recipe_modules.omero_upload
Loaded 3rd party recipe module pyme_omero.recipe_modules.omero_upload
Trying to load 3rd party recipe module pyme_rcnn.recipe_modules.rcnn_segment
ERROR:PYME.recipes.modules:Error loading plugin: pyme_rcnn.recipe_modules.rcnn_segment
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/modules.py", line 42, in <module>
    m = __import__(mn, fromlist=mn.split('.')[:-1])
  File "/Users/Andrew/code/pyme-mrcnn/pyme_rcnn/recipe_modules/rcnn_segment.py", line 6, in <module>
    from mrcnn.config import Config
ModuleNotFoundError: No module named 'mrcnn'
Trying to load 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
Loaded 3rd party recipe module quant_condensate.recipe_modules.cluster_quant
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'
quaternion module not found, disabling custom clip plane orientations
quaternion module not found, disabling custom clip plane orientations
New Canvas
DEBUG:PYME.LMVis.visCore:Creating VisGUI menu bar
DEBUG:PYME.LMVis.Extras:Initializing QPobjectSegment plugin
DEBUG:PYME.LMVis.Extras:Initializing animation plugin
DEBUG:PYME.LMVis.Extras:Initializing clusterAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing dockedPanel plugin
DEBUG:PYME.LMVis.Extras:Initializing extra_layers plugin
DEBUG:PYME.LMVis.Extras:Initializing fiducial plugin
DEBUG:PYME.LMVis.Extras:Initializing flagZMotionArtifacts plugin
DEBUG:root:Adding menu items for event filters
DEBUG:PYME.LMVis.Extras:Initializing fusionUtils plugin
DEBUG:PYME.LMVis.Extras:Initializing multiColourAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing multiviewMapping plugin
DEBUG:root:Adding menu items for multi-view manipulation
DEBUG:PYME.LMVis.Extras:Initializing objectMeasurements plugin
DEBUG:PYME.LMVis.Extras:Initializing particleTracking plugin
DEBUG:PYME.LMVis.Extras:Initializing perFrameVariable plugin
DEBUG:PYME.LMVis.Extras:Initializing photophysics plugin
DEBUG:PYME.LMVis.Extras:Initializing pointSetGeneration plugin
DEBUG:PYME.LMVis.Extras:Initializing pointwiseColoc plugin
DEBUG:PYME.LMVis.Extras:Initializing saved_views plugin
Saved Views are deprecated -  use the ones with animation instead and remove me!
DEBUG:PYME.LMVis.Extras:Initializing scaleBox plugin
DEBUG:PYME.LMVis.Extras:Initializing sequentialSpecies plugin
DEBUG:PYME.LMVis.Extras:Initializing setBGImage plugin
DEBUG:PYME.LMVis.Extras:Initializing shiftmapGenerator plugin
DEBUG:PYME.LMVis.Extras:Initializing showShortcuts plugin
DEBUG:PYME.LMVis.Extras:Initializing snapshot plugin
DEBUG:PYME.LMVis.Extras:Initializing spherical_harmonics plugin
DEBUG:root:Adding menu items for spherical harmonic (shell) fitting
DEBUG:PYME.LMVis.Extras:Initializing subsets plugin
DEBUG:PYME.LMVis.Extras:Initializing surface_fitting plugin
DEBUG:PYME.LMVis.Extras:Initializing temperaturePlot plugin
DEBUG:PYME.LMVis.Extras:Initializing vibration plugin
ERROR:PYME.LMVis.Extras:Error loading plugin: #package_name.visgui_modules.NAME_OF_FILE_WITHOUT_EXTENSION
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/LMVis/Extras/__init__.py", line 46, in InitPlugins
    m = __import__(mn, fromlist=mn.split('.')[:-1])
ModuleNotFoundError: No module named '#package_name'
DEBUG:root:Adding menu items for OMERO loading
Creating fold panel
GenPanels
Creating datasource panel
RecipeView._layout
Input datasources:  []
/Users/Andrew/code/python-microscopy/PYME/LMVis/filterPane.py:111: wxPyDeprecationWarning: Call to deprecated item. Use InsertItem instead.
  ind = self.lFiltKeys.InsertStringItem(ind+1, key)
/Users/Andrew/code/python-microscopy/PYME/LMVis/filterPane.py:112: wxPyDeprecationWarning: Call to deprecated item. Use SetItem instead.
  self.lFiltKeys.SetStringItem(ind, 1, '%3.2f' % value[0])
[... 423 lines cut ...]
/Users/Andrew/code/python-microscopy/PYME/LMVis/filterPane.py:112: wxPyDeprecationWarning: Call to deprecated item. Use SetItem instead.
  self.lFiltKeys.SetStringItem(ind, 1, '%3.2f' % value[0])
/Users/Andrew/code/python-microscopy/PYME/LMVis/filterPane.py:113: wxPyDeprecationWarning: Call to deprecated item. Use SetItem instead.
  self.lFiltKeys.SetStringItem(ind, 2, '%3.2f' % value[1])
Input datasources:  ['_color_mapped', 'FitResults']
Bus error: 10

```

</details>

**To Reproduce**
Steps to reproduce the behavior:
1. update PYME to head, setup.py develop from your macos framework build
2. PYMEVis mydata.hdf -r myrecipe.yaml (or open it, copy paste into the text, apply text, or add it one module at a time, issue persists regardless)
4. See error

**Expected behavior**
PYMEVis not trip over itself

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] osx10.15
 - Python version: 3.7.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

barentine, 2021-02-20:
note that my recipe runs fine on the same data when executed in the bakeshop

David-Baddeley, 2021-02-20:
Would be useful to have a bit more info here...as it stands there is not enough to make useful conclusions about the problem

- What recipe/recipes?
- dimensions and type of input data
- what runtime (mins, longer?)

Ultimately were looking for a minimum set of conditions that reliably reproduce the error. 
A bus error/segfault points pretty strongly to an error in compiled code, and suggests that it could be a bug in one of our dependencies.

barentine, 2021-02-21:
<details> <summary> recipe </summary>

```
- localisations.AddPipelineDerivedVars:
    inputEvents: ''
    inputFitResults: FitResults
    outputLocalizations: Localizations
- localisations.ProcessColour:
    input: Localizations
    output: colour_mapped
- tablefilters.FilterTable:
    filters:
      A:
      - 5
      - 20000
      error_x:
      - 0
      - 30
      error_y:
      - 0
      - 30
    inputName: colour_mapped
    outputName: filtered_localizations
- localisations.ExtractTableChannel:
    channel: chan0
    inputName: filtered_localizations
    outputName: chan0
- surface_fitting.SphericalHarmonicShell:
    input_name: chan0
    output_name: harmonic_shell
    output_name_mapped: shell_mapped
- surface_fitting.ImageMaskFromSphericalHarmonicShell:
    input_image_bound_source: chan1
    input_shell: harmonic_shell
    output: nucleus_mask
- localisations.ExtractTableChannel:
    channel: chan1
    inputName: filtered_localizations
    outputName: chan1
- pointcloud.Ripleys:
    inputMask: nucleus_mask
    inputPositions: chan1
    nbins: 25
    outputName: ripleys
    threaded: true
    three_d: true
- output.HDFOutput:
    filePattern: '{output_dir}/{file_stub}_ripleys.hdf'
    inputVariables:
      harmonic_shell: shell
      ripleys: ripleys

```

</details>

which uses one module not in main (in PR #880), though I'm not sure if that's strictly necessary.

Just ran again without that module (so no mask on ripleys) and got an alert that I'd exhausted all system memory (was into swap for sure, at 120 GB used by PYMEVis alone)

barentine, 2021-02-21:
[example_hdf.zip](https://github.com/python-microscopy/python-microscopy/files/6017862/example_hdf.zip)

David-Baddeley, 2021-02-21:
If you're using 120GB, then there's probably also a memory leak in play. From the recipe, I'd initially suspect Ripleys  - at least if it is our error (the only module in the recipe which uses compiled code). Might be worth trying to run single threaded (threading=False) as there's a chance that a) we're not doing our reference counting properly and b) that this gets triggered by a race condition when we have released the GIL.

barentine, 2021-03-01:
works fine in a different conda environment. Sorry for the time sink David

closed 2021-03-01 by barentine