python-microscopy/python-microscopy #881: [DISCUSSION/BUG]: What is our z reference frame for localizations?

barentine, 2021-02-21:
Basically I'm hoping we can resolve this fixme in renderers
```
#record the pixel origin in nm from the corner of the camera for futrue overlays
ox, oy, oz = MetaDataHandler.origin_nm(mdh)
if not imb.z0 == 0:
    # single plane in z stack
    # FIXME - what is z for 3D fitting at a single focal plane? Check for pipeline['focus']==0 instead?
    oz = 0

mdh['Origin.x'] = ox + imb.x0
mdh['Origin.y'] = oy + imb.y0
mdh['Origin.z'] = oz + imb.z0
```

x, y origin is ulc of active camera chip ROI, and the chip offset is recorded so we can map back to physical space even if we change roi etc. Renderers set their `Origin` metadata to account for any additional offset. Cool.

z origin from `MetaDataHandler.origin(mdh)` is (for a `Time/Blinking series` acq) is `Positioning.z`, or wherever the piezo was when you started that acquisition. Problem is that this is not reflected in the actual localization 'z', which is referenced from piezo origin. We special-case our way out of it in renderers so you get `Origin.z: 0`, but if you actually try and use any of this together, e.g. to ripley's from a 3D image mask, you'll run into a wall because even if you have image bounds / origins that overlap nicely between your mask (`image`, below) and your points (`mapped`)
```
>>> MetaDataHandler.origin_nm(mapped.mdh)
(21160.0, 103040.0, 49971.3)
>>> image.origin
(20870.058197021484, 104417.21313476562, 48183.455455455456)
>>> image.imgBounds.bounds
(-290.0, 1377.0, 29860.0, 30252.0, 48183.0, 54108.0)
>>> image.imgBounds.estimateFromSource(mapped).bounds
(-289.9418, 1272.7356, 29720.553, 30145.607, 48183.455455455456, 54111.455455455456)
```
you only have xy + origin lining up between the two and you need to ignore the z origin for one to get things to line up (which in ripleys.py we don't, see line 126. @zacsimile I'm guessing this works for imported 4Pi data though?).

I'm not sure what I'd like to see differently. XY roi based localizations and recording the origin makes sense. For Z it's a bit weirder to think of a difference between physical / ROI coordinates, which is probably why we've fallen into this halfway point, but I think for consistency it would make most sense to subtract off Positioning.z from 'focus' during pipeline mapping, but for continuity of people working with their data as usual maybe the preference would be to change `MetaDataHandler.origin`'s return for z to be zero even if there's a `Positioning.z` in the metadata?

barentine, 2021-02-21:
Note that our origin handling changed on 4e2ae22 in #202, which introduced z not returning 0 (for my data anyway) which also breaks `coordinate_tools.pixel_index_of_points_in_image` and consequently `localisations.LabelsFromImage`.

David-Baddeley, 2021-02-24:
### Definitive status quo:

- **pixel co-ordinates in a stack are referenced to stack starting position**
- **localisations are referenced to piezo 0**
- `MetaDataHandler.origin` returns the origin of the **pixel** data - i.e. `Positioning.z`. It cannot return 0 if we have any information to determine the pixel origin, not the localisation data.
- origin handling did not change in #202 (we simply fixed an omission that had crept in when changing to more state-based control in PYMEAcquire)
- `coordinate_tools.pixel_index_of_points_in_image` is buggy (note the original, non-refactored version of this code @barentine removed in 696f595bd5419a48ad77b96206ecf174e162059f worked correctly and I should have caught this at the time).

We should make sure all existing code matches the above (ie fix the bug in Ripleys fixes) with some urgency, and can look at what the best option moving forward is later. Even if not super elegant, there are some arguments for the status quo. Although `MetaDataHandler.origin` can't change, we could definitely look at improving the docs and/or adding a helper function to get the localisation origin to reduce the number of places we can make mistakes.

David-Baddeley, 2021-02-24:
### Arguments for the status-quo:

- an absolute origin (as for localisation z coordinates) is nice when possible.
- This is possible for z as the piezo is accurate, and interpretation is simple (no complications like ROIs etc...)
- this is not really possible for images (pixel coordinates are relative and quantized by default)
- It could theoretically be done for localisation x,y coordinates but has pragmatic issues including stage precision which is generally worse than localisation precision.
- certain scanning regimes make a fixed origin for z less pragmatic (you don't necessarily have a clearly defined 'bottom' slice)
- if we did reference z localisations to a particular slice, we'd get odd ranges of z values when doing 3D localisation (they would include values at a -ve offset from the bottom slice or origin). This is not a major problem, but is not super intuitive.

barentine, 2021-02-25:
closing this, though converting some of it to docs might not be a bad idea

closed 2021-02-25 by barentine