python-microscopy/python-microscopy #890: [ENHANCEMENT] Automatically select text in editable boxes when I hit tab

zacsimile, 2021-02-24:
For example, if I use the hist limits editor and open the bounds, I want to edit them. It would be nice if when I hit tab the bound number was selected so I just have to type a number to edit it, rather than selecting it first.

Currently this:
[image]

Want this:
[image]

David-Baddeley, 2021-02-24:
I suspect is going to be pretty hard to do without hacking each and every text control (you'd need to handle `wx.EVT_SET_FOCUS` on the control and select the text in the handler). A quick google suggests that there might be timing issues with this and you'd need to either send yourself another event, or use, e.g. `wx.CallAfter` to actually set the selection.

David-Baddeley, 2021-02-24:
We could potentially subclass `wx.TextCtrl` and replace all our hard-coded uses, but traits etc ... would still use the native control.

zacsimile, 2021-02-24:
I figured. I just wanted to put this down as something that would be nice to have in case we ever get around to it. Definitely low, low priority.

David-Baddeley, 2021-12-05:
Just added the won't fix label, as I don't see any clean way of doing it - particularly not in a cross-platform manner.

closed 2023-09-18 by David-Baddeley