python-microscopy/python-microscopy #894: [DISCUSSION] ok that CameraInfoManager.getDarkMap(mdh) bombs if mdh contains filename not present on system?

csoeller, 2021-02-25:
## Discussion if behaviour described below is intended or not
Note this is not a bug report per se but a question if an issue encountered should come up like that or not. Looking for genuine discussion if I am doing the right or wrong thing.

**Describe the issue**
I am trying to retrieve the darkmap for an image given its metadata. This is in a plugin that I have written.
I use a `CameraInfoManager` object from remFitBuf.py in my code and do somethink like:

```python
   ci = CameraInfoManager()
   darkf = ci.getDarkMap(mdh2)
```

When I do this with an image that has a file name for the darkmap, but that was acquired on a different machine, e.g.

```
DarkMapID: F:\Data\CALIBRATION\VSC-02858\dark_100ms.tif
```

and open that, say on my mac, I get an error about this file not existing, full error at bottom.

## Questions

1. Is `CameraInfoManager()` the right object to provide the API or has this changed?
2. If still the right API, is it intended behaviour for the call to fail, or should the APIU at least provide an option like `ignoreMissingFile=True` or similar?

**To Reproduce**

See description above.

**Expected behavior**
Not sure.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**

Latest PYME running on 2.7.x

### Appendix - Full error

```
  File "/Users/csoe002/Documents/src/PYME-src/PYME-extra/PYMEcs/experimental/mapTools.py", line 475, in OnPhotonConvert
    darkf = self.ci.getDarkMap(mdh2)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/localization/remFitBuf.py", line 216, in getDarkMap
    mp = self._getMap(md, darkMapName)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/localization/remFitBuf.py", line 189, in _getMap
    full_map = self._fetchMap(md, mapName)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/localization/remFitBuf.py", line 174, in _fetchMap
    map = ImageStack(filename=fn, haveGUI=False).data[:,:,0].squeeze() #this should handle .tif, .h5, and a few others
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/image.py", line 238, in __init__
    self.Load(filename, prompt=load_prompt, haveGUI=self.haveGUI)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/image.py", line 1054, in Load
    self._loadTiff(filename)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/image.py", line 796, in _loadTiff
    mdfn = self._findAndParseMetadata(filename)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/image.py", line 690, in _findAndParseMetadata
    tf = TIFFfile(filename)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/contrib/gohlke/tifffile.py", line 747, in __init__
    self._fd = unifiedIO.openFile(filename)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/IO/unifiedIO.py", line 211, in openFile
    raise IOError('File does not exist or URI not understood: %s' % filename)
IOError: File does not exist or URI not understood: /Users/csoe002/PYMEData/F:/Data/CALIBRATION/VSC-02858/dark_100ms.tif
```

David-Baddeley, 2021-02-25:
`CameraInfoManager` in remFitBuf was only ever designed to be used within fits, in which case a hard fail on a missing map is the correct and desirable behaviour. Any other usage is currently "off-label". That said, I can certainly see an argument that the functionality would be useful in a broader context, and that `CameraInfoManager` could be refactored/adapted to support that.

Even with "off-label" usage, I'm not sure that raising an exception for a missing map is the wrong behaviour - if a map is specified in the metadata, I think it's reasonable to assume that the user wants to use that map, and that getting the map should fail if it's not present (rather than silently falling back on the defaults). If a map is specified in the metadata I think that any fallback should be handled explicitly by the caller.

One potential solution would be to refactor the defaults into functions which could be called externally, so that your code could, e.g. do:

```python
ci = CameraInfoManager()
try:
    darkf = ci.getDarkMap(mdh2)
except IOError:
    logger.exception('Dark map specified but not found, falling back on defaults')
    darkf = ci.get_dark_default(mdh2)
```

David-Baddeley, 2021-02-25:
In a more general API context, we should probably always warn when we are falling back on the defaults (although this is probably not wanted in the fits as it would be very spammy).

csoeller, 2021-02-25:
Great, this makes good sense.

I'd welcome that we have some API interface to maps that it ideally won't change in completely unexpected ways.

The code snippet that you suggest looks very good, I believe we only need to add the `get_XXX_default` methods, right?

```python
ci = CameraInfoManager()
try:
    darkf = ci.getDarkMap(mdh2)
except IOError:
    logger.exception('Dark map specified but not found, falling back on defaults')
    darkf = ci.get_dark_default(mdh2)
```

csoeller, 2021-02-26:
I have now a quick implementation outside of all modules for testing as below.

Quick discussion about the approach:

1. I let the get_default methods search in the camera calibration dir first, as that would be the preferred default
2. This brings a dependency on PYME.Analysis.gen_sCMOS_maps
3. Is that dependeny ok in remFitBuf.py?

Once we have discussed the points above I will integrate into the relevant PYME code and create a pull request.

### External test implementation

```python
import os
import PYME.Analysis.gen_sCMOS_maps as gmaps

import logging
logger = logging.getLogger('main')

def _getDefaultMap(ci,mdh,maptype = 'dark'):
    # logic: try map in default calibration dir first
    #        if not present just return default value
    mapname = gmaps.mkDefaultPath(maptype, mdh, create=False)
    mdh2 = NestedClassMDHandler(mdh)
    logger.debug("checking map file %s" % mapname)
    
    if maptype == 'dark':
        id = 'Camera.DarkMapID'
    elif maptype == 'variance':
        id = 'Camera.VarianceMapID'
    elif maptype == 'flatfield':
        id = 'Camera.FlatfieldMapID'
    else:
        raise RuntimeError('unknown map type %s' % type)

    if os.path.isfile(mapname):
        mdh2[id] = mapname
    else:
        mdh2[id] = ''
        logger.debug("map file %s does not exist" % mapname)

    if maptype == 'dark':
        return ci.getDarkMap(mdh2)
    elif maptype == 'variance':
        return ci.getVarianceMap(mdh2)
    elif maptype == 'flatfield':
        return ci.getFlatfieldMap(mdh2)

def get_dark_default(ci,mdh):
    return _getDefaultMap(ci,mdh,maptype = 'dark')

def get_variance_default(ci,mdh):
    return _getDefaultMap(ci,mdh,maptype = 'variance')
    
def get_flatfield_default(ci,mdh):
    return _getDefaultMap(ci,mdh,maptype = 'flatfield')

# Example use
try:
    darkf = ci.getDarkMap(nr.mdh)
except IOError:
    logger.exception('Dark map specified but not found, falling back on defaults')
    darkf = get_dark_default(ci,nr.mdh)
```

closed 2023-12-09 by csoeller