python-microscopy/python-microscopy #898: [discussion/bug/enhancements] events pushing / recipe rule pushing / timing / retries on failed recipes

barentine, 2021-02-26:
**Describe the thing**
We've gone through some robustness improvements lately with the timing of signals/flags/etc in HTTPSpooler, which has helped a lot, but I still timed out 2/96 of the last detections on the first run through (they auto-retriggered fresh overviews, which is great, but obviously we'd be even more robustness if we didn't need that) because the recipes were looking for events files before they were there.

Notably I connect to the onSpoolStop signal to push rules, which currently gets sent before the new `finalise`. One of the things I was going for in #850, which as @David-Baddeley mentioned had it's own potential flaw, was to send the spoolstop signal after the events have hit the cluster which is our current mark of completion. There's now a TODO in Spooler about whether to call finalise before the signal instead, which I'll PR shortly as I think that would be grand. 

I did additionally want to just toss out there a couple other thoughts about events though, without taking control of the pyramid distribution discussion. If we aren't OK pushing events in batches as we go in the same way we push frames in groups as we do, I think we'll limit ourselves in terms of how 'smart' we can be. For example, I want to get the z height right at every ROI because I messed it up last week and spoiled a dataset (shouldn't have been imaging so late in the evening I guess). Anyway, in our currently framework, my options are to a) run a widefield protocol chained to a height-determining recipe which also queues back SMLM, b) image one z cycle in SMLM chained to a recipe to figure out whether we got the right height and requeue the remaining cycles with different stack settings (which notably would also take some stacksettings infrastructure) or c) just hack that stacksettings infrastructure into my initscript, i.e. don't really do anything in any framework. If we pushed events as we went though, and some recipe/protocol when can i release this task magic, we could adjust on the fly based on where we actually get localizations. Obviously this sort of feedback ability would be helpful potentially for 405 / other laser intensities as well. 

So:
- I'll sub a PR to swap spoolstop signal and finalise order
- I'll maybe sub a PR to retry on clusterPZFdatasource event searching
- things like #852 would be nice
-  longer term I am wondering what our vision is for 'smart'ness if we aren't going to push events as we go

David-Baddeley, 2021-02-28:
There's lots of different and only tangentially related bits in here, I'm going to focus on the event spooling, why spooling as you go is not going to be really helpful, and how this relates to feedback/smartness.

### PYME prioritises speed / throughput over latency

Although this in not immediately intuitive, high speed and low latency often place competing demands. As a general rule PYME prioritises speed over latency. e.g.:

- running the camera in continuous (untriggered) mode and processing protocol tasks in a semi-synchronised fashion
- batching frames in the spooler and ruleservers to ensure data locality introduces latency
- distributed analysis intrinsically incurs latency (e.g. http round-trips) 

Many of the optimisations we've made recently (e.g. increasing timeouts) come with an implicit acceptance of potentially higher latency. In general the design philosophy has been that cluster analysis tasks should have a latency roughly <= the length of an acquisition (i.e. you have the results of the previous acquisition before the current one is finished).

For simple analysis tasks at low load, the latency is typically in the 1-10s range. There is a hard 1s lower bound on latency due to clusterIO cache persistence, and there is no upper bound on latency for expensive computational tasks or when the cluster is busy.

### Latency makes cluster-derived feedback *within* a series impractical

With a 10s acquisition time (@800Hz), a few seconds latency would mean that you couldn't really expect to update, e.g. z-stepping settings within a series. Because the latency is unbounded, any mechanism which relied on cluster feedback within a series would not be robust, even if we did reduce cluster latency (at the expense of overall throughput). In practice this means that queued actions should be regarded as "atomic", non-modifiable operations (at least in the cluster context)

The "smart" imaging scenario in which we poll live cells and queue acquisitions if we see something interesting is (mostly) robust with current cluster latencies as the process we are looking at must be slow compared with the imaging time (and latency) for us to expect to capture it in any case, and we have a robust means of expiring queued actions if a given latency is exceeded (resulting in the stochastic dropping of a few events).   

### Events are very flexible, but have too much overhead to read/process on a per-frame basis

- The nature of events (text description which needs to be parsed, values for a frame can depend on all previous values, need to read all events in sequence to get to the one(s) which determine values for the current frame) means that they are relatively expensive to process (probably higher than the cost of localising a single frame). This is fine when it's done once per-sequence, but problematic when done once per frame.
- Due to the lack of hard synchronisation between events and frames, a realtime consumer has no guarantee that the events file will contain all events relevant to the current frame when they receive the frame. They either have to poll the event data until they see an event with a frame number > than the frame they are processing (which would result in an unacceptable load on servers) or they need to introduce a delay sufficient to ensure that any events they would be interested in have been written prior to reading the event data (extra latency).  
- if you were doing something on a per-frame basis you'd need a non-events based way of, e.g. determining z position. This could be, e.g. using a slower triggered, synchronous acquisition so that you could map frame # to z without events.

### How to address the adaptive stack bounds problem

Now enough of what can't be done, how about some solutions to the actual problem. As I see it there are 3 possible approaches:
- a 3D / 2.5D overview so that you can estimate z depths along with x/y positions
- an in-process adaptive z stack function which used a cheap, low-latency, approximate metric for the number of localisations (e.g. comparing image std. deviation to what you'd expect for Poisson noise with the same mean value)
- widefield z stack then localisation image. Bounds would either need to be computed from z stack in-process, or you'd need an explicit wait on the cluster. Probably the hardest to implement.

barentine, 2021-03-01:
Thanks for all of the detailed thoughts here David - lots of good points. I actually took a crack at the third option you suggested this weekend which 'works' with the exception of the actual metric (sort of important). The workflow I tried was tiling at nice 11, chained to a recipe to queue widefield tasks at nice 9, chained to a recipe to queue SMLM tasks at nice 10. I think what that suffers from is the precision you need to set the bounds on a per-cell basis is maybe non-trivial for a widefield stack (depends on what you're looking at maybe). An alternative which might be aided by higher z resolution would be to take a quick SMLM cycle in z, look at were you actually get localizations, and then requeue a longer SMLM stack at the same location and 'drift correct' them together later if you want. 

Anyway, I am dropping the push for now as I think we both know I need to push hard on other things, but after messing up the z bounds for an entire plate last week I took a quick crack at it. Regardless I appreciate the conversation about events and generally adaptive imaging.