python-microscopy/python-microscopy #903: [BUG]

adrianhollow, 2021-02-27:
**Describe the bug**
Using the option "Spool to h5 on cluster" in PYMEAcquire leads to file error on the cluster.

**To Reproduce**
1. Start PYMEClusterOfOne and PYMEAcquire.
2. Select `simul488.py` protocol, "Z stepped" and "Spool to h5 on cluster"
3. Click "Start"
4. See error

**Expected behavior**
Successful save to h5 on the cluster.

**Screenshots**
[image]
Stacktrace from PYMECluster:
```
----------------------------------------
DEBUG:root:pytables open call: /home/tphan/PYMEData/tphan/2021_2_27/27_2_series_A.h5
HDF5-DIAG: Error detected in HDF5 (1.10.4) thread 139755500361472:
  #000: H5F.c line 509 in H5Fopen(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #001: H5Fint.c line 1400 in H5F__open(): unable to open file
    major: File accessibilty
    minor: Unable to open file
  #002: H5Fint.c line 1615 in H5F_open(): unable to lock the file
    major: File accessibilty
    minor: Unable to open file
  #003: H5FD.c line 1640 in H5FD_lock(): driver lock request failed
    major: Virtual File Layer
    minor: Can't update object
  #004: H5FDsec2.c line 941 in H5FD_sec2_lock(): unable to lock file, errno = 11, error message = 'Resource temporarily unavailable'
    major: File accessibilty
    minor: Bad file ID accessed
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 45646)
Traceback (most recent call last):
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/socketserver.py", line 650, in process_request_thread
    self.finish_request(request, client_address)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/http/server.py", line 646, in __init__
    super().__init__(*args, **kwargs)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "/home/tphan/thesis/python-microscopy/PYME/cluster/HTTPDataServer.py", line 485, in do_GET
    f = self.send_head()
  File "/home/tphan/thesis/python-microscopy/PYME/cluster/HTTPDataServer.py", line 570, in send_head
    return self.list_h5(path)
  File "/home/tphan/thesis/python-microscopy/PYME/cluster/HTTPDataServer.py", line 657, in list_h5
    with h5File.openH5(path.rstrip('/').rstrip('\\')) as h5f:
  File "/home/tphan/thesis/python-microscopy/PYME/IO/h5File.py", line 29, in openH5
    h5rFile.file_cache[key] = H5File(filename, mode)
  File "/home/tphan/thesis/python-microscopy/PYME/IO/h5rFile.py", line 68, in __init__
    self._h5file = tables.open_file(filename, mode)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/tables/file.py", line 315, in open_file
    return File(filename, mode, title, root_uep, filters, **kwargs)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/tables/file.py", line 778, in __init__
    self._g_new(filename, mode, **params)
  File "tables/hdf5extension.pyx", line 492, in tables.hdf5extension.File._g_new
tables.exceptions.HDF5ExtError: HDF5 error back trace

  File "H5F.c", line 509, in H5Fopen
    unable to open file
  File "H5Fint.c", line 1400, in H5F__open
    unable to open file
[... 301 lines cut ...]
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/multiprocessing/pool.py", line 121, in worker
    result = (True, func(*args, **kwds))
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/multiprocessing/pool.py", line 44, in mapstar
    return list(map(*args))
  File "/home/tphan/thesis/python-microscopy/PYME/IO/clusterIO.py", line 248, in _listSingleDir
    r = s.get(url, timeout=timeout)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/requests/sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/requests/adapters.py", line 498, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Ubuntu 20.04.1 LTS
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - pytables version: 3.6.1

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-02-28:
I suspect this is a HDF library issue, likely specific to your version of linux and / or the installed version of libhdf5. Notably I cannot reproduce on Ubuntu 18.04.5, Python 3.7.7 (or on OSX).

David-Baddeley, 2021-02-28:
@adrianhollow - Could you double check that this is reproducible on your system. 

The alternative option to a libhdf5 bug is that, e.g., there had been a previous attempt to write to that file which had resulted in a hard failure (e.g. via a segfault) leaving a stale HDF lock in place. This would be expected to happen very rarely.

adrianhollow, 2021-03-01:
Curiously, I can't reproduce this error today. I'm not aware of changing anything except switching git branches, though I took careful note that the error only appeared when I pulled the latest commits from `master`. I'll check if I made any other changes and try again, but it seems very strange to me.

David-Baddeley, 2021-03-01:
The other option is a race condition when the file timed out and was closed followed by a write which tried to open it before closing had fully finished.