python-microscopy/python-microscopy #917: [BUG] menuinst doesn't work on conda install >=py37

zacsimile, 2021-03-02:
**Describe the bug**
```
Traceback (most recent call last):
  File "C:\Users\PYME_test2\miniconda3\lib\site-packages\conda\gateways\disk\create.py", line 245, in make_menu
    import menuinst
  File "C:\Users\PYME_test2\miniconda3\lib\site-packages\menuinst\__init__.py", line 23, in <module>
    from .win32 import Menu, ShortCut
  File "C:\Users\PYME_test2\miniconda3\lib\site-packages\menuinst\win32.py", line 19, in <module>
    from .winshortcut import create_shortcut
ModuleNotFoundError: No module named 'menuinst.winshortcut
```

Appears to be related to https://github.com/ContinuumIO/menuinst/issues/16.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Windows10
 - Python version: 3.7.10
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

barentine, 2022-09-19:
Not that when uninstalling the executable installer I see in the details:
```
Deleting python-microscopy menus...
Error: install() got an unexpected keyword argument 'root_prefix'
Traceback:
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\Lib\_nsis.py", line 129, in mk_menus
    menuinst.install(shortcut, remove, prefix=prefix,
TypeError: install() got an unexpected keyword argument 'root_prefix'

Failed to process C:\Users\aesb\AppData\Local\python-microscopy\Menu\menu-windows.json...
Execute: "C:\Users\aesb\AppData\Local\python-microscopy\pythonw.exe" -E -s "C:\Users\aesb\AppData\Local\python-microscopy\Lib\_nsis.py" pre_uninstall
Execute: "C:\Users\aesb\AppData\Local\python-microscopy\pythonw.exe" -E -s "C:\Users\aesb\AppData\Local\python-microscopy\Lib\_nsis.py" rmpath
Execute: "C:\Users\aesb\AppData\Local\python-microscopy\pythonw.exe" -E -s "C:\Users\aesb\AppData\Local\python-microscopy\Lib\_nsis.py" rmreg
Removing files and folders...

```

I see we pinned 1.4.16, but in the executable installer I get: 1.14.18
[image]

and then I see the same error thrown on installation

[image]

barentine, 2022-09-19:
Why is our build config for pyme-depends-strict only on py 3.6 and then we're packaging 3.7 installers?

barentine, 2022-09-19:
Issue is presumably that we don't pin menuinst version in pyme-conda-recipes/installers/construct.yaml

David-Baddeley, 2022-09-19:
We don't use pyme-depends-strict in building the installers (or pretty much for anything now - it's a relic of trying to offer a failsafe option for the PYMEVis paper). A standard conda install on windows (`conda create -n PYME python=3.7 python-microscopy`) worked fine for me last week.

barentine, 2022-09-22:
Current automated build installer 2022.09.20 still fails, in the same way, though now trying with menuinst 1.4.19
[image]

Worth noting there is some discussion about menuinst / conda version compatibility [here](https://github.com/ContinuumIO/anaconda-issues/issues/582)

The conda packaged with the installer is 4.14.0, which appears to be the most current Anaconda-built version. 

My base env conda is is 4.13.0. When I `conda create -n pyme20220922 python=3.7 python-microscopy` I pull the latest 20220920 pyme, and the shortcuts work 
[image]
there is not conda associated with that env,
<details>

```
(base) C:\Users\aesb>conda activate pyme20220922

(pyme20220922) C:\Users\aesb>conda list
# packages in environment at C:\Users\aesb\Miniconda3\envs\pyme20220922:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0    anaconda
babel                     2.9.1              pyhd3eb1b0_0    anaconda
backcall                  0.2.0              pyhd3eb1b0_0    anaconda
blas                      1.0                         mkl    anaconda
blosc                     1.19.0               h7bd577a_0    anaconda
bottleneck                1.3.4            py37h080aedc_0    anaconda
brotli                    1.0.9                ha925a31_2    anaconda
brotlipy                  0.7.0           py37h2bbff1b_1003    anaconda
bzip2                     1.0.6            vc14hdec8e7a_1  [vc14]  anaconda
ca-certificates           2022.4.26            haa95532_0    anaconda
cairo                     1.14.12              hf171d8a_3    anaconda
certifi                   2022.6.15        py37haa95532_0    anaconda
cffi                      1.15.0           py37h2bbff1b_1    anaconda
charset-normalizer        2.0.4              pyhd3eb1b0_0    anaconda
cheroot                   8.5.2            py37haa95532_1    anaconda
cherrypy                  18.6.0           py37haa95532_0    anaconda
cloudpickle               2.0.0              pyhd3eb1b0_0    anaconda
colorama                  0.4.4              pyhd3eb1b0_0    anaconda
cryptography              37.0.1           py37h21b164f_0    anaconda
cycler                    0.11.0             pyhd3eb1b0_0    anaconda
cython                    0.29.28          py37hd77b12b_0    anaconda
cytoolz                   0.11.0           py37he774522_0    anaconda
dask-core                 2021.10.0          pyhd3eb1b0_0    anaconda
decorator                 5.1.1              pyhd3eb1b0_0    anaconda
django                    2.1                      py37_0    anaconda
docutils                  0.17.1           py37haa95532_1    anaconda
fftw                      3.3.9                h2bbff1b_1    anaconda
fonttools                 4.25.0             pyhd3eb1b0_0    anaconda
freetype                  2.10.4               hd328e21_0    anaconda
fsspec                    2022.3.0         py37haa95532_0    anaconda
future                    0.18.2                   py37_1    anaconda
hdf5                      1.8.18           vc14h7a021fe_0  [vc14]  anaconda
icc_rt                    2019.0.0             h0cc432a_1    anaconda
icu                       58.2             vc14hc45fdbb_0  [vc14]  anaconda
idna                      3.3                pyhd3eb1b0_0    anaconda
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0              pyhd3eb1b0_0    anaconda
imagesize                 1.3.0              pyhd3eb1b0_0    anaconda
importlib-metadata        4.11.3           py37haa95532_0    anaconda
intel-openmp              2021.4.0          haa95532_3556    anaconda
ipython                   7.31.1           py37haa95532_0    anaconda
jaraco.classes            3.2.1              pyhd3eb1b0_0    anaconda
jaraco.collections        3.3.0              pyhd3eb1b0_0    anaconda
jaraco.functools          3.3.0              pyhd3eb1b0_0    anaconda
jaraco.text               3.5.1            py37haa95532_2    anaconda
jedi                      0.18.1           py37haa95532_1    anaconda
jinja2                    3.0.3              pyhd3eb1b0_0    anaconda
joblib                    1.1.0              pyhd3eb1b0_0    anaconda
jpeg                      9b               vc14h4d7706e_1  [vc14]  anaconda
kiwisolver                1.4.2            py37hd77b12b_0    anaconda
libpng                    1.6.37               h2a8f88b_0    anaconda
[... 85 lines cut ...]
urllib3                   1.26.9           py37haa95532_0    anaconda
vc                        14.2                 h21ff451_1    anaconda
vs2015_runtime            14.27.29016          h5e58377_2    anaconda
wcwidth                   0.2.5              pyhd3eb1b0_0    anaconda
wheel                     0.37.1             pyhd3eb1b0_0    anaconda
win_inet_pton             1.1.0            py37haa95532_0    anaconda
wincertstore              0.2              py37haa95532_2    anaconda
wxpython                  4.0.4            py37ha925a31_0    anaconda
xz                        5.2.5                h8cc25b3_1    anaconda
yaml                      0.1.7            vc14h4cb57cf_1  [vc14]  anaconda
zc.lockfile               2.0                pyhd3eb1b0_0    anaconda
zeroconf                  0.24.0                   py37_3    david_baddeley
zipp                      3.8.0            py37haa95532_0    anaconda
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
zstd                      1.4.4                ha9fde0e_3    anaconda
```

</details>

I updated conda to 4.14.0 and then tried again, and once again get menu shortcuts:
[image]

My guess then is that this is only an issue when pyme is being installed in the root conda environment

sidenote: do we really want to have a `launchWorkers` shortcut at this point?

David-Baddeley, 2022-09-22:
Should definitely scrub the `launchWorkers` shortcut

David-Baddeley, 2022-09-22:
This is super, super, odd, as the `menuinst.install()` does take a `root_prefix` argument in any of the versions we care about.

David-Baddeley, 2022-09-22:
(verified by looking at the menuinst code).

The only thing I can think of is that something else has installed a module whose name shadows menuinst.

David-Baddeley, 2022-09-22:
But I don't see anything in constructor which would suggest this ...

barentine, 2022-09-22:
[image]

where that `menuinst.install` lives in Miniconda3\lib\site-packages\menuinst\__init__.py

barentine, 2022-09-22:
Should have also noted that this was from `C:\Users\aesb\AppData\Local\python-microscopy\Lib\_nsis.py`

David-Baddeley, 2022-09-22:
cf:

https://github.com/conda/menuinst/blob/1.4.19/menuinst/__init__.py

Ditto for 1.4.18

Although, not present in 1.4.17

David-Baddeley, 2022-09-22:
Which pretty much leaves us with the option/hypothesis that conda has packaged an old menuinst as 1.4.19 (in one of the channels), and that channel priority is different with constructor and a normal install.

barentine, 2022-09-22:
Here is the info from the (failed) installer

<details>

```
(base) C:\Users\aesb>conda info --envs
# conda environments:
#
                         C:\Users\aesb\AppData\Local\python-microscopy
base                  *  C:\Users\aesb\Miniconda3
pyme20220922             C:\Users\aesb\Miniconda3\envs\pyme20220922
pyme20220922_2           C:\Users\aesb\Miniconda3\envs\pyme20220922_2

(base) C:\Users\aesb>conda activate C:\Users\aesb\AppData\Local\python-microscopy

(C:\Users\aesb\AppData\Local\python-microscopy) C:\Users\aesb>conda list
# packages in environment at C:\Users\aesb\AppData\Local\python-microscopy:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0
babel                     2.9.1              pyhd3eb1b0_0
backcall                  0.2.0              pyhd3eb1b0_0
blas                      1.0                         mkl
blosc                     1.15.0               h7bd577a_0
bottleneck                1.3.5            py37h080aedc_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2022.07.19           haa95532_0
cairo                     1.16.0               he04af86_2
certifi                   2022.9.14        py37haa95532_0
cffi                      1.15.1           py37h2bbff1b_0
charset-normalizer        2.0.4              pyhd3eb1b0_0
cheroot                   8.5.2            py37haa95532_1
cherrypy                  18.6.0           py37haa95532_0
cloudpickle               2.0.0              pyhd3eb1b0_0
colorama                  0.4.5            py37haa95532_0
conda                     4.14.0           py37haa95532_0
conda-package-handling    1.9.0            py37h8cc25b3_0
cryptography              37.0.1           py37h21b164f_0
cycler                    0.11.0             pyhd3eb1b0_0
cython                    0.29.32          py37hd77b12b_0
cytoolz                   0.11.0           py37he774522_0
dask-core                 2021.10.0          pyhd3eb1b0_0
decorator                 5.1.1              pyhd3eb1b0_0
django                    2.1                      py37_0
docutils                  0.18.1           py37haa95532_3
fftw                      3.3.9                h2bbff1b_1
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.10.4               hd328e21_0
fsspec                    2022.7.1         py37haa95532_0
future                    0.18.2                   py37_1
glib                      2.69.1               h5dc1a3c_1
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                 ha925a31_3
idna                      3.3                pyhd3eb1b0_0
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.19.3           py37haa95532_0
imagesize                 1.4.1            py37haa95532_0
importlib-metadata        4.11.3           py37haa95532_0
intel-openmp              2021.4.0          haa95532_3556
ipython                   7.31.1           py37haa95532_1
jaraco.classes            3.2.1              pyhd3eb1b0_0
jaraco.collections        3.3.0              pyhd3eb1b0_0
jaraco.functools          3.3.0              pyhd3eb1b0_0
jaraco.text               3.5.1            py37haa95532_2
[... 108 lines cut ...]
traitlets                 5.1.1              pyhd3eb1b0_0
traits                    6.2.0            py37h2bbff1b_0
traitsui                  7.1.0                      py_0
typing_extensions         4.3.0            py37haa95532_0
ujson                     5.4.0            py37hd77b12b_0
urllib3                   1.26.11          py37haa95532_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.37.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2              py37haa95532_2
wxpython                  4.0.4            py37ha925a31_0
xz                        5.2.5                h8cc25b3_1
yaml                      0.2.5                he774522_0
zc.lockfile               2.0                pyhd3eb1b0_0
zeroconf                  0.24.0                   py37_3    david_baddeley
zipp                      3.8.0            py37haa95532_0
zlib                      1.2.12               h8cc25b3_3
zstd                      1.5.2                h19a0ad4_0

(C:\Users\aesb\AppData\Local\python-microscopy) C:\Users\aesb>
```

</details>

David-Baddeley, 2022-09-22:
This gets weirder and weirder ... I've checked the individual packages in the conda channels

- 1.4.19 version in the `anaconda` channel is correct (**but** we don't use the anaconda channel when creating the  installer)
- 1.4.19 version in the `main` channel (the one we use) is correct.

David-Baddeley, 2022-09-22:
(downloaded the the package, extracted, looked at menuinst.__init__)

barentine, 2022-09-22:
Right. Interestly, my other environments don't explicitly list menuinst? I believe that's because it's tied up with conda, and only lives in the base environment (same as conda)

<details>

```
(pyme20220922) C:\Users\aesb>conda list
# packages in environment at C:\Users\aesb\Miniconda3\envs\pyme20220922:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0    anaconda
babel                     2.9.1              pyhd3eb1b0_0    anaconda
backcall                  0.2.0              pyhd3eb1b0_0    anaconda
blas                      1.0                         mkl    anaconda
blosc                     1.19.0               h7bd577a_0    anaconda
bottleneck                1.3.4            py37h080aedc_0    anaconda
brotli                    1.0.9                ha925a31_2    anaconda
brotlipy                  0.7.0           py37h2bbff1b_1003    anaconda
bzip2                     1.0.6            vc14hdec8e7a_1  [vc14]  anaconda
ca-certificates           2022.4.26            haa95532_0    anaconda
cairo                     1.14.12              hf171d8a_3    anaconda
certifi                   2022.6.15        py37haa95532_0    anaconda
cffi                      1.15.0           py37h2bbff1b_1    anaconda
charset-normalizer        2.0.4              pyhd3eb1b0_0    anaconda
cheroot                   8.5.2            py37haa95532_1    anaconda
cherrypy                  18.6.0           py37haa95532_0    anaconda
cloudpickle               2.0.0              pyhd3eb1b0_0    anaconda
colorama                  0.4.4              pyhd3eb1b0_0    anaconda
cryptography              37.0.1           py37h21b164f_0    anaconda
cycler                    0.11.0             pyhd3eb1b0_0    anaconda
cython                    0.29.28          py37hd77b12b_0    anaconda
cytoolz                   0.11.0           py37he774522_0    anaconda
dask-core                 2021.10.0          pyhd3eb1b0_0    anaconda
decorator                 5.1.1              pyhd3eb1b0_0    anaconda
django                    2.1                      py37_0    anaconda
docutils                  0.17.1           py37haa95532_1    anaconda
fftw                      3.3.9                h2bbff1b_1    anaconda
fonttools                 4.25.0             pyhd3eb1b0_0    anaconda
freetype                  2.10.4               hd328e21_0    anaconda
fsspec                    2022.3.0         py37haa95532_0    anaconda
future                    0.18.2                   py37_1    anaconda
hdf5                      1.8.18           vc14h7a021fe_0  [vc14]  anaconda
icc_rt                    2019.0.0             h0cc432a_1    anaconda
icu                       58.2             vc14hc45fdbb_0  [vc14]  anaconda
idna                      3.3                pyhd3eb1b0_0    anaconda
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0              pyhd3eb1b0_0    anaconda
imagesize                 1.3.0              pyhd3eb1b0_0    anaconda
importlib-metadata        4.11.3           py37haa95532_0    anaconda
intel-openmp              2021.4.0          haa95532_3556    anaconda
ipython                   7.31.1           py37haa95532_0    anaconda
jaraco.classes            3.2.1              pyhd3eb1b0_0    anaconda
jaraco.collections        3.3.0              pyhd3eb1b0_0    anaconda
jaraco.functools          3.3.0              pyhd3eb1b0_0    anaconda
jaraco.text               3.5.1            py37haa95532_2    anaconda
jedi                      0.18.1           py37haa95532_1    anaconda
jinja2                    3.0.3              pyhd3eb1b0_0    anaconda
joblib                    1.1.0              pyhd3eb1b0_0    anaconda
jpeg                      9b               vc14h4d7706e_1  [vc14]  anaconda
kiwisolver                1.4.2            py37hd77b12b_0    anaconda
libpng                    1.6.37               h2a8f88b_0    anaconda
libtiff                   4.0.8           vc14h04e2a1e_10  [vc14]  anaconda
libwebp                   1.2.2                h2bbff1b_0    anaconda
locket                    1.0.0            py37haa95532_0    anaconda
lz4-c                     1.9.3                h2bbff1b_1    anaconda
lzo                       2.10             vc14h0a64fa6_1  [vc14]  anaconda
markupsafe                2.1.1            py37h2bbff1b_0    anaconda
matplotlib                3.5.1            py37haa95532_1    anaconda
matplotlib-base           3.5.1            py37hd77b12b_1    anaconda
matplotlib-inline         0.1.2              pyhd3eb1b0_2    anaconda
mkl                       2021.4.0           haa95532_640    anaconda
[... 68 lines cut ...]
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.1              py37h2bbff1b_0    anaconda
traitlets                 5.1.1              pyhd3eb1b0_0    anaconda
traits                    6.2.0            py37h2bbff1b_0    anaconda
traitsui                  7.1.0                      py_0    anaconda
typing_extensions         4.1.1              pyh06a4308_0    anaconda
ujson                     5.1.0            py37hd77b12b_0    anaconda
urllib3                   1.26.9           py37haa95532_0    anaconda
vc                        14.2                 h21ff451_1    anaconda
vs2015_runtime            14.27.29016          h5e58377_2    anaconda
wcwidth                   0.2.5              pyhd3eb1b0_0    anaconda
[... cut ...]
xz                        5.2.5                h8cc25b3_1    anaconda
yaml                      0.1.7            vc14h4cb57cf_1  [vc14]  anaconda
zc.lockfile               2.0                pyhd3eb1b0_0    anaconda
zeroconf                  0.24.0                   py37_3    david_baddeley
zipp                      3.8.0            py37haa95532_0    anaconda
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
zstd                      1.4.4                ha9fde0e_3    anaconda
```

</details>

David-Baddeley, 2022-09-22:
Hmm - did you get menu shortcuts with those?

barentine, 2022-09-22:
I did! Presumably for those, it used my base environment menuinst `menuinst                  1.4.16           py37he774522_1    anaconda`

David-Baddeley, 2022-09-22:
I'm thinking it might be related to this ... https://github.com/conda/constructor/pull/474

constructor uses `conda-standalone` rather than conda, which is a py2exed version of conda. Maybe it's built with an old menuinst ...

David-Baddeley, 2022-09-22:
Much as I'm loath to suggest it, we might try using `conda-forge` for `conda-standalone`

David-Baddeley, 2022-09-22:
https://github.com/conda-forge/conda-standalone-feedstock/commits/main/recipe/meta.yaml

They seem to be using a recent menuinst

David-Baddeley, 2022-09-22:
@barentine  - Done a build using conda-forge's conda-standalone (didn't bump the version so it will have over-ridden the previous installer in the `automated_builds` folder). Could you give that a try?

barentine, 2022-09-23:
Nicely done, David! 
[image]

Opened the one without a conda env name after it
[image]

and it would appear to be the installer one:
[image]

-----

I would say our to-do here is then just to remove `launchworkers`

jaimergp, 2022-09-23:
Hi team! Sorry about the conda-standalone cross-bundling complexities. `menuinst` is bundled in that frozen conda executable, as you have now figured out. The good news is that you don't need to process the channel metadata or include conda-forge in the solver at all because you can grab conda-standalone directly (via curl or whatever), extract it and point to the `.exe` with the `--conda-exe` flag in `constructor`. You can also use `micromamba` for that.

I have been working on `constructor` and `menuinst` for a few months and we are preparing a big release. I'd love to hear early feedback from the new versions once we start testing, so let me know if you are interested in the betas and so on.

Also, feel free to ping me for the `constructor` related issues! <3

jaimergp, 2022-09-23:
>  I did! Presumably for those, it used my base environment menuinst `menuinst 1.4.16 py37he774522_1 anaconda`

This is definitely a bug. conda-standalone should be... standalone :D If you can reproduce, I'd be super grateful if you can post an issue in `conda/constructor` @barentine

David-Baddeley, 2022-09-26:
@jaimergp - thanks for your work on constructor and menuinst - it looks like support for shortcuts on Mac as well as windows is in the pipeline which will be very, very, welcome. Unfortunately we probably don't have the bandwidth to do much beta testing - these days it seems to be about all we can do to keep our current build pipeline functioning. 

Thanks again,
David

closed 2023-09-18 by David-Baddeley