python-microscopy/python-microscopy #918: Downloading H5 from PYME Cluster[BUG]

ecourchaine, 2021-03-03:
**Describe the bug**
When attempting to download from the cluster, the window hangs for a little while and then errors with the message pasted below. 
Environment:

Request Method: GET
Request URL: http://10.150.7.130:8889/files/download/Bergamot/2021_3_2/RW_overview_210302/488.pcs/?type=h5

Django Version: 2.1.7
Python Version: 3.6.9
Installed Applications:
['django.contrib.admin',
 'django.contrib.auth',
 'django.contrib.contenttypes',
 'django.contrib.sessions',
 'django.contrib.messages',
 'django.contrib.staticfiles',
 'clusterbrowser',
 'clusterstatus',
 'localization',
 'recipes',
 'accounts']
Installed Middleware:
[]

Traceback:

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/exception.py" in inner
  34.             response = get_response(request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  126.                 response = self.process_exception_by_middleware(e, request)

File "/home/smeagol/miniconda3/lib/python3.6/site-packages/django/core/handlers/base.py" in _get_response
  124.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/home/smeagol/code/python-microscopy/PYME/cluster/clusterUI/clusterbrowser/views.py" in file
  43.             response['Content-Length'] = os.path.getsize(outf.name)

File "/home/smeagol/miniconda3/lib/python3.6/genericpath.py" in getsize
  50.     return os.stat(filename).st_size

Exception Type: FileNotFoundError at /files/download/Bergamot/2021_3_2/RW_overview_210302/488.pcs/
Exception Value: [Errno 2] No such file or directory: '/tmp/tmpx7d13nus.h5'

David-Baddeley, 2021-03-03:
Is this a real cluster or ClusterOfOne?

barentine, 2021-03-03:
real cluster, clusterUI's running on my desktop. Think it's #746

David-Baddeley, 2021-04-18:
duplicate of #956

closed 2021-04-18 by David-Baddeley