python-microscopy/python-microscopy #923: [BUG] pymevis doesn't launch on py36 due to wx flag error

zacsimile, 2021-03-04:
```Traceback (most recent call last):
  File "C:\Users\zrc4\Miniconda3\envs\pyme-latest-36\lib\site-packages\PYME\LMVis\VisGUI.py", line 443, in OnInit
    self.main = VisGUIFrame(None, self.filename, use_shaders=self.use_shaders, cmd_args=self.cmd_args)
  File "C:\Users\zrc4\Miniconda3\envs\pyme-latest-36\lib\site-packages\PYME\LMVis\VisGUI.py", line 186, in __init__
    self._recipe_editor = recipeGui.RecipeView(self, self._recipe_manager)
  File "C:\Users\zrc4\Miniconda3\envs\pyme-latest-36\lib\site-packages\PYME\recipes\recipeGui.py", line 345, in __init__
    vsizer.Add(self.bApply, 0, wx.ALL|wx.ALIGN_CENTER_VERTICAL, 2)
wx._core.wxAssertionError: C++ assertion "!(flags & wxALIGN_CENTRE_VERTICAL)" failed at ..\..\src\common\sizer.cpp(2147) in wxBoxSizer::DoInsert(): Vertical alignment flags are ignored in vertical sizers
OnInit returned false, exiting...
Error in atexit._run_exitfuncs:
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at ..\..\src\common\wincmn.cpp(470) in wxWindowBase::~wxWindowBase(): any pushed event handlers must have been removed```

David-Baddeley, 2021-03-04:
is this a conda install, of a source install? pyme-depends or pip? You'd usually see this when using a newer version of wx than our pinned version (see also #657).

zacsimile, 2021-03-04:
Conda install. Just realized it installed `wxpython=4.1.1`. Downgrading to `4.0.4` fixed it. `pyme-depends-strict` should take care of this.

David-Baddeley, 2021-03-04:
Conda install still shouldn't install the wrong version, as it's should also be pinned in pyme-depends (I see that it isn't - we should fix).

David-Baddeley, 2021-03-04:
now pinned in `pyme-depends`, should revisit once we look at #657 again. Need some way of finding all the bad sizer configs (these don't raise deprecation warnings on 4.0.x, but are a hard fail on on 4.1.x).  Really need search logic which evaluates to something like `wx.ALIGN_CENTER_VERTICAL in sizer.Add() and sizer.has_flag(wx.VERTICAL)` which is going to be hard to do in a regex or similar.

closed 2021-03-04 by David-Baddeley